NVIDIA/open-gpu-kernel-modules #1: Welcome to Linux

aliyuchang33, 2022-05-11:
Finally, you reached out to us. Congrats. Btw I use arch.

MedzikUser, 2022-05-11:
Oh yes

danielgarciagarcia, 2022-05-11:
I'm so glad this is happening. Thank you for open sourcing this driver. This can be a true game changer for Linux Desktop in the future!

NayamAmarshe, 2022-05-11:
Made my day😁 Things are headed in the right direction, for both Nvidia and Linux

vtheskeleton, 2022-05-11:
Truly a arch btw moment

BieHDC, 2022-05-11:
Lets see if linus changes his mind aswell.

pointydev, 2022-05-11:
the year of the linux desktop, now 20xx instead of 22xx.

KaiserBloo, 2022-05-11:
gaming

cinnx, 2022-05-11:
LET'S GOOOOOO

rootqa, 2022-05-11:
this is great!

NextWork123, 2022-05-11:
I hope they do something good, we will see btw i use gentoo

bluewww, 2022-05-11:
2022 is not so bad after all

Thepiotug, 2022-05-11:
mfw nvidia actually open sourced their driver
[image]

a-weeb-programmer, 2022-05-11:
yayay
you won't regret this nvidia dudes. You all will have hundreds of potential contributors rather than being limited to a single team of devs

Lucia-Ziesler, 2022-05-11:
Nothing is certain anymore

nodgear, 2022-05-11:
Nvidia just made a great step towards dominating another market, a small one, but yet another market.
Those drivers should now (hopefully) receive great help from the community

MartinK7, 2022-05-11:
Finally! Thank you Nvidia.

makejohn, 2022-05-11:
what？the world coming to the end？？？
这是要世界末日了吗？

IMS212, 2022-05-11:
is user-mode eventually coming?

pixelplanetdev, 2022-05-11:
[image]

seadog007, 2022-05-11:
Good job Nvidia. Hope to see more open source contribution overtime.

NextWork123, 2022-05-11:
We need to do a party guys WELCOME TO LINUX NVIDIAAA! 
[image]

alexisgeoffrey, 2022-05-11:
this repo issue tracker is going to be unusable for a bit lol

RealTheSunCat, 2022-05-11:
We love you Nvidia!!!

hwa-ryun, 2022-05-11:
You just made my day Nvidia

leetfin, 2022-05-11:
wauw

pixelplanetdev, 2022-05-11:
Thread Theme
https://www.youtube.com/watch?v=y9kkXTucnLU

Zanthed, 2022-05-11:
the ransomware actually worked

KookiesGH, 2022-05-11:
Amazing work! The community will be very grateful

Wonderfall, 2022-05-11:
I was there

Zanthed, 2022-05-11:
@Wonderfall cap

lberrymage, 2022-05-11:
Poggers

That is all. Truly a historic occasion.

flawedworld, 2022-05-11:
add me to the screenshot

nodgear, 2022-05-11:
> this repo issue tracker is going to be unusable for a bit lol

nah, everyone is being polite and using this specific issue.
it should be a topic.... but at least it's contained.

mammothrept, 2022-05-11:
[image]

NextWork123, 2022-05-11:
> add me to the screenshot

Add me too please i want to be famous xd

iusearch, 2022-05-11:
Somebody submits to `AUR`. NOW!!!!!!!!!!!!

pappacena, 2022-05-11:
<3

Zi-SH, 2022-05-11:
A surprise, to be sure, but a welcome one.

Hi-Ray, 2022-05-11:
<3

bigswag420, 2022-05-11:
among us

bram-pkg, 2022-05-11:
❤️

goosvorbook, 2022-05-11:
[image]

TwinniDev, 2022-05-11:
very cool nvidia :)
[image]

2minwia, 2022-05-11:
among us sussy approved

NextWork123, 2022-05-11:
i'm the impostor

> among us sussy approved

gaussandhisgun, 2022-05-11:
340 driver when
(quadro fx 570m user here)

ubscal, 2022-05-11:
This is so exciting

koyuawsmbrtn, 2022-05-11:
I'm more concerned about my GTX 750 and GTX 970

Hatry1337, 2022-05-11:
I hope this finally kill w*ndows.
(use arch btw)

1Lighty, 2022-05-11:
interesting

LukasDoesDev, 2022-05-11:
Very cool move from Nvidia (i use Arch, btw) (I use an AMD card because they had an official FOSS driver before)

pixelplanetdev, 2022-05-11:
[image]

systemscape, 2022-05-11:
> I hope this finally kill w*ndows.
> (use arch btw) 

[image]

lynxnathan, 2022-05-11:
Well, here is an early christmas present. Thanks for all those involved in pushing and approving this!

LukasDoesDev, 2022-05-11:
> [image]
> 

And you made #2 ... Interesting

bittin, 2022-05-11:
nice using an old GTX 760, thats not supported in linux-mainline, so have to use linux-lts on my Arch Linux box
but with this news maybe i will buy an NVIDIA card and not an AMD one, as i been a loyal NVIDIA fan but my card is old and crusty and goes end of support for real in October 2023, so have to buy a new GPU this or next year

thebeanogamer, 2022-05-11:
[image]

Aang23, 2022-05-11:
That was so unexpected... But honestly, can only say thanks NVIDIA on this one! With some luck the userspace (MESA) aspect will go opensource sometime as well :-)

ThatOneCalculator, 2022-05-11:
Yea, this is "opened source" but I'm perusing through the code now and a lot of it links back to binary blobs. Dissapointing.

nidefawl, 2022-05-11:
YES! Finally!
Now throw the legacy one out there as well please

[280 more replies]

closed 2022-05-27 by iusearch