NVIDIA/open-gpu-kernel-modules #11: DKMS Infrastructure

sempervictus, 2022-05-11:
As far as i know, most users consume the old proprietary driver as a DKMS package for their distro to permit building against upstream/binary kernels which they are not compiling from source every time they update the kernel or driver pacakge. If the relevant DKMS build files and structure are in this repository, it's less likely to end up with the "per-distro-implementation" schism normally left up to the downstream packagers.

aritger, 2022-05-11:
Good idea.  Tagging as an enhancement.

aritger, 2022-05-11:
For reference, we've filed NVIDIA-internal bug 3642567 for this request.

sempervictus, 2022-05-11:
Thank you.
I'm guessing you could rip the DKMS setups from Arch Linux packaging (its just `.sh` scripts) or Ubuntu/RHEL-types pretty easily.
If the Makefile then contains a `make dkms` option which takes a `DESTDIR` paramter for chroot-targets (pacakge scaffold roots), then all downstream users could leverage it to build their respective DKMS packages from this repo.

svenstaro, 2022-05-12:
I might be misunderstanding the intention of this issue, but as far as I'm concerned, I think I managed to get dkms working fine in the [Arch package](https://archlinux.org/packages/testing/x86_64/nvidia-open-dkms/). Please give it a test and see whether it's working as intended. See [here](https://github.com/archlinux/svntogit-packages/blob/packages/nvidia-open/trunk/PKGBUILD) for the PKGBUILD.

ryao, 2022-05-12:
Not every distribution uses DKMS, so please do not make DKMS a hard requirement. Otherwise, you might see some per distribution patching in the opposite direction. Doing it like ZFSOnLinux where DKMS is a wrapper around the existing build system would be great.

Making DKMS a wrapper around the existing build system rather than rewriting the build system to be DKMS should go without saying, but just in case, I am saying it anyway.