NVIDIA/open-gpu-kernel-modules #12: error: implicit declaration of function 'sys_close' [-Werror=implicit-function-declaration]

breisig, 2022-05-11:
**NVIDIA Driver Version**
Latest version of 'main' as of today.

**GPU**
3090

**Describe the bug**
Trying to compile the code and getting the following error.

```
make[1]: Entering directory '/tmp/open-gpu-kernel-modules/kernel-open'
make[2]: Entering directory '/usr/src/linux-5.17.6'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 11.3.1 20220429
  You are using:           cc (GCC) 11.3.1 20220429
  CC [M]  /tmp/open-gpu-kernel-modules/kernel-open/nvidia/nv-caps.o
/tmp/open-gpu-kernel-modules/kernel-open/nvidia/nv-caps.c: In function 'nv_cap_close_fd':
/tmp/open-gpu-kernel-modules/kernel-open/nvidia/nv-caps.c:567:5: error: implicit declaration of function 'sys_close' [-Werror=implicit-function-declaration]
  567 |     sys_close(fd);
      |     ^~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:288: /tmp/open-gpu-kernel-modules/kernel-open/nvidia/nv-caps.o] Error 1
make[2]: *** [Makefile:1831: /tmp/open-gpu-kernel-modules/kernel-open] Error 2
make[2]: Leaving directory '/usr/src/linux-5.17.6'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/tmp/open-gpu-kernel-modules/kernel-open'
make: *** [Makefile:50: modules] Error 2
```

Running with the following
gcc 11.3.1
glibc 2.35
binutils 2.38
Linux kernel: 5.17.6

**To Reproduce**
Download the latest code from git and type 'make'

**Expected behavior**
Compiles without any issue,

aritger, 2022-05-11:
Thanks for the report, breisig.  I assume you see the same problem installing the 515.43.04 NVIDIA driver?  (i.e., this problem is not specific to open-gpu-kernel-modules repo?)

sempervictus, 2022-05-11:
I'm building fine on GCC 12 and 11, for Grsecurity and vanilla targets. Guessing there's a header missing somewhere.

atiensivu, 2022-05-11:
Does it compile if you remove the -Werror requirement?

aritger, 2022-05-12:
@breisig I don't think this is a general compatibility problem between these modules and 5.16.7.

sys_close() is called in nv-cap.c if neither close_fd() nor __close_fd() are available.

Could you please paste the output of these commands?

`cat /tmp/open-gpu-kernel-modules/kernel-open/conftest/symbols.h`

`grep close_fd /lib/modules/`uname -r`/build/Module.symvers`

breisig, 2022-05-12:
@aritger Ok, Quick update. I just updated to Linux kernel 5.17.7 this morning, re-downloaded the code from git and it worked without issue without having to do my hacking below

I did manage to fix it yesterday doing the following hack.
1) I didn't have a Module.symvers [I cleaned out my kernel source as I compile from source] initially until I recompiled the modules from scratch [/lib/modules/uname -r/build/Module.symvers]
2) I had to edit the kernel source Makefile [/usr/src/linux-5.17.7/Makefile] and removed the -Werror=implicit-function-declaration from that file and it compiled without any issue.

closed 2022-05-12 by breisig

ptr1337, 2022-05-13:
@aritger 

Is this the correct way? Actually after upgrading my kernel from 5.16.6 to 5.16.7 my dkms was broken. Actually I thought it is caused by the nv-ap.c bug. 

Also building the dkms module with clang, does write many warnings. If its not the same I going to create a issue with logs

aritger, 2022-05-16:
> @aritger
> 
> Is this the correct way? Actually after upgrading my kernel from 5.16.6 to 5.16.7 my dkms was broken. Actually I thought it is caused by the nv-ap.c bug.
> 
> Also building the dkms module with clang, does write many warnings. If its not the same I going to create a issue with logs

I don't think the issue you are describing matches the originally reported problem in this Issue.

peads, 2024-08-07:
if you're still experiencing this issue (especially in Gentoo w/kernel v5.15.x),  rebuilding the kernel with `CONFIG_TRIM_UNUSED_KSYMS=y`, or--if you're not lazy like me--whitelisting it via `CONFIG_UNUSED_KSYMS_WHITELIST` might be avenues to explore.