NVIDIA/open-gpu-kernel-modules #14: Extranous \`\` in a makefile script

kiroma, 2022-05-11:
**NVIDIA Driver Version**
515.43.04

**Describe the bug**
When building the kernel modules, an error popped up `/bin/sh: line 1: localhost.localdomain: command not found` caused by extra \`\` around `$$(HOSTNAME)` in the command `GENERATE_NVIDSTRING` in [utils.mk:L527](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/utils.mk#L527)

aritger, 2022-05-11:
Thanks for the report, kiroma.  It looks like utils.mk treats $(HOSTNAME) as a program, rather than a value.  Does it work if you make this change?

```
diff --git a/utils.mk b/utils.mk
index d09807eb2657..22eedeb8a004 100644
--- a/utils.mk
+++ b/utils.mk
@@ -107,7 +107,9 @@ CHMOD                 ?= chmod
 OBJCOPY               ?= objcopy
 XZ                    ?= xz
 WHOAMI                ?= whoami
-HOSTNAME              ?= hostname
+ifndef HOSTNAME
+  HOSTNAME := $(shell hostname)
+endif
 
 NV_AUTO_DEPEND        ?= 1
 NV_VERBOSE            ?= 0
@@ -524,7 +526,7 @@ define GENERATE_NVIDSTRING
   # g_nvid_string.c depends on all objects except g_nvid_string.o, and version.mk
   $(NVIDSTRING): $$(filter-out $$(call BUILD_OBJECT_LIST,$$(NVIDSTRING)), $(3)) $$(VERSION_MK)
        $(at_if_quiet)$$(MKDIR) $$(dir $$@)
-       $(at_if_quiet)$$(ECHO) "const char $(1)[] = \"nvidia id: NVIDIA $$(strip $(2)) for $$(TARGET_ARCH)  $$(NVIDIA_VERSION)  $$(NVIDSTRING_BUILD_TYPE_STRING)  (`$$(WHOAMI)`@`$$(HOSTNAME)>
+       $(at_if_quiet)$$(ECHO) "const char $(1)[] = \"nvidia id: NVIDIA $$(strip $(2)) for $$(TARGET_ARCH)  $$(NVIDIA_VERSION)  $$(NVIDSTRING_BUILD_TYPE_STRING)  (`$$(WHOAMI)`@$$(HOSTNAME))>
        $(at_if_quiet)$$(ECHO) "const char *const p$$(strip $(1)) = $(1) + 11;" >> $$@;
 endef
 
```

kiroma, 2022-05-11:
Yep works just fine.

aritger, 2022-05-11:
Thanks for testing it.  I'll get that change applied internally, and it should be included in a future release.

PAR2020, 2022-05-17:
Internal bug 3642605. Fix in progress.

aritger, 2022-05-31:
This should be fixed in 515.48.07.  Please let me know if you continue to have troubles.  Marking fixed.

closed 2022-05-31 by aritger