NVIDIA/open-gpu-kernel-modules #17: In another update, activate RTX IO for deep learning

johnnynunez, 2022-05-11:
For people who work with deep learning, it would be amazing that you would enable RTX IO for Turing and Ampere.

PAR2020, 2022-05-12:
Thanks for the suggestion. We are exploring whether or not this is possible, and will get back to you.

johnnynunez, 2022-05-13:
https://github.com/NVIDIA/MagnumIO/issues/5 
It should be supporting by RTX Turing and Ampere as my RTX 3090 FE, because RTX IO is the same function for gaming...

PAR2020, 2022-05-16:
Still digging into this, @johnnynunez. As you pointed out, RTX IO couples with Microsoft Direct Storage for gaming, but MagnumIO works with RTX IO GPUs. Please be patient a little while longer as we sort through all of our issues/PRs and get to this info for you. Thanks.

johnnynunez, 2022-05-17:
@PAR2020 Yes, don't worry, I'm not in a hurry, as long as it's in the pipeline for future versions. I think it's something that can benefit the community a lot.

johnnynunez, 2022-05-19:
@weareai Linux always. But now that linux gaming is also booming, it should be able to be used for both cases.

johnnynunez, 2022-05-19:
> @johnnynunez got it. It seems to be that you are interested in accelerating data transfer from storage to the GPU, right ? Have you considered [GPU Direct Storage for CUDA](https://developer.nvidia.com/gpudirect-storage) on Linux ? Let me know as I am trying to make sure we capture your request if its not already supported for future releases :-)

I understand that magnum IO is not compatible with RTX, only quadro, tesla, A100, etc, in resume, professional GPU's as indicated on the page. I have also seen the fact of using DALI, que es compatible con directstorage, although I think the idea or my idea is that the developer has to touch little code and that it comes integrated directly with Pytorch, for example.

weareai, 2022-06-28:
@johnnynunez  sorry missed this response. 
DALI is supported on many GPUs. In some cases, DALI relies on GDS. GeForce GPUs like the 3090FE aren't supported by the CUDA user mode driver or runtime. In such cases, DALI wont be able to support these features (DALI use of GDS).
Please let me know how we can enable your use case and perhaps find alternate solutions to your application.