NVIDIA/open-gpu-kernel-modules #18: No documentation on which GPUs this driver supports

RobertCochran, 2022-05-11:
There is no bleedingly obvious list of what GPU models this driver does and does not support. It's of course safe to assume that this driver supports all of the latest GPUs and at least a generation back, but beyond that it's difficult to ascertain whether you're in the green here. I know that the proprietary driver fairly regularly drops support for older cards. What options are available for folks that have said older cards?

Please provide a clear list of GPUs are that supported by this driver, thanks.

Aang23, 2022-05-11:
It's very 515, so it will support what the equivalent proprietary driver supports. Architecture-wise, that includes : 
Maxwell, Pascal, Turing, Ampere, Volta.

Personally though, I wouldn't mind if the code for Kepler could be released as well.

Binary-Eater, 2022-05-11:
My personal understanding is that it essentially supports any card that is able to use the GSP firmware. That information seems to be available in the documentation provided with each NVIDIA driver release.

https://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/gsp.html

sempervictus, 2022-05-11:
Based on the source code here and in the last binary driver set, it looks to "just be the next version" of the same driver, so the same chipsets.

9p4, 2022-05-11:
This is limited to only GSP-enabled cards, so 20xx and 30xx cards (Turing and above).

From https://www.phoronix.com/scan.php?page=article&item=nvidia-open-kernel&num=1

> Well, it wouldn't entirely since let's not forget this new kernel driver only works with Turing GPUs and newer. Meanwhile the older sweet spot currently for the Nouveau kernel driver is GTX 600/700 Kepler series (and GTX 750 Maxwell1) but the newer GPUs is where it lacks proper power management firmware / re-clocking and thus the performance is in bad shape. Nouveau on recent generations of NVIDIA GPUs is basically only good enough for driving a display as due to the ability to re-clock to the peak performance state its 3D graphics performance is frankly junk. So the NVIDIA Open Kernel Driver is certainly superior for GeForce RTX 20/30 series while GTX 900 / GTX 10 graphics cards will likely be left in an awkward state outside of the proprietary driver stack.

Aang23, 2022-05-11:
> This is limited to only GSP-enabled cards, so 20xx and 30xx cards (Turing and above).
>
> From https://www.phoronix.com/scan.php?page=article&item=nvidia-open-kernel&num=1

Interesting. I do see some files for Maxwell / Pascal in there though. Was just about to test it on my 1070.

Binary-Eater, 2022-05-11:
> Interesting. I do see some files for Maxwell / Pascal in there though. Was just about to test it on my 1070.

This is from the NVIDIA documentation.

>Turing and later GPUs are capable of using the GSP firmware by setting the kernel module parameter NVreg_EnableGpuFirmware=1. However, note feature and GPU support limitations below.

PAR2020, 2022-05-11:
Hopefully this helps clarify a response, RobertCochran, from the announcement links: https://developer.nvidia.com/blog/nvidia-releases-open-source-gpu-kernel-modules/
Which GPUs are supported by Open GPU Kernel Modules?
Open kernel modules support all Ampere and Turing GPUs. Datacenter GPUs are supported for production, and support for GeForce and Workstation GPUs is alpha quality. Please refer to the Datacenter, NVIDIA RTX, and GeForce [product tables](https://developer.nvidia.com/cuda-gpus) for more details (Turing and above have compute capability of 7.5 or greater).

RobertCochran, 2022-05-11:
Seems clear enough. I'd still like if this were mentioned in the README so that the information stays up to date with the driver and removes the guesswork.

xnox, 2022-05-11:
Shipping `supported-gpus/supported-gpus.json` would be nice, cause then distributions who package this module can correctly declare / detect matching GPUs.

Currently supported-gpus.json is shipped in the .run script, but it would be nice to have it here too.

PAR2020, 2022-05-11:
Both good suggestions, thanks.

aritger, 2022-05-11:
Thanks for the discussion.  Yes, the open kernel modules support Turing and later GPUs.  For datacenter GPUs, that list is currently: https://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/gsp.html

For GeForce and Workstation, product names can make it difficult to infer what GPU architecture is in a product.

Extending supported-gpus/supported-gpus.json with an identifier, and including it in this github repository, is a good idea.

Tagging as an enhancement request.

MrBeeBenson, 2022-05-11:
If I was a beginner I would not personally know where to look to see whether my GTX 760 is supported or not (last supported Linux driver was 470) - I recommend a simple markdown file with a list of supported and unsupported GPUs

iam0day, 2022-05-11:
I agree, I'd like to see an updated list with all officially supported graphics cards.

ghtesting2020, 2022-05-12:
> This is limited to only GSP-enabled cards, so 20xx and 30xx cards (Turing and above).
>....

My laptop has a 1650 max-q which is Turing TU117. So it looks like the 16xx are also supported unless @aritger wants to correct me.

mtijanic, 2022-05-12:
Yes, the 16xx GeForce series are based on Turing and have a GSP; as such they can use the open source driver.

However, GeForce 16xx are not datacenter GPUs and support for them is experimental in this release, so you will have to use [NVreg_OpenRmEnableUnsupportedGpus](http://us.download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/kernel_open.html) module param for now.

I have personally found this wikipedia page to be an excellent mapping between product names and chip/architecture ones:
https://en.wikipedia.org/wiki/List_of_Nvidia_graphics_processing_units#GeForce_16_series

Anything with a codename of `TUxxx` or `GAxxx` should have a GSP microcontroller.

mtijanic, 2022-05-12:
@tajetaje Please see [here](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/19#issuecomment-1124905002), cards prior to 16xx and 20xx simply do not have the necessary HW to use this driver.

tajetaje, 2022-05-12:
> @tajetaje Please see [here](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/19#issuecomment-1124905002), cards prior to 16xx and 20xx simply do not have the necessary HW to use this driver.

Ah, missed that; well hopefully some of the back-ported changes help out then

MartinX3, 2022-05-12:
> > @tajetaje Please see [here](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/19#issuecomment-1124905002), cards prior to 16xx and 20xx simply do not have the necessary HW to use this driver.
> 
> Ah, missed that; well hopefully some of the back-ported changes help out then

They won't backport the driver to older GPU's without the GSP.
Only cards are supported where the important code remains closed source thanks to firmware blob support.
https://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/gsp.html

Nvidia just released this "open souce" skeleton full of blob usage to workaround the kernel restriction.
https://www.phoronix.com/scan.php?page=news_item&px=Linux-Kernel-Blocking-NV-NetGPU

So
1. Nvidia tries to workaround the GPL restriction
2. Nvidia tries to advertise us this PR action as the holy "nvidia supports open source and linux" grail.

smxi, 2022-05-13:
This is a useful discussion. Note that I'm working on adding support for some of these questions into inxi at the moment, but still am collecting data. So far there is no online resource that combines nvidia product IDs with microarchitecture names, so I have to combine those lists manually and use some guesswork to match the legacy microarchitecture names to product ids. Some of this is running in pinxi, the development branch of inxi, already, and more will be added as I get more information and data, and figure out how to present this data in the most useful form.

I should have something basic working today or in the coming next days, but it won't be perfect due to lack of accurate data online. I've done this type of data assembly many times, any intern could have that page of matching tables up in about 2 days, maybe 3 on the outside, but I would only do that if I were paid because it's so boring, I'm looking for more general matches.

It sounds like all cards with product ids listed in the 5.15 page will be supported, I'll have to double check their microarchitectures to confirm that, though that's a real pain due to lack of one single reference resource for this information.

smxi, 2022-05-15:
This is my raw working code with Product IDs. Needless to say, it would be a lot easier to get this data if per microarchitecture product IDs > microarch tables or whatever were available.

These are based on the data in the 515 driver readme. Because I have to use manually generated regex to get each microarch using the data on the nvidia wikipedia pages, I probably missed some product IDs. I honestly scratch my head and wonder why multibillion dollar tech companies can't publish such information in a clear format, it would cost them probably a few hundred dollars max to do it. But nvidia publishes better than most, so there is that. 

```
	%nv_arch = (
	'Maxwell' => {
	'ids' => '06a3|1025|1028|103c|1043|107a|10de|1198|1199|11c9|11dd|137a|137b|' .
	'1380|1381|1382|1391|1392|139a|139b|139d|13b0|13b1|13b2|13b4|13b6|13c0|13c2|13d8|' .
	'13da|13f0|13f1|13f8|13f9|13fa|13fb|1401|1402|1406|1407|1427|1430|1436|1458|1462|' .
	'161a|174d|174e|17aa|17c8|17f0|17f1|19da|2232|2b4c|362c|362f|363f|3640|3647|36b9|' .
	'36bf|505a|861e|86d9|b282|b284|b286|c248|d003',
	'code' => 'GMxxx',
	'process' => 'TSMC-28nm',
	'series' => '515.xx+',
	},
	'Pascal' => {
	'ids' => '07c0|0802|0803|0825|0827|0876|087c|0885|0886|0887|088b|08a1|08a2|' .
	'09c1|1025|1028|1031|1032|103c|1043|1059|10de|114f|11e8|11e9|1217|1221|1225|1226|' .
	'1227|1232|123c|123e|123f|1244|1245|126c|126d|1284|1297|1341|1351|1458|1462|1481|' .
	'149f|14a1|152d|1558|15f7|15f8|15f9|1651|1653|16f0|1707|17aa|17c2|18c1|1a58|1b00|' .
	'1b02|1b06|1b31|1b5e|1b80|1b81|1b82|1b83|1b84|1ba0|1ba1|1ba2|1be0|1be1|1bf0|1c02|' .
	'1c03|1c04|1c06|1c20|1c21|1c22|1c23|1c60|1c61|1c62|1c81|1c82|1c83|1c8c|1c8d|1c8f|' .
	'1c90|1c91|1c92|1c94|1c96|1cce|1d05|1d11|1d13|1d16|1d52|1d72|2000|2001|2009|2266|' .
	'2267|355b|3842|39b9|39ff|8390|8467|8478|84db|84e9|84eb|8519|856a|8581|86e3|9501|' .
	'95e1|95e5',
	'code' => 'GP10x',
	'process' => 'TSMC-16nm',
	'series' => '515.xx+',
	},
	'Volta' => {
	'ids' => '10de|12eb|1d81|1dba',
	'code' => 'GV1xx',
	'process' => 'TSMC-12nm',
	'series' => '515.xx+',
	},
	'Turing' => {
	'ids' => '0032|08a1|08a2|08ea|08eb|08ec|08ed|08ee|08ef|093b|093c|0949|098f|' .
	'0990|09c3|09d1|09d2|09e2|09f3|1001|1003|1006|1007|1025|1028|103c|1043|105e|1069|' .
	'1070|1089|10de|1220|1274|1277|127e|1281|1284|1285|128a|128b|128d|1297|129c|12ae|' .
	'12b0|12b4|12bd|12c5|12c6|12cb|12cc|12cf|12d2|12d9|12f4|1302|130d|131b|131f|1326|' .
	'132a|132d|132e|132f|1342|136f|1375|137f|138f|13b2|13cf|13d9|1401|1402|1414|141f|' .
	'1432|1442|1446|144d|1458|1462|1479|147a|147b|147c|147d|149c|1511|1512|152d|1558|' .
	'156f|15c1|1658|165f|1660|1661|1662|1663|1664|171f|174f|1751|17aa|17ef|185c|1881|' .
	'189c|18d1|1902|1a58|1d05|1d61|1dd1|1dff|1e04|1e07|1e11|1e1f|1e30|1e36|1e51|1e61|' .
	'1e6e|1e78|1e81|1e82|1e83|1e84|1e87|1e89|1e90|1e91|1e93|1eb0|1eb1|1eb5|1eb6|1ec2|' .
	'1ec7|1ed0|1ed1|1ed3|1ef5|1f01|1f02|1f03|1f06|1f07|1f08|1f0a|1f10|1f11|1f12|1f14|' .
	'1f15|1f36|1f42|1f47|1f50|1f51|1f54|1f55|1f76|1f82|1f91|1f95|1f96|1f97|1f98|1f99|' .
	'1f9c|1f9d|1f9f|1fa0|1fdd|2002|2005|2007|2008|2009|200a|200c|2011|2087|2182|2184|' .
	'2187|2188|2191|2192|21c4|21d1|229f|22b8|22ba|22c0|22c1|22c3|22c5|22da|25a6|25a7|' .
	'25a9|25aa|3000|3001|3002|3012|3801|3802|3803|3806|380c|381d|381e|3837|3e30|3f1a|' .
	'3f93|3f95|3fce|3fcf|3fd0|3fee|3ffe|50d3|70d1|75a4|75a5|75a6|75a7|75b1|75b2|75b3|' .
	'75b4|75b5|75b6|76b0|76b1|76b2|76b3|76b4|76b5|78a0|78a1|78a2|78a3|78a4|78a5|8086|' .
	'8572|8573|8574|85fb|85fe|8600|8602|8605|8606|8607|863e|86a4|86a5|86a6|86a7|86b0|' .
	'86b1|86b2|86b3|86b4|86b5|86c6|86c7|86d6|86e7|86e8|8736|8738|8741|8744|8746|8772|' .
	'8780|8782|8783|8785|878a|878b|878d|878e|87a6|87a7|87af|87b3|880e|8815|95e1|97e1|' .
	'c192|c822|c838',
	'code' => 'TUxxx',
	'process' => 'TSMC-12nm',
	'series' => '515.xx+',
	},
	'Ampere' => {
	'ids' => '0032|0990|0baf|1028|103c|1043|10de|1192|1194|1195|1196|13cf|1414|' .
	'1450|145a|145f|1463|147f|1532|1533|157e|1d05|1f36|1f76|20b0|20b2|20b5|20b7|20f1|' . 
	'2203|2204|2206|2208|220a|2216|2235|2414|2420|2460|2482|2484|2486|2487|2488|2489|' .
	'249c|249d|24a0|24dc|24dd|24e0|2503|2504|2507|2508|2520|2523|2560|2563|25a0|25a2|' .
	'25a5|25b6|25bb|25e0|25e2|25e5|8736|8738|8772',
	'code' => 'GAxxx',
	'process' => 'TSMC-7nm',
	'series' => '515.xx+',
	},
	# no ids yet
	# 'Hopper' => {
	# 'ids' => '',
	# 'code' => 'GH1xx',
	# 'process' => 'TSMC-n4',
	# },
	# 'Lovelace' => {
	# 'ids' => '',
	#	'code' => '',
	# 'process' => '',
	# },
	);
```

mtijanic, 2022-05-15:
We detect architectures at runtime by reading the [PMC_BOOT_0](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/src/common/inc/swref/published/nv_ref.h#L106) and comparing the architecture bits with [these values](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/src/common/inc/swref/published/nv_arch.h#L40-L46) (there's also individual implementation constants below).

We currently don't have a handy offline reference table to translate device IDs to architectures, but it certainly sounds like a good thing to publish. We do have a list of all published products and their IDs, from which such a list could be constructed: [g_nv_name_released.h](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/src/nvidia/generated/g_nv_name_released.h)

smxi, 2022-05-15:
That method would be preferred, but this is Perl, so unless it's available in /sys or something like that I couldn't use it, but a real dynamic detection is obviously far superior to a product/pci id detection, which will always miss stuff, and be more difficult to update.

I'll poke around in some nvidia user datasets and see if those architecture bits are exposed anywhere that is readable.

Given how important knowing your card microarchitecture is beccming, simply adding one more column to your otherwise exceptionally well done product name/id files created with each new driver would really be all it takes, those are relatively easy to parse, that's what I'm using, but it's very difficult combining the wikipedia data directly in with that data because many product names and series are actually different architectures depending on various situations. The wikipedia pages don't have product ids, and are also vastly varying in quality and completeness depending on microarchitecture, they were clearly generated by different people with different ideas of what constitutes consistent data, so not ideal.

I'm going to convert my shell filters to perl today to make it easier to build up these lists, until ideally they get published in a more reliable format for 71 > 470 legacy and 515 > product id lists. 

Note that is quite literally just adding one more table column to the published html tables, which should not take more than an hour two to do, it only took me as an outsider having to merge data and no access to internal resources a few hours.

Note that one very strange thing in the wikipedia data is that a page for say kepler will show the product code for fermi or something, a few mixed in, at which point it's hard to know if they made a mistake, or some codes are not actually referring to the microarch names even though they are clearly meant to like TU, GV, GA, and so on. I haven't found any docs for Tegra anywhere so far unless I'm missing it, I see by the header file that is a different category altogether, but one thing at a time.

For now, I'll whip up a perl script to make this more fool proof and repeatable to generate the per arch. id lists, which will make it easy if in the future those arch. are added to the existing html tables, or json data sources, or whatever. 

I believe these changes will already cover most nvidia users in Linux however, it's just labor intensive to maintain it over time, not ideal.

smxi, 2022-05-15:
I've updated the IDs above if they are useful to anyone, those will be going live fairly soon, just need to work out a few last details. Made a Perl tool to do the parsing, that works much better, and is easier to adjust and tweak.

This output is designed to copy/paste directly into inxi data structures, but should be easy for anyone to use / modify, at least until something better shows up.

I'm adding in Fahrenheit > Ampere microarchitectures, though I believe the Tesla IDs are going be off, maybe also the Curie ones, but those are not well documented and are confusing to figure out. I might use the new tool at least to lock down Fermi/Kepler IDs as well, I'll see.

aritger, 2022-05-18:
A subsequent release will include a list of compatible GPUs in the README.md.  Self assigning.

aritger, 2022-06-01:
@smxi, for a machine-parseable format, please see the 'kernelopen' token in supported-gpus/supported-gpus.json of NVIDIA-Linux-x86_64-515.48.07.run and later.

I'm going to mark this Issue fixed:
- In the 515.48.07 release of open-gpu-kernel-modules, the README.md contains a table for human readers.
- In the 515.48.07 release of NVIDIA-Linux-x86_64-515.48.07.run, supported-gpus.json contains the 'kernelopen' token for computer readers.

closed 2022-06-01 by aritger

smxi, 2022-06-01:
Thanks for that one, your current HTML tables are actually fine to parse, I'm done with all the backend stuff to make updates fairly trivial in the future. But I'll check that one out, though it requires downloading and extracting the driver package, then grabbing the file, which is actually a lot more work than the method I'm using, but I'll take a look at it, thanks.

Note that inxi is out now and is showing nvidia gpu architectures, and next inxi will even it out, and also show Intel and AMD gpu architecture data.

Most of the work involved with this stuff is actually more in the research area than the data sources, your HTML tables have always been useful to me in terms of updating support pci ids.

smxi, 2022-06-01:
Just on a technical level, speaking as someone who did gnu/linux nvidia installers for far too long, since around 2008, this data is actually needed before the driver run package is downloaded, not after, since it does no good to know your card is legacy after you've downloaded the full latest driver. My stuff always automated all of this of course so it was transparent to the user. But this json source file should help people who are still interested in making installers since it can be fed into the actual installer that determines which run package to download and install the driver with.

This would be more useful if it were also added as a downloadable resource somewhere, like the way you can click to see the supported products HTML table page if you know how to find it, that same appendix could have a link to this json file as well. I'm sure there are some people who would benefit from this. I doubt I'd download the latest driver every time myself just to get this file, that's kind of overkill.

Is there a  license attached to this data file? public domain, etc? 

The kernelopen 'feature' I assume means it's a turing or ampere microarchitecture? I may add a json parser tool to the nvidia logic, but not unless it actually has data the table doesn't have, but this should help some people I think.  A 'microarch' field name: value would have been nice since that can be a pain to dig up the data on, but I've done that work mostly so from my side it's largely completed, albeit probably has an error or two in the data, but it's good enough.

aritger, 2022-06-01:
Yes, the "kernelopen" feature indicates that the GPU has GSP and therefore can be driven by the open kernel modules.  I.e., is >= Turing.

Note that supported-gpus.json has been present in the .run file for several years; it is only the "kernelopen" feature that is new in 515.48.07.

smxi, 2022-06-01:
Thanks for clarifying. So there's no actual new data in there beyond what I have working already, that's good to know. I don't generally download and look inside the run packages for the drivers, so I would never have found this one, but your HTML tables I think are actually easier to work with as data anyway. 

These json files could be a lot more data rich, but in general, nvidia does the best job of releasing this type of practical and useful data, so it's more wishing it were more rich than anything else, the research to find the data that most of the devs inside nvidia probably know off the top of their heads, or have ready access to, is really tedious and time consuming, but again, nvidia does a better job of it than intel or amd, so your overall efforts are appreciated anyway.

Hoping you can make good progress on the desktop gpu gsp open kernel modules, will be curious to see how long it will take to pull them to beta state at least. But given the gigantic size of the current driver it's probably not going to be something we'll see a full stack free driver for anytime soon, if ever, but still one step at a time.