NVIDIA/open-gpu-kernel-modules #20: Mention OpenRmEnableUnsupportedGpus in README

PiMaker, 2022-05-11:
The error from [this line](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/1739a20efc4acb55fd1dc53dcc66057b70c2613c/src/nvidia/arch/nvalloc/unix/src/osinit.c#L1895) references info about `OpenRmEnableUnsupportedGpus` being in the README, but it isn't mentioned anywhere.

For reference, to load the module on non-datacenter cards the `OpenRmEnableUnsupportedGpus` setting in NVReg must be set to 1. This can be done with module parameters, e.g. a file in `/etc/modprobe.d` reading `options nvidia NVreg_OpenRmEnableUnsupportedGpus=1` seems to do the trick.

I'm writing this from Xorg running on a consumer 2080 Ti with the open source module loaded, so it seems to be not entirely broken, would be nice to have this mentioned somewhere then :)

ryao, 2022-05-12:
It is in the README:

http://us.download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/kernel_open.html

PAR2020, 2022-05-12:
Thanks for pointing out the discrepancy @PiMaker. Issue has been tagged as a documentation bug.

PAR2020, 2022-07-26:
This has been implemented, so closing this issue.

closed 2022-07-26 by PAR2020