NVIDIA/open-gpu-kernel-modules #9: In-tree build option

sempervictus, 2022-05-11:
Given that the current upstream attempt at interacting with NV hardware is basically 
[image],
it would be great to be able to pull this repository into the kernel tree similar to what can be done with OpenZFS, PF_RING, XTables-Addons, etc.
This is mostly a matter of converting the makefiles to the kbuild structure and having a copy script which will throw the relevant files into appropriate locations in a user-provided/default kernel tree.

aritger, 2022-05-11:
Thanks for the suggestion; good idea.  Tagging as a future enhancement.

sempervictus, 2022-05-11:
It looks like your repo has a CLA, and i don't really want to pay hundreds of dollars per hour and email involved to our attorneys to review that, so i can't make pulls requests here. Strongly suggest getting rid of that - people with patent rights and intellectual property efforts can't risk some fine print in a CLA transferring their rights to the repo owner (or changing down the line to make such assignment). Not suggesting thats the intent here, but society has shed any pretense of civility as of late and in this dog-eat-dog paradigm of the 2020's its only prudent to avoid signing documents without legal review (which is expensive when done right).

zfsbot, 2022-05-11:
yep, asking us to sign off on a _specific license_, is way easier than signing off on a CLA.

sempervictus, 2022-05-12:
Totally happy to sign something that says "any contribution to this repository is provided under the dual MIT/GPL license scheme without retention of any rights not defined by those licenses or requirement for the contributor to support users of the code." I'm sure there's a more legally appropriate format, but we're all used to contributing to projects under their license semantics, just not all great at grokking legalese.

ryao, 2022-05-12:
@sempervictus They made it clear that they want any changes made to this driver to go into the proprietary driver, so Nvidia needs a CLA for that. Honestly, I am just happy to see them open sourcing the kernel driver.

MilesBHuff, 2022-05-12:
The CLA should probably be discussed in another issue, so that this one is not bogged-down in tangential discussions.

sempervictus, 2022-05-17:
@ryao - you lurking so and so :grin:, how've you been?
I'm all for them putting whatever code comes in here under BSD/MIT/etc terms into their non-free codebase too. The issue i have with any of these CLAs is that they _can_ be used as a trojan horse against the IP rights of contributors. Highly doubt that NV would think of doing something like that given all of the hype around this repo, but "legal has advised" me not to touch CLAs without review/cost attached.
@MilesBHuff - you're probably right, though it does have bearing on the request since i have a proclivity for just building such things myself and the best i'll be able to do is post a diff in this issue for someone else to contribute under their own name/signature while CLAs are in play.