NVIDIA/open-gpu-kernel-modules #100: Confusion about some file name formats

YixinSong-e, 2022-05-12:
Just confused, I see that most of the files in the generated directory have the "nvoc" suffix, so which specific words is this an abbreviation for? I would be very grateful if I could get an answer.

mtijanic, 2022-05-12:
Hello @YixinSong-e , thank you for your interest.

Explanation follows, but as you read it please bear in mind:
* Same files published here are used by other NVIDIA code, including Windows driver and various microcodes
* Many of the design decisions come with over 20 years of history

What I'm saying is, anyone reading the code here and my explanation will rightly thing "why!?", but I'm asking you to trust me that it makes (more) sense with context :)

Anyway, NVOC stands for "NVIDIA Object Compiler" and it is effectively a preprocessor that allows us to have some syntax sugar in our headers. Practically what it does is allow us to annotate some "methods" of an "object" in ways that later allow for lookups or feature toggle flags, or define per-chip specific behaviors.

This isn't used very much in the current codebase, but once you have a nice hammer a lot of things can be made into a compatible nail, so we have a lot of "objects" that use NVOC. How it typically works is that we have an NVOC-enabled header, like `system.h` which contains annotated C code. Then, the code is preprocessed and NVOC produces `g_system_nvoc.{h,c}` files. The .c file is included in the build, and `system.h` is redirected into `g_system_nvoc.h`.

Most common use here is to annotate "control" methods that are exposed by the kernel driver to usermode. Think of it as a way to annotate ioctl() handlers and have a preprocessor generate the plumbing that dispatches to the correct function based on ioctl command ID.

As you can imagine, this complicates the build process significantly, so we have decided it is better for everyone involved if we just ran NVOC once and published the generated files. This way we didn't have to require custom tooling which would make compiling this project a nightmare.

In the future we hope to modify the tool so that the generated code is more readable and less noisy.

I hope this helps. Please feel free to ask about any specifics.

YixinSong-e, 2022-05-12:
Thank you for your detailed reply, your explanation helped me a lot in understanding the driver code.😀

X547, 2022-05-12:
Are NVOC code generator source code published somewhere?

mtijanic, 2022-05-12:
> Are NVOC code generator source code published somewhere?

They are not. I doubt it would be of any use to anyone else other than to satisfy curiosity, and unfortunately there is a cost with open sourcing anything developed internally. We feel those resources are better spent on improving the published code.

For what it's worth, it is a fork of clang 3.x, extended to understand some custom syntax and produce C output. If you're particularly curious, I can give an example input and output that it produces so you have an idea on what you're missing out on.

mtijanic, 2022-05-13:
Closing this, let's use #148 to consolidate all similar questions. Thanks!

closed 2022-05-13 by mtijanic