NVIDIA/open-gpu-kernel-modules #102: PVS-Studio big list of warnings

ValZapod, 2022-05-12:
[projectcrazy.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8678843/projectcrazy.txt)

For example, 

```
./src/common/nvswitch/kernel/lr10/lr10.c	5957	warn	V519 The 'status' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 5952, 5957.
./src/common/nvswitch/kernel/lr10/soe_lr10.c	2191	warn	V519 The 'pHal->getEmemSize' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 2189, 2191.
./src/nvidia-modeset/src/nvkms-evo.c	2936	warn	V519 The 'appliedHeadMask' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 2927, 2936.
./src/nvidia/arch/nvalloc/unix/src/osapi.c	4867	note	V519 The 'fbSize' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 4856, 4867.
./src/nvidia/src/kernel/gpu/bus/arch/ampere/kern_bus_ga100.c	417	warn	V519 The 'status' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 411, 417.
./src/nvidia/src/kernel/gpu/ce/arch/ampere/kernel_ce_ga100.c	526	warn	V519 The 'status' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 523, 526.
./src/nvidia/src/kernel/gpu/fifo/arch/volta/usermode_api_gv100.c	60	note	V519 The 'status' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 55, 60.
./src/nvidia/src/kernel/gpu/mem_mgr/arch/maxwell/mem_utils_gm107.c	114	warn	V519 The 'memAllocParams.attr' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 113, 114.
./src/nvidia/src/kernel/gpu/mem_mgr/arch/maxwell/mem_utils_gm107.c	797	warn	V519 The 'rmStatus' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 789, 797.
./src/nvidia/src/kernel/gpu/mem_mgr/heap.c	3047	warn	V519 The 'status' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 3042, 3047.
./src/nvidia/src/kernel/gpu/timer/timer.c	1428	warn	V592 The expression was enclosed by parentheses twice: ((expression)). One pair of parentheses is unnecessary or misprint is present.
./src/nvidia/src/kernel/mem_mgr/gpu_vaspace.c	1182	warn	V519 The 'pGVAS->vaLimitInternal' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 1179, 1182.
./src/nvidia/src/kernel/mem_mgr/video_mem.c	182	warn	V519 The 'status' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 170, 182.
./src/nvidia/src/kernel/rmapi/nv_gpu_ops.c	8187	warn	V519 The 'status' variable is assigned values twice successively. Perhaps this is a mistake. Check lines: 8173, 8187.

```

On of those reported before.

mtijanic, 2022-05-12:
Thank you @ValZapod , this is very helpful! I have filed bug 3643455 internally to track these as well.

From a quick skim I already see some that look like real issues.

qarmin, 2022-05-12:
Can you export results as html?
It is a lot of easier to read them with source code attached - https://pvs-studio.com/en/blog/posts/0539/

ValZapod, 2022-05-12:
Sure. Repo should be in /opt/SourceCode
[fullhtml.zip](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8679101/fullhtml.zip)

birdie-github, 2022-05-25:
I've uploaded the HTML to my website: https://itvision.altervista.org/nvidia-vs-pvs-studio/