NVIDIA/open-gpu-kernel-modules #105: G-Sync shown as not supported with opensource driver.

sandikata, 2022-05-12:
**NVIDIA Driver Version**
NVRM version: NVIDIA UNIX Open Kernel Module for x86_64  515.43.04  Release Build  (@)

**GPU**
RTX 3050

**Describe the bug**
G-SYNC reported as unsupported while it is supported.

**To Reproduce**
Open Nvidia Control Panel -> OpenGL (There should be checkbox for G-SYNC) but it's missing

**Expected behavior**
It should have G-SYNC checkbox like on proprietary version

Check screenshots below with open and proprietary version.

https://i.imgur.com/Cwqcovr.png -> opensource
https://i.imgur.com/jFFC2oV.png -> proprietary
https://i.imgur.com/LFJiz1M.png -> opensource
https://i.imgur.com/o9ff7lW.png -> proprietary

https://i.ibb.co/RQ278qH/2022-05-12-15-13-38.jpg -> lenovo g27q-20

Regards,
Rosen Aleksandrov

sandikata, 2022-05-12:
Hello,

That's what I've read before, begin with the opensource version of the driver.

Regards

sandikata, 2022-06-29:
Is that feature supported in any of the recent versions, for example 515-57 ?

mtijanic, 2022-06-29:
Hey there. All releases since the initial opensource driver were point fixes (i.e. 515.xx) and generally don't come with big features. Supporting consumer features, such as G-Sync, is a big upcoming milestone that will definitely be a major release. I can't promise which one specifically, but we'll be sure to feature it prominently in the release notes, and update this issue here.

Thanks.

jrgiacone, 2022-12-17:
Still no gsync support as of today

sandikata, 2022-12-18:
> Still no gsync support as of today

Haven't you realize, this is nvidia. If not works it will not work never ever.
They don't care about customer opinions. They care only about the money.

Monsterovich, 2023-03-11:
Driver 525.85.05. I have the same issue.

aritger, 2023-05-30:
G-Sync is expected to work with the open kernel modules as of 535.43.02, at least on desktop GPUs.  We still have some loose ends to address for notebook; G-Sync on notebooks should come in a future release.

Could I trouble you to retest?  Thanks.

kodatarule, 2023-05-31:
> G-Sync is expected to work with the open kernel modules as of 535.43.02, at least on desktop GPUs. We still have some loose ends to address for notebook; G-Sync on notebooks should come in a future release.
> 
> Could I trouble you to retest? Thanks.

Here it seems to be working, unfortunately the bug with not able to boot into wayland if monitor is above 120hz is still present.

gardotd426, 2023-06-01:
> Here it seems to be working, unfortunately the bug with not able to boot into wayland if monitor is above 120hz is still present.

That explains why I've not been able to log into a Wayland session in months, my monitors run at 165Hz. 

Is one able to lower the RR to 120Hz, log in, and then raise the refresh rate without issue? 

Also, what's up with GSync on Wayland? Does it work at all?

kleidiss, 2023-06-01:
Can VRR even work on AMD + Nvidia laptops

VRR is handled by the AMD i-gpu on Windows as far as i know  on Windows but the way optimus works , nvidia frames are rendered through the i-gpu

kleidiss, 2023-06-03:
I managed to enable freesync on linux by copying the display edid from windows and freesync with nvidia on laptops seems to work on X11 but i got no idea on Wayland since forced v-sync is on

BlueskyFR, 2025-03-24:
What is the supported way to enable G-Sync on Wayland?