NVIDIA/open-gpu-kernel-modules #109: Null pointer dereference in nv-acpi.c

tallossos, 2022-05-12:
**NVIDIA Driver Version**
Git

https://github.com/NVIDIA/open-gpu-kernel-modules/blob/8d7f201155cc257bbef784a7887a6ff737f2fb5d/kernel-open/nvidia/nv-acpi.c#L718-L720
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/8d7f201155cc257bbef784a7887a6ff737f2fb5d/kernel-open/nvidia/nv-acpi.c#L737-L746 There isn't any validation on the return value from `acpi_bus_get_device` unlike other places where  `acpi_bus_get_device` is being used.
`device` might be null when dereferencing.

amada95, 2022-05-12:
I've addressed this issue in [this pull request](https://github.com/NVIDIA/open-gpu-kernel-modules/pull/22)

PAR2020, 2022-07-12:
Since PRs are closed, will close this issue as well.

closed 2022-07-25 by PAR2020