NVIDIA/open-gpu-kernel-modules #114: [Request] RISC-V architecture

jackghg, 2022-05-12:
Support for risc-v arch please.

RealAstolfo, 2022-05-12:
please change label from bug to enhancement. this is not a bug. it has been explicitly said in the README that the supported arches are `TARGET_ARCH=aarch64|x86_64`

Leseratte10, 2022-05-12:
The label is auto-generated by Github for issue reports, there is no way for the reporter to pick a label other than "bug". This needs to be fixed by the Nvidia team to allow for other report types; in the meantime someone with commit access has to correct all these labels for new reports.

PAR2020, 2022-05-12:
We just added a Feature Request template, which should help. Our team is going through and manually changing any other issues affected. Thank you for your patience.

ValZapod, 2022-05-13:
> is no way for the reporter to pick a label other than "bug".

You can just select to open a new bare issue. Sigh.

mtijanic, 2022-05-13:
#152  :)