NVIDIA/open-gpu-kernel-modules #119: Are there plans to incorporate the GPL driver into linux-next?

Neirth, 2022-05-12:
I'm curious to know if your driver will eventually be incorporated into the development branch of the Linux kernel, as AMD and Intel already do about it.

I fully understand that it may not be a priority given that you will be focusing efforts on managing the community around this new driver, but that being said, concerns.

Evernow, 2022-05-12:
They mentioned that is a long term goal: https://developer.nvidia.com/blog/nvidia-releases-open-source-gpu-kernel-modules/

> There are plans to work on an upstream approach with the Linux kernel community and partners such as Canonical, Red Hat, and SUSE.

airlied, 2022-05-12:
" The current codebase does not conform to the Linux kernel design conventions and is not a candidate for Linux upstream."

karolherbst, 2022-05-12:
In the meantime until something, whatever that may be in the end, happens, the best random users can do is to help figuring out nouveau bugs with this code, as we (as in nouveau developers) are planning to do the same thing as well.

This should really help out fixing some nasty bugs.

aritger, 2022-05-12:
Yes, what @arlied and @karolherbst said.  Closing.

closed 2022-05-12 by aritger