NVIDIA/open-gpu-kernel-modules #120: NVRM nvAssertOkFailedNoLog: Assertion failed: Generic operating system error

Peter-Levine, 2022-05-12:
**NVIDIA Driver Version**
Please write the version of the NVIDIA driver you are using.

515.43.04

Gentoo Linux package  `x11-drivers/nvidia-drivers-515.43.04:0/515::gentoo`.

**GPU**
Please write the particular model of NVIDIA GPU you are using.

NVIDIA GeForce RTX 3070 Laptop GPU

**Describe the bug**
Please write a clear and concise description of what the bug is.

Binary works fine, but when built with open-source (i.e., `kernel-open` is set as a USE flag), and rebooted in discrete GPU mode,  this error is displayed in `dmesg`:
```
[    4.668053] NVRM nvAssertOkFailedNoLog: Assertion failed: Generic operating system error [NV_ERR_OPERATING_SYSTEM] (0x00000059) returned from ((rpc_message_header_v *)pKernelGsp->pRpc->message_buffer)->rpc_result @ kernel_gsp.c:2061
[    4.668059] NVRM nvAssertOkFailedNoLog: Assertion failed: Generic operating system error [NV_ERR_OPERATING_SYSTEM] (0x00000059) returned from kgspWaitForRmInitDone(pGpu, pKernelGsp) @ kernel_gsp_ga102.c:211
[    4.668062] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ kernel_gsp_ga102.c:216
[    4.668065] NVRM kgspInitRm_IMPL: cannot bootstrap riscv/gsp: 0x59
[    4.668067] NVRM RmInitAdapter: Cannot initialize GSP firmware RM
[    4.672462] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x63:0x59:1689)
[    4.675866] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[    4.675989] [drm:nv_drm_format_array_alloc [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to allocate NvKmsKapiDevice
[    4.676148] [drm:nv_drm_probe_devices [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to register device
```

There also is nothing in  `/dev/dri`. I can `rmmod` and `insmod/modprobe` the modules but it has no effect and nothing pertainent is posted to `journald`.

System is [Lenovo Legion 5 Pro-16ACH6H Laptop](https://www.lenovo.com/us/en/p/laptops/legion-laptops/legion-5-series/legion-5-pro-16ach6h/wmd00000468)
Operating System: Gentoo Linux 2.8
Kernel Version: 5.17.6-gentoo (64-bit)
Processors: 16 × AMD Ryzen 7 5800H with Radeon Graphics
Memory: 27.3 GiB of RAM

**To Reproduce**
Please write the steps to reproduce the behavior.

    ACCEPT_KEYWORDS=** CFLAGS="-march=native -O2 -pipe" CXXFLAGS="-march=native -O2 -pipe" USE="X driver kernel-open static-libs tools wayland -dist-kernel -persistenced" emerge  x11-drivers/nvidia-drivers`
    dracut -f
    reboot

**Expected behavior**
Please write a clear and concise description of what you expected to happen.

The driver should load without error on boot and the display manager should show.

**Please reproduce the problem, run nvidia-bug-report.sh, and attach the resulting nvidia-bug-report.log.gz.**

PAR2020, 2022-05-12:
Please run nvidia-bug-report.sh, and attach the resulting nvidia-bug-report.log.gz. Thank you.

Peter-Levine, 2022-05-12:
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8683124/nvidia-bug-report.log.gz)

PAR2020, 2022-05-13:
Thanks @Peter-Levine - will get it checked out.

PAR2020, 2022-05-13:
@Peter-Levine, please run it again but append NVreg_RmMsg=":" and post back the updated logs with some more info. Thank you.

Peter-Levine, 2022-05-14:
With `NVreg_RmMsg=":"`:

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8691879/nvidia-bug-report.log.gz)

LunNova, 2022-05-14:
Same issue with a laptop 3060 `01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)`.

Log with NVreg_RmMsg=":": https://gist.github.com/LunNova/ae36b3e90c7d4cbffbf2cb45f0b129bd

neophack, 2022-05-20:
With same problem

System is [Intel® NUC X15 Laptop Kit](https://www.intel.com/content/www/us/en/products/details/nuc/laptop-kits.html)
Operating System: Pop os! 20.04
Kernel Version: Linux pop-os 5.15.15-76051515-generic
GPU: RTX 3070 Laptop

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8736950/nvidia-bug-report.log.gz)

PAR2020, 2022-07-11:
Internal bug 3661421

FlorianFranzen, 2022-08-23:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Operating System and Version

NixOS Unstable 22.11.20220818.8ea014a (Raccoon)

### Kernel Release

5.18.18

### Hardware: GPU

NVIDIA Corporation GA104M [GeForce RTX 3070 Mobile / Max-Q] 

### Describe the bug

Here are the relevant entries in the dmesg log:
```
[    0.000000] Linux version 5.18.18 (nixbld@localhost) (gcc (GCC) 11.3.0, GNU ld (GNU Binutils) 2.38) #1-NixOS SMP PREEMPT_DYNAMIC Wed Aug 17 12:42:36 UTC 2022
[    0.000000] Command line: initrd=\efi\nixos\rl3vmwwkgjifw60m4an1lzsw66aipm7d-initrd-linux-5.18.18-initrd.efi init=/nix/store/d079d36197cscmg55imadpx35vc6ywiy-nixos-system-satoshi-22.11.20220820.b8eb2d4/init amdgpu.backlight=0 rfkill.default_state=1 loglevel=4 nvidia-drm.modeset=1 nvidia.NVreg_PreserveVideoMemoryAllocations=1 nvidia.NVreg_OpenRmEnableUnsupportedGpus=1
[    ....    ]
[   23.989750] nvidia: loading out-of-tree module taints kernel.
[   24.020428] nvidia-nvlink: Nvlink Core is being initialized, major device number 239
[   24.020433] NVRM cpuidInfoAMD: Unrecognized AMD processor in cpuidInfoAMD
[   24.021693] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   24.059459] intel_rapl_common: Found RAPL domain package
[   24.059463] intel_rapl_common: Found RAPL domain core
[   24.066898] NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  515.65.01  Release Build  (nixbld@)  Thu Aug 18 19:48:26 UTC 2022
[   24.073208] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[   24.073374] snd_hda_intel 0000:01:00.1: Disabling MSI
[   24.073382] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[   24.073502] snd_hda_intel 0000:05:00.6: enabling device (0000 -> 0002)
[   24.134480] input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input16
[   24.134529] input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input17
[   24.134575] input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input18
[   24.134610] input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input19
[   24.134642] input: HDA NVidia HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input20
[   24.134675] input: HDA NVidia HDMI/DP,pcm=11 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input21
[   24.158580] nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for x86_64  515.65.01  Release Build  (nixbld@)  Thu Aug 18 19:48:52 UTC 2022
[   24.212719] input: HDA Digital PCBeep as /devices/pci0000:00/0000:00:08.1/0000:05:00.6/sound/card1/input22
[   24.212767] input: HD-Audio Generic Mic as /devices/pci0000:00/0000:00:08.1/0000:05:00.6/sound/card1/input23
[   24.212801] input: HD-Audio Generic Headphone as /devices/pci0000:00/0000:00:08.1/0000:05:00.6/sound/card1/input24
[   24.215008] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[   25.766786] NVRM nvAssertOkFailedNoLog: Assertion failed: Generic operating system error [NV_ERR_OPERATING_SYSTEM] (0x00000059) returned from ((rpc_message_header_v *)pKernelGsp->pRpc->message_buffer)->rpc_result @ kernel_gsp.c:2081
[   25.766797] NVRM nvAssertOkFailedNoLog: Assertion failed: Generic operating system error [NV_ERR_OPERATING_SYSTEM] (0x00000059) returned from kgspWaitForRmInitDone(pGpu, pKernelGsp) @ kernel_gsp_ga102.c:229
[   25.766801] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ kernel_gsp_ga102.c:235
[   25.766804] NVRM kgspInitRm_IMPL: cannot bootstrap riscv/gsp: 0x59
[   25.766806] NVRM RmInitAdapter: Cannot initialize GSP firmware RM
[   25.768277] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x63:0x59:1689)
[   25.768793] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[   25.768933] nvidia-uvm: Loaded the UVM driver, major device number 236.
[   25.768942] [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to allocate NvKmsKapiDevice
[   25.770078] [drm:nv_drm_probe_devices [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to register device
[   25.774858] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0810 flags=0x0020]
[   25.775943] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0818 flags=0x0020]
[   25.777016] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0828 flags=0x0020]
[   25.778076] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0000 flags=0x0000]
[   25.779122] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0000 flags=0x0020]
[   25.780164] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0830 flags=0x0020]
[   25.781204] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0838 flags=0x0020]
[   25.782233] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0000 flags=0x0000]
[   25.783249] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcffe0000 flags=0x0020]
[   25.784301] nvidia 0000:01:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0xcfff18d8 flags=0x0020]
[   26.023990] NVRM s_executeBooterUcode_TU102: Booter failed with non-zero error code: 0x88
[... cut ...]
[   26.217362] NVRM nvAssertOkFailedNoLog: Assertion failed: Failure: Generic Error [NV_ERR_GENERIC] (0x0000FFFF) returned from kgspExecuteFwsecFrts_HAL(pGpu, pKernelGsp, pKernelGsp->pFwsecUcode, pKernelGsp->pWprMeta->frtsOffset) @ kernel_gsp_ga102.c:164
[   26.217371] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ kernel_gsp_ga102.c:235
[   26.217373] NVRM kgspInitRm_IMPL: cannot bootstrap riscv/gsp: 0xffff
[   26.217376] NVRM RmInitAdapter: Cannot initialize GSP firmware RM
[   26.218345] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x63:0xffff:1689)
[   26.218867] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[    ....    ]
```

mtijanic, 2022-08-23:
Hello @FlorianFranzen ! Since you're using NixOS, can you check if your gsp.bin is compressed? And if so, try uncompressing it?
Could be a dupe of #331

constroy, 2023-02-01:
### NVIDIA Open GPU Kernel Modules Version
525.85.05

```
options nvidia NVreg_OpenRmEnableUnsupportedGpus=1                                                                                                                                                                                             
options nvidia NVreg_EnableGpuFirmware=0
options nvidia NVreg_RmMsg=":"
```

### Operating System and Version
Arch Linux

### Kernel Release
6.1.8-arch1-1

### Hardware: GPU
NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q]

### Describe the bug
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/10559713/nvidia-bug-report.log.gz)

Here are the relevant entries in the dmesg log:

```
Feb 02 01:08:04 Host-24 kernel: NVRM kgspExecuteBooterUnloadIfNeeded_TU102: skipping executing Booter Unload as WPR is not up                                                                                                                  
Feb 02 01:08:04 Host-24 kernel: NVRM kgspInitRm_IMPL: skipping allocating Scrubber ucode as pre-scrubbed memory (0x10000000 bytes) is sufficient (0x7700000 bytes needed)
Feb 02 01:08:04 Host-24 kernel: NVRM kgspExecuteBooterLoad_TU102: executing Booter Load, gspFwWprMetaOffset 0x117c37000
Feb 02 01:08:04 Host-24 kernel: NVRM s_executeBooterUcode_TU102: before Booter mailbox0 0x00000000, mailbox1 0x00000000
Feb 02 01:08:04 Host-24 kernel: NVRM s_executeBooterUcode_TU102: starting Booter with mailbox0 0x17c37000, mailbox1 0x00000001
Feb 02 01:08:05 Host-24 kernel: NVRM s_executeBooterUcode_TU102: after Booter mailbox0 0x00000000, mailbox1 0x00000001
Feb 02 01:08:05 Host-24 kernel: NVRM kgspBootstrapRiscvOSEarly_GA102: GSP ucode loaded and RISCV started.
Feb 02 01:08:05 Host-24 kernel: NVRM kgspBootstrapRiscvOSEarly_GA102: Waiting for GSP fw RM to be ready...
Feb 02 01:08:05 Host-24 kernel: NVRM GspStatusQueueInit: Status queue linked to command queue.
Feb 02 01:08:06 Host-24 kernel: NVRM nvAssertOkFailedNoLog: Assertion failed: Generic operating system error [NV_ERR_OPERATING_SYSTEM] (0x00000059) returned from ((rpc_message_header_v *)pKernelGsp->pRpc->message_buffer)->rpc_result @ kern
el_gsp.c:2909
Feb 02 01:08:06 Host-24 kernel: NVRM nvAssertOkFailedNoLog: Assertion failed: Generic operating system error [NV_ERR_OPERATING_SYSTEM] (0x00000059) returned from kgspWaitForRmInitDone(pGpu, pKernelGsp) @ kernel_gsp_ga102.c:237
Feb 02 01:08:06 Host-24 kernel: NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ kernel_gsp_ga102.c:243
Feb 02 01:08:06 Host-24 kernel: NVRM kgspInitRm_IMPL: cannot bootstrap riscv/gsp: 0x59 
Feb 02 01:08:06 Host-24 kernel: NVRM RmInitAdapter: Cannot initialize GSP firmware RM
Feb 02 01:08:06 Host-24 kernel: NVRM kbusDestructVirtualBar2_VBAR2: MapCount: 0 Bar2 Hits: 0 Evictions: 0
Feb 02 01:08:06 Host-24 kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x62:0x59:1615)
Feb 02 01:08:06 Host-24 kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```

constroy, 2023-02-01:
I disabled the GSP firmware by `NVreg_EnableGpuFirmware=0`, but the open-source kernel driver still loaded it.

FlorianFranzen, 2023-05-07:
Seems to be fixed for me on 530.41.03

Peter-Levine, 2023-05-09:
530.41.03 is still not working for me with GeForce RTX 3070 Mobile.

Peter-Levine, 2023-11-09:
This seems to  be finally working for me after testing with 545.29.02.  Reopen if needed.

closed 2023-11-09 by Peter-Levine