NVIDIA/open-gpu-kernel-modules #124: Fix if-else statements in various files

Connor-GH, 2022-05-12:
**Is your feature request related to a problem? Please describe.**
Not a problem, but an enhancement. 
I see lines like this:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/1dbdaaef8f33dbc35da574850e17877c0e63fa63/src/common/modeset/hdmipacket/nvhdmipkt.c#L232
that use brackets for a singular function on the "if" statements.
(example: ``if(n) { printf("n\n"); }`` )
In reality, these brackets don't have to be there, since it's just one function for the conditional statement.

**Describe the solution you'd like**
Instead of 
```
if (libHandle == NVHDMIPKT_INVALID_HANDLE)
    {
        return NVHDMIPKT_LIBRARY_INIT_FAIL;
    }
```
use
```
if (libHandle == NVHDMIPKT_INVALID_HANDLE)
        return NVHDMIPKT_LIBRARY_INIT_FAIL;
```

**Describe alternatives you've considered**
Ternary operators can also be used for if-else statements. 

**Additional context**
This all falls under better code style, and makes no functional difference. 
In my eyes, it helps for readability.

TrickyDmitriy, 2022-05-12:
This is not an enhancement and do not follows the current code style. 
So issue should be closed in order to de-clutter "Issues" board ;)

Connor-GH, 2022-05-12:
Fair enough.

closed 2022-05-12 by Connor-GH

Zezombye, 2022-05-13:
https://www.zdnet.com/article/apples-goto-fail-tells-us-nothing-good-about-cupertinos-software-delivery-process/

Always use braces and if you don't, put the statement on the same line so you're not misled by indentation.