NVIDIA/open-gpu-kernel-modules #25: Can't use MIT or GPL licenses

vinniefalco, 2022-05-12:
Would it be possible to switch the MIT license to the Boost Software License? I can't incorporate any of this code because it requires attribution in binaries. See: https://pdimov.github.io/blog/2020/09/06/why-use-the-boost-license/

RealAstolfo, 2022-05-12:
why not just say you used nvidia code???

vinniefalco, 2022-05-12:
> why not just say you used nvidia code???

https://pdimov.github.io/blog/2020/09/06/why-use-the-boost-license/

Evernow, 2022-05-12:
> > why not just say you used nvidia code???
> 
> https://pdimov.github.io/blog/2020/09/06/why-use-the-boost-license/

Doesn't really answer the question of why you can't simply include the attribution.

KisaragiEffective, 2022-05-12:
At least, the title should be like "migrate to Boost Software License?" IMO.

airlied, 2022-05-12:
this code is for a Linux kernel module, it has to be GPL/MIT, why would you be reusing it somewhere else?

nicolas17, 2022-05-12:
> https://pdimov.github.io/blog/2020/09/06/why-use-the-boost-license/

Good thing this is not C++ nor a library then.

vinniefalco, 2022-05-12:
> Good thing this is not C++ nor a library then.

Yep, Peter's blog post is specified to C++ but the BSL is applicable to any software (and better than the MIT license). Going with the BSL allows it to be used in more contexts, so if the goal is to make the source code as widely available as possible (which, I am assuming it is) - then BSL is better.

vinniefalco, 2022-05-12:
> this code is for a Linux kernel module, it has to be GPL/MIT, why would you be reusing it somewhere else?

The BSL is more permissive. And the reason for nVIDIA to use the BSL is to use its prominence as a large company to promote the more permissive license, and raise awareness of it.

vinniefalco, 2022-05-12:
> That license isn't going to work for that.

That isn't quite right though. The Boost Software License is a drop-in replacement for any project which currently uses the MIT License. It is applicable in exactly the same scenarios that the MIT License is applicable. To put it in different terms, regardless of the specifics of the project - if it is currently using the MIT License, then it is better off using the Boost Software License. Switching costs nothing (I'm making a big assumption here that there are not already a bunch of contributors who submitted code under the MIT License, so apologies if that is not correct) and the potential benefits are greater than zero.

logan2611, 2022-05-12:
You could just attribute them?

vinniefalco, 2022-05-12:
Not all organizations can put attributions in binaries, for various reasons. Microsoft for example cannot use MIT licensed code as part of its standard library (because that would obligate anyone who compiles the standard library into their executable to show attribution in the binary).

logan2611, 2022-05-12:
> because that would obligate anyone who compiles the standard library into their executable to show attribution in the binary

And why is that an issue? Especially here, this is primarily intended for use in the Linux kernel which is GPL licensed.
Not to mention that Microsoft uses MIT licensed code and this has yet to be an issue for them.

closed 2022-05-12 by vinniefalco

Eugeny, 2022-05-13:
@vinniefalco these organizations need to get their shit together then.

cbeuw, 2022-05-13:
> Not all organizations can put attributions in binaries, for various reasons.

Such as? I'm genuinely curious as to when such circumstances arise. I can't think of any

hramrach, 2022-05-13:
That's an interesting question. 

Where do you put the required MIT attribution in a kernel module?
Do you print it to the kernel log?
For each module separately if multiple are loaded?
Or does it suffice to put it in the module description that modinfo would show?

The main argument against these clauses that require attribution in some specific form is that the attribution list grows ad absurdum over time as number of contributors grows and you are not legally allowed to move it somewhere where it is reasonably available to the user but does not get in the way.

Arguably requiring attribution in binaries may become unwieldy over time because there isn't always a convenient place to stuff really long list of attributions.

On the other hand, with carefully worded contributor agreement and tight control over what code gets into the project you may be able to avoid the attribution explosion problem.

Using a license that prevents it entirely might be preferable to some but it is more important for some projects than others.

cbeuw, 2022-05-13:
> requiring attribution in binaries

MIT does not require that, and almost no one is doing that, including copyright holders (authors). A copy of a software can contain more than one files. You just need to put the licence text in a separate file when you distribute it.

hramrach, 2022-05-13:
Is it OK to build a ramdisk that contains the nVidia module then, and put it on an installation medium?

That medium may not have the full copy of the binary distribution of the driver, mind you.

cbeuw, 2022-05-13:
You can place the licence text in the installation medium alongside the ramdisk image.

hramrach, 2022-05-13:
But then you have to go out of your way to examine the license of the modules needed for a particular system, find that it requires that the license text be placed on the medium, fish it from somewhere when building the medium, and place it there.

Not that it's not doable but avoiding such minefield would be preferable.

cbeuw, 2022-05-13:
> fish it from somewhere when building the medium, and place it there

That's the feature, not a bug! If the author licences something under MIT then you should assume they _want_ to be attributed. They want you to put their name in your medium.

hramrach, 2022-05-13:
There is a distinction between 'accompany it' as required by GPL for sources and 'included in all copies or substantial portions' as required for the MIT notices.

cbeuw, 2022-05-13:
You can argue over the wording of a licence for as long as you have money to pay lawyers. Nonetheless, the reality is that all of the richest tech companies employing the most expensive lawyers regularly use MIT-licenced dependencies without embedding the licence in their binary distributions, nor do they embed the licence in binaries of their own software. I really don't think anyone can win in court against you for having the licence placed alongside the binary.

hramrach, 2022-05-13:
Where are the court cases or at least legal analyses supporting your claim?

vimpostor, 2022-05-13:
> Microsoft for example cannot use MIT licensed code as part of its standard library (because that would obligate anyone who compiles the standard library into their executable to show attribution in the binary).

That's valid for standard libraries and I would maybe even consider this argument valid for libc implementations.
But this is not a valid argument for this project, I can't imagine any scenario where it would be moral to use Nvidia code and not attribute anyone. **The Nvidia driver is not a STL**, so your point does not apply here (noone will compile the Nvidia driver into their own binary).

Also your blog post is an absolute shit tier take, the only morally correct thing would be to license everything with GPL /s :clown_face:

aaronp24, 2022-05-13:
Thank you for your interest in this driver. The license will remain as stated in [COPYING](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/COPYING) for now.