NVIDIA/open-gpu-kernel-modules #27: User Linter and CI

6543, 2022-05-12:
It would be a **good practice** to enforce linter to detect/avoid things like pointed out in #24

6543, 2022-05-12:
#31 or #108 ?

mmaneetsingh, 2022-05-13:
Consolidating comment from @ghtesting2020 (here):
I'm seeing dozens of tickets that could be avoided if Nvidia used a linter in their own internal dev environment. GitHub Actions could also be implemented to autocheck code for compliance, security issues, quality, etc. At my work we use cppcheck for static code analysis and it's been amazing so far. SonarQube/Lint is also an option as well as being open source/free.

sTiKyt, 2022-05-13:
> Consolidating comment from @ghtesting2020 (here): I'm seeing dozens of tickets that could be avoided if Nvidia used a linter in their own internal dev environment. GitHub Actions could also be [...]

You may have forgotten to add link to "(here)" section? I may be getting it wrong

mmaneetsingh, 2022-05-13:
updated, thanks for pointing out.