NVIDIA/open-gpu-kernel-modules #33: Add dependencies to README

blueOkiris, 2022-05-12:
I was trying to build on Arch, and I ran into a weird issue with the Makefile not finding the command "hostname"

Turns out I needed the "inetutils" package (I think other distros put it in bind-utils).

It would be helpful if we had some idea of what the build dependencies were beyond gcc

Doomsdayrs, 2022-05-12:
The driver compiles fine with Fedora for me. It might be just your edge case?

rmnscnce, 2022-05-12:
It's just the regular kernel build dependencies. Anyone willing to hack around this driver should already know that, I think.

But a dependency list won't hurt.

PAR2020, 2022-07-26:
Thanks @blueOkiris for the issue. Per aritger in PR #28, we will not be implementing dependency lists. Closing this issue as WNF.

closed 2022-07-26 by PAR2020