NVIDIA/open-gpu-kernel-modules #35: Depends on proprietary firmware binaries

throwaway1037, 2022-05-12:
Please may you also release the firmware source code under a free software license?
(Those at https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/nvidia)
The driver requires them to work, and without them core features like reclocking will not function, which hindered Nouveau for years.

OS distributions such as [Parabola GNU+Linux-libre](https://parabola.nu) use Linux-libre, a version of Linux without blobs, and do not distribute blobs.
In essence, without the firmware also being free software, this awesome free software driver release is crippled in the free world, as it relies on proprietary binaries.

PAR2020, 2022-05-12:
Thank you for your comments, @throwaway1037. NVIDIA has just begun our open source journey, so please be patient.
For now, we are not planning to upload GSP firmware at linux-firmware but this may change in the future.
Today, the firmware, kernel driver, and user-mode drivers must match versions within a release.
Note that firmware interfaces are not backward or forward compatible.

throwaway1037, 2022-05-12:
We've been waiting for over a decade for our freedom, although I thank NVIDIA for making this positive step. :)

msva, 2022-05-12:
> for over a decade 
@throwaway1037  Two, actually ;)

> Today, the firmware, kernel driver, and user-mode drivers must match versions within a release.
@PAR2020 yup, and this is pretty annoying thing that prevents it to package it like any other ordinary opensource package in many distros (well, it is still possible, but it makes mainainer's work harder).

And official instruction still recommends to use `blablabla.run` "installer" (the script with bundled archives), which is "non-recommended" way of installed anything in linux distros, and installing anything in that way (not with package manager) makes that system "not supported" (as package manager doesn't know anything about that files and can face a collision someday).

So, TL;DR: it is super good thing, that NVidia finally released kernel-side sources, and we're all congratulate and thank NVidia. But at the same time we're hoping NVidia will move further in "opensourcing", and will make it's drivers "compatible" with "standard" way of linux distro mainteance.

breisig, 2022-05-12:
Thank you Nvidia for moving forward with a true open source kernel module. This was needed for a LONG time. GSP firmware should be included in linux-firmware and be flexible to use the same GSP firmware for multiple driver versions.

AaronWebster, 2022-05-12:
I'm really pleased to see this as well and going forward it will only make your product better.

Don't worry, you'll still provoke the ire of Linus, just for different reasons :-)

PAR2020, 2022-05-12:
Thank you all for your support, and patience. We have many more issues to work through in the short term, so will close this one for now.

closed 2022-05-12 by PAR2020