NVIDIA/open-gpu-kernel-modules #38: SSL error while signing modules

blueOkiris, 2022-05-12:
Currently using 510 on Arch w/ Linux kernel 5.17.6-arch1-1 on an RTX 3080m. Trying to install 515 from this repo.

I'm running into an issue when running `sudo make modules_install`

```
make -C kernel-open modules_install
make[1]: Entering directory '<where I downloaded the repo>/open-gpu-kernel-modules/kernel-open'
make[2]: Entering directory '/usr/lib/modules/5.17.6-arch1-1/build'
  INSTALL /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-drm.ko
  INSTALL /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-modeset.ko
  INSTALL /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-peermem.ko
  INSTALL /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-uvm.ko
  INSTALL /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia.ko
  SIGN    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-peermem.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
  SIGN    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-modeset.ko
  SIGN    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-drm.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
  ZSTD    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-peermem.ko.zst
  ZSTD    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-modeset.ko.zst
  ZSTD    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-drm.ko.zst
  SIGN    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
  ZSTD    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia.ko.zst
  SIGN    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-uvm.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
  ZSTD    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia-uvm.ko.zst
  DEPMOD  /lib/modules/5.17.6-arch1-1
make[2]: Leaving directory '/usr/lib/modules/5.17.6-arch1-1/build'
make[1]: Leaving directory '<the download directory>/open-gpu-kernel-modules/kernel-open'
```
It appears to be an issue with "signing:"

```
  SIGN    /lib/modules/5.17.6-arch1-1/kernel/drivers/video/nvidia.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: ./certs/signing_key.pem: No such file or directory
```

joshniec, 2022-05-12:
You are missing a signing key: `sign-file: ./certs/signing_key.pem: No such file or directory`

https://superuser.com/questions/1214116/no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules/1322832#1322832

aritger, 2022-05-12:
Thanks, @blueOkiris and @joshniec.  I don't believe this is an issue with the open-gpu-kernel-modules, but rather your signing configuration.  If you disagree, please let me know.  But, for now I'm going to close this as not a bug in open-gpu-kernel-modules.

closed 2022-05-12 by aritger