NVIDIA/open-gpu-kernel-modules #39: Also support T234 in the same driver

woachk, 2022-05-12:
**NVIDIA Driver Version**
34.1 and 515.43.04
**GPU**
AGX Orin and RTX 2080 Ti

**Describe the bug**
AGX Orin already ships with an open-source NVIDIA kernel driver for the display. This means that when we install this driver, we lose the display output tied to the display controller already present on the SoC.

**To Reproduce**

On an AGX Orin:

`$ modinfo 5.10.65-tegra/extra/opensrc-disp/nvidia.ko
filename:       /lib/modules/5.10.65-tegra/extra/opensrc-disp/nvidia.ko
firmware:       nvidia/34.1.0/gsp.bin
alias:          char-major-195-*
version:        34.1.0
supported:      external
license:        Dual MIT/GPL
srcversion:     92576267C511B79A495FD51
alias:          pci:v000010DEd*sv*sd*bc06sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
alias:          of:N*T*Cnvidia,tegra234-displayC*
alias:          of:N*T*Cnvidia,tegra234-display
`

This conflicts with this driver, which doesn't have T234D support included. (and the driver in L4T doesn't have dGPU support included!)

**Expected behavior**

This driver should support T234D too (nvidia,tegra234-display). Having two incompatible open-source drivers this early is quite unusual.

PAR2020, 2022-05-12:
Good suggestion.
We do plan to do this in the future, however, we haven't finalized when this would be done. T234D support was added at a time when a major part of this driver (today's release) was not ready, hence we did a lot of pre-processing - some of which needs to be cleaned up before we can have a single driver that supports both big GPUs and T234D.

mmalyska, 2025-02-26:
Any update on this?

aritger, 2025-02-26:
Sorry, no current update.  But this is still something we hope to do eventually.