NVIDIA/open-gpu-kernel-modules #40: Consider supporting older GPUs

nonetrix, 2022-05-12:
It's great part of the drivers are now open source however, this doesn't effect many uses since this only applies to Ampere GPUs and higher. 

Many people still use older GPUs me included and many others judging off the Steam hardware survey:
https://store.steampowered.com/hwsurvey/Steam-Hardware-Software-Survey-Welcome-to-Steam

I understand this might take a lot of effort but it would really be great if this was included thanks ☺️

nonetrix, 2022-05-12:
However it might still effect older GPUs indirectly (just guessing I don't know much about GPUs) if they are similar enough that they code could be hacked together to work on older GPUs? Am I correct in assuming this?

wyatt8740, 2022-05-12:
@nonetrix it's possible _something_ of value could come out of this for, say, the Nouveau project. But they'd still need to get binary firmware signed for everything Maxwell and newer except for the GTX 750/750 Ti Maxwell cards.

As for repurposing this code for older GPU's, that seems less likely but I have yet to deep dive into these sources.

woachk, 2022-05-12:
@wyatt8740 

For code for older GPUs you might want to look at this codebase: https://nv-tegra.nvidia.com/r/gitweb?p=linux-nvgpu.git;a=summary which _does_ support some older GPUs. 

That said, it doesn't come with firmware for it either, so that issue still remains.

(nvgpu is the GPU driver used for Tegra today, across Linux, QNX and Nintendo Horizon)

wyatt8740, 2022-05-12:
Interesting.

I know about tegra, but did not know about that repository. How new is that?

woachk, 2022-05-12:
> Interesting.
> 
> I know about tegra, but did not know about that repository. How new is that?
> 
> Anyway, it seems like that might be _somewhat_ useful, still not a complete solution, though.

All the Tegra since the K1 use it to drive the GPU block, so since quite a while now. And yep, not a complete solution, used for Tegra iGPUs almost exclusively by the public*... and GPLv2 only instead of being dual-licensed

PAR2020, 2022-05-12:
Thanks for your feedback.
Today's release supports data center GPUs for Turing and Ampere architectures. There are hardware limitations in our older GPUs which negates support of open kernel modules for pre-Turing architectures. Please refer to aritger's comments at https://github.com/NVIDIA/open-gpu-kernel-modules/issues/19 for his detailed explanation.
Also see further responses regarding Tegra devices in https://github.com/NVIDIA/open-gpu-kernel-modules/issues/39

closed 2022-05-12 by PAR2020

wyatt8740, 2022-05-12:
By "hardware limitations" do we actually mean "licensing/legal limitations?"

aaronp24, 2022-05-12:
> By "hardware limitations" do we actually mean "licensing/legal limitations?" Because I don't know any other reasons from an engineering standpoint that would preclude sharing source code.

It's because this driver uses the new [GPU System Processor](https://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/gsp.html) which is not present on GPUs older than Turing. Please refer to [the blog post](https://developer.nvidia.com/blog/nvidia-releases-open-source-gpu-kernel-modules/) from today for more info.

StayBlueee, 2022-05-12:
> It's because this driver uses the new [GPU System Processor](https://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/gsp.html) which is not present on GPUs older than Turing. Please refer [...]

I think the main question people have is if there will ever be a way provided by Nvidia to allow for reclocking support on Maxwell and Pascal without the need to use the GPU System Processor.

Daasin, 2022-05-12:
That's a shame for GTX10 Series cards on the Pascal architecture, open-source resources could've helped with a lot of issues like on DXVK/VKD3D

wyatt8740, 2022-05-13:
> It's because this driver uses the new [GPU System Processor](https://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/gsp.html) which is not present on GPUs older than Turing. Please refer [...]

I understood that; That wasn't my question. My question is if there's any intention to make the non-GSP driver open source, or even release documentation/provide help with authoring a free (as in 'freedom') one for non-GSP GPU's.

ptr1337, 2022-05-13:
Actually I did from the current nvidia proprietary driver (dkms) and that is what I not understand. It uses the dma buf, yes.

Here if some is interested in. 

https://www.toptal.com/developers/hastebin/ralegehixi.diff