NVIDIA/open-gpu-kernel-modules #41: NVIDIA I2C driver issues

CalcProgrammer1, 2022-05-12:
Copy/paste from an email I sent to linux-bugs@nvidia.com, I haven't verified this against this new open driver, but it has been an issue for a while with the proprietary kernel driver.

I am the lead developer of OpenRGB, an open source RGB lighting control application for Windows and Linux.  Our goal is to create a universal RGB control app, talking directly to as many RGB lighting devices as possible.  As RGB control is often only supported by official software in Windows, Linux users get left out.  That's where we come in.

As you are probably aware, a lot of graphics cards have built-in RGB lighting that features software control.  Most cards implement RGB control using the GPU's I2C interface.  We're facing an issue controlling certain RGB devices over the NVIDIA GPU's I2C interface in Linux with the proprietary NVIDIA driver.  The same code is working fine using NvAPI on Windows and using the Nouveau I2C implementation on Linux, so we believe this to be an issue specific to NVIDIA's proprietary Linux driver.

The cards we've been focused on lately are the ASUS 3xxx series cards, which all use a similar I2C RGB chip that is also found on some ASUS motherboards and various manufacturers' RGB DRAM modules.  The chip comes from ENE.  The I2C protocol used by this chip is a 16-bit address scheme where you first write a 16-bit address to the 0x00 register of the ENE chip, then perform either a read or write operation to a fixed chip register.  This chip appears to be an SMBus chip, so we're using the SMBus functions in the Linux kernel as shown in these two accessor functions:

(ene_dev_id is the 8-bit I2C address of the ENE chip, ene_register is the 16-bit address in the chip that we are reading or writing)

```
unsigned char ENESMBusInterface_i2c_smbus::ENERegisterRead(ene_dev_id dev, ene_register reg)
{
    //Write ENE register
    bus->i2c_smbus_write_word_data(dev, 0x00, ((reg << 8) & 0xFF00) | ((reg >> 8) & 0x00FF));

    //Read ENE value
    return(bus->i2c_smbus_read_byte_data(dev, 0x81));
}

void ENESMBusInterface_i2c_smbus::ENERegisterWrite(ene_dev_id dev, ene_register reg, unsigned char val)
{
    //Write ENE register
    bus->i2c_smbus_write_word_data(dev, 0x00, ((reg << 8) & 0xFF00) | ((reg >> 8) & 0x00FF));

    //Write ENE value
    bus->i2c_smbus_write_byte_data(dev, 0x01, val);
}
```

The issue here appears to be with regards to i2c_smbus_write_word_data.  To detect and verify that the chip exists on the bus, we do a series of i2c_smbus_read_byte_data calls and these all work fine on the NVIDIA proprietary Linux driver.  However, after detecting the chip we then write a 16-bit address and attempt to read from it.  Specifically, we try to read a region of the ENE chip's memory known to contain a version string.  The expectation is that the word data write puts the 16-bit address of the byte we want to read into the chip, and the following byte data read from 0x81 returns one byte from the 16-bit address.

With other SMBus host controllers (Intel and AMD chipsets) as well as the NVIDIA GPU I2C on both Windows (NvAPI) and Linux (Nouveau), this works fine and we successfully retrieve the ENE RGB controller's version string.  With the NVIDIA proprietary Linux driver, we read garbage.  Since we know the i2c_smbus_read_byte_data function works with other manufacturers' NVIDIA GPU boards and for detecting the chip, I can only assume the issue is that the i2c_smbus_write_word_data function isn't working correctly.  Note that i2c_smbus_write_byte_data does appear to work on several other manufacturers' GPU RGB chips so I have to assume it's specific to word data.

We have also observed issues with SMBus block operations, though doing the same block operations using I2C_RDWR ioctl (thus avoiding the SMBus layer) seem to work on the NVIDIA proprietary driver.

wyatt8740, 2022-05-12:
The only way I have ever been able to get I²C or DDC working on my nvidia card has been to put `options nvidia NVreg_RegistryDwords=RMUseSwI2c=0x01;RMI2cSpeed=100` in `/etc/modprobe.d/nvidia.conf`, or else set up an xorg.conf containing something like:

````
Section "Device"
        Identifier      "Device0"
        Driver          "nvidia"
        VendorName      "NVIDIA Corporation"
        BoardName       "GeForce GTX 750 Ti"
        Option          "RegistryDwords"        "RMUseSwI2c=0x01; RMI2cSpeed=100"
EndSection
````
I prefer the first method.

CalcProgrammer1, 2022-05-12:
The lighting controller is usually on an unused I2C bus (one not connected to a display output for DDC).  I can communicate with the controllers on many cards just fine, so I don't think it has to do with I2C speed.  It seems to be related to the SMBus subset of the I2C functionality in that word and block transfers do not occur correctly (but byte transfers do).  I don't personally have an affected card unfortunately, it's an issue that I had others who owned an appropriate card test.  The ASUS RTX3xxx series are the most affected as the lighting controller they use requires 16-bit (word) SMBus writes.

atiensivu, 2022-05-12:
This explains what I saw when I was fiddling with my ASUS 3090. Interesting.

PAR2020, 2022-05-12:
Hello gentlemen,
To get the best focus for our open source effort, we would like to keep the focus of these GitHub issues to the Open GPU Kernel Modules components. As you may know, we do have a Linux Forum (https://forums.developer.nvidia.com/c/gpu-graphics/145) where more generic issues should be discussed. Submitting the issue through linux-bugs@nvidia.com get directly to the team of developer, QA, and triage folks who are able to resolve them best.
So respectfully, we will close this issue and direct you to continue using the forum for these types of problems/discussions.

CalcProgrammer1, 2022-05-12:
I2C is part of the kernel components...

https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/kernel-open/nvidia/nv-i2c.c

CalcProgrammer1, 2022-05-12:
I actually think removing the block of code at line 191 here would do the trick, as without the I2C_SMBUS capabilities provided by NVIDIA's driver, the kernel will fall back to using the non-SMBus I2C calls through emulation, using an interface we have confirmed working (raw I2C block transfers).

https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/kernel-open/nvidia/nv-i2c.c#L191

The issue is why do I2C SMBUS word operations fail when going through NVIDIA's implementation?

ZetaPhoenix, 2022-05-12:
EVGA 30xx cards use `I2C_SMBUS_I2C_BLOCK_DATA` commands to talk to the RGB controller and this is not supported under Linux but is supported through nvapi.dll

See this patch that works under Windows for OpenRGB to talk to EVGA cards through NVIDIA's .dll but will fail under Linux: https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/752

Support for `I2C_SMBUS_I2C_BLOCK_DATA` commands is also missing in this code.

CalcProgrammer1, 2022-05-12:
Also look into https://github.com/NVIDIA/open-gpu-kernel-modules/blob/69791e9bf5161c786d620312fd6396756c37cff9/kernel-open/nvidia/nv-i2c.c#L58

Comment indicates "we only support basic I2C reads/writes, reject any other commands"

aritger, 2022-05-12:
Hi @CalcProgrammer1 I'm sorry you didn't receive a response to your linux-bugs email.  I agree this merits investigation.  Reopening.

ValZapod, 2022-05-12:
Can I clarify? We can finaly get RGB colors instead of only green for my founders edition 2080 Ti? https://youtu.be/bcKQ3YFnYM4

That always made me so angry! 😋

amrit1711, 2022-05-16:
We have filed a bug 3646481 internally for tracking purpose. You can refer it for status update.

aritger, 2022-05-23:
@CalcProgrammer1: can you test whether the following change fixes the problem for you?

```
$ git diff
diff --git a/kernel-open/nvidia/nv-i2c.c b/kernel-open/nvidia/nv-i2c.c
index f227d180e149..8bc6cf6314e7 100644
--- a/kernel-open/nvidia/nv-i2c.c
+++ b/kernel-open/nvidia/nv-i2c.c
@@ -140,8 +140,9 @@ static int nv_i2c_algo_smbus_xfer(
         case I2C_SMBUS_WORD_DATA:
             if (read_write != I2C_SMBUS_READ)
             {
-                data->block[1] = (data->word & 0xff);
-                data->block[2] = (data->word >> 8);
+                u16 word = data->word;
+                data->block[1] = (word & 0xff);
+                data->block[2] = (word >> 8);
             }
 
             rmStatus = rm_i2c_transfer(sp, nv, (void *)adapter,
```
(the same change should apply to both the open kernel modules here and to the proprietary driver)

amrit1711, 2022-05-30:
@CalcProgrammer1 
Can you please help to test the change fixe as suggested in last comment and share results.

CalcProgrammer1, 2022-05-30:
Unfortunately, as I said earlier in this issue thread, I personally do not have a compatible GPU to test this on.  I'm working based on feedback from other users on the OpenRGB Discord server.  I asked a week or so ago if anyone was available to test, but I don't know if the users who had the issue are familiar enough with Linux to patch the modules.

I can ask again.  If anyone here has an ASUS RTX 3 series card with Aura lighting, it would be awesome if you could test the above patch.  All you would need to do is launch the latest OpenRGB (version 0.7 or newer) with the patched module loaded and then check the version string in the Information tab.  If it is garbage (non-ASCII characters) then the patch did not work.

rockowitz, 2022-06-01:
Like @CalcProgrammer1, I too lack a device to test this on.  Because of the difficulties presented by the proprietary driver, I have not purchased a Nvidia based card in years. Also, **ddcutil** does not use I2C SMBUS ioctl's, and tries to avoid even probing a /dev/i2c device that appears to be a SMBUS device, both for efficiency and because of the potential for lockups.

devilsclaw, 2022-06-05:
So I went and looked at the patch.

I manually applied the patch to /usr/src/nvidia-510.73.05/nvidia/nv-i2c.c
which is the current install proprietary nvidia driver.

after that I did:  
`cd /usr/src/nvidia-510.73.05`  
`sudo make modules -j24`  
`sudo make modules_install`  
`sudo update-initramfs -c -k $(uname -r)`  

I originally added a printk("DC: Testing\n"); to the code as well so that I could do a dmesg and grep that string. I was able to see that string so it shows that the code and the patches took. after that I removed the print statement and did the same thing as the above.

I ran the patched version of openrgb and I also ran the one I patched which is really no difference. Sadly the patch did not fix it for my video card.

Below is what that section of the code currently looks like.  
`        case I2C_SMBUS_WORD_DATA:`  
`            if (read_write != I2C_SMBUS_READ)`  
`            {`  
`                //data->block[1] = (data->word & 0xff);`  
`                //data->block[2] = (data->word >> 8);`  
`                u16 word = data->word;`  
`                //printk("DC: Testing\n");`  
`                data->block[1] = (word & 0xff);`  
`                data->block[2] = (word >> 8);`  
`            }`

NOTE: If need be I can switch to the open-gpu-kernel-modules version of the nvidia driver. It might take me a bit to get it running on ubuntu but I am sure I can get it done.

Chr1sNo, 2022-06-05:
Further to @devilsclaw's comments it's worth noting that the same GPU (10DE:220A 1043:886E) has been [confirmed as working](https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/1243) on Windows.

devilsclaw, 2022-06-05:
This is a dump of what happens when I run openrgb and do nothing else

Note the first line as there before I open the program but I think it may be relevant to why it is not working for me, not sure.
[   11.651031] [drm:nv_drm_master_set [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000900] Failed to grab modeset ownership  
[  107.463050] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.464127] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.465661] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.466819] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.468771] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.470270] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.471682] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.472890] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.474193] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.475468] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.476780] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.478218] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.479667] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.480993] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.483073] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.484369] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.485706] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.486826] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.487922] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.489004] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.490108] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.491191] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.492277] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.493373] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.494471] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.495555] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.496632] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.497715] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.499610] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.501203] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.502936] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.504309] DC: I2C_WRITE: addr 0x67 : WORD 0x1010 : 0x10,0x10  
[  107.505838] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.507267] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.508624] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.509991] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.511298] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.512651] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.513793] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.515523] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.516844] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.518049] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.519515] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.520608] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.521694] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.522790] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.523877] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.524962] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.526081] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.527171] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.528665] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.529752] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.530834] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.532796] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.534383] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.535923] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.537631] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.539035] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.540467] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.541757] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.543127] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.544471] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.545808] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.546909] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.547998] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.549814] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[... 9 lines cut ...]
[  107.563626] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.564804] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.566459] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.567918] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.569945] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.571244] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.572576] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.574101] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.575441] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[... cut ...]
[  107.588751] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.589909] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.591011] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.592091] DC: I2C_WRITE: addr 0x67 : WORD 0x1C1C : 0x1C,0x1C  
[  107.593240] DC: I2C_WRITE: addr 0x67 : WORD 0x8080 : 0x80,0x80  
[  107.594333] DC: I2C_WRITE: addr 0x67 : WORD 0x8080 : 0x80,0x80  
[  107.595423] DC: I2C_WRITE: addr 0x67 : WORD 0x8080 : 0x80,0x80  
[  107.596507] DC: I2C_WRITE: addr 0x67 : WORD 0x8080 : 0x80,0x80

devilsclaw, 2022-06-05:
Here is a dump:

[   57.761987] DC: I2C_WRITE_SMBUS_BYTE: addr 0x67 : BYTE 0x00  
[   57.762610] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA0 : BYTE 0x00  
[   57.763227] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA1 : BYTE 0x01  
[   57.763865] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA2 : BYTE 0x02  
[   57.764480] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA3 : BYTE 0x03  
[   57.765101] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA4 : BYTE 0x04  
[   57.765715] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA5 : BYTE 0x05  
[   57.766331] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA6 : BYTE 0x06  
[   57.766948] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA7 : BYTE 0x07  
[   57.767560] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA8 : BYTE 0x08  
[   57.768183] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xA9 : BYTE 0x09  
[   57.768791] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xAA : BYTE 0x0A  
[   57.769416] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xAB : BYTE 0x0B  
[   57.770020] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xAC : BYTE 0x0C  
[   57.770635] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xAD : BYTE 0x0D  
[   57.771250] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xAE : BYTE 0x0E  
[   57.771860] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0xAF : BYTE 0x0F  
[   57.771871] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.772953] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x8C  
[   57.772962] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.774040] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x80  
[   57.774048] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.775773] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0xC5  
[   57.775783] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.777650] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x80  
[   57.777660] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.779338] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x8C  
[   57.779345] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.780814] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x80  
[   57.780821] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.782100] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.782108] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.783491] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.783498] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.784832] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.784838] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.787136] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.787144] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.788692] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.788699] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.790054] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.790062] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.791563] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.791570] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.792716] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.792723] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.793788] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.793797] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.794921] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.794929] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.796019] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.796027] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.797103] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.797110] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.798222] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.798229] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.799417] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.799424] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.800505] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.800511] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[   57.801599] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0x23  
[   57.801607] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x1010 : BYTES 0x10,0x10  
[... cut ...]
[   57.911291] DC: I2C_WRITE_SMBUS_WORD_DATA: addr 0x67 : CMD 0x00 : WORD 0x8080 : BYTES 0x80,0x80  
[   57.912571] DC: I2C_READ_SMBUS_BYTE_DATA: addr 0x67 : CMD 0x81 : BYTE 0xAA  
[   57.932933] DC: I2C_WRITE_SMBUS_BYTE: addr 0x67 : BYTE 0x00  
[   57.957259] DC: I2C_WRITE_SMBUS_BYTE: addr 0x67 : BYTE 0x00  
[   57.977866] DC: I2C_WRITE_SMBUS_BYTE: addr 0x67 : BYTE 0x00  
[   57.978786] DC: I2C_WRITE_SMBUS_BYTE: addr 0x67 : BYTE 0x00  
[   57.979330] DC: I2C_WRITE_SMBUS_BYTE: addr 0x67 : BYTE 0x00

emko, 2022-07-01:
just moved to Linux and now can't control leds, if you need someone to test i am willing

ZetaPhoenix, 2022-07-02:
Seems like i2ctransfer commands work to talk to the EVGA 30 series GPUs:

```
# Initialize LED control
i2ctransfer -y 5 w6@0x2d 0xb2 0x04 0xc6 0xeb 0xea 0x15
# Command control for ALL OFF
i2ctransfer -y 5 w11@0x2d 0xc0 0x09 0x00 0x00 0x00 0x00 0xff 0xff 0xff 0xff 0x00
```

Maps to this command here: https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/EVGAAmpereGPUController/EVGAGPUv3Controller.cpp#L297
```c++
void EVGAGPUv3Controller::initCard()
{
    // This command needs to be sent before the card will respond to OpenRGB commands
    // NvAPI_I2CWriteEx: Dev: 0x2D RegSize: 0x01 Reg: 0xB2 Size: 0x05 Data: 0x04 0xC6 0xEB 0xEA 0x15
    uint8_t data_pkt[5] = {0x04, 0xC6, 0xEB, 0xEA, 0x15};
    bus->i2c_smbus_write_i2c_block_data(dev, EVGA_GPU_V3_REG_ENABLE, sizeof(data_pkt), data_pkt);
    LOG_TRACE("[%s] Sending SW int packet", evgaGPUName);
    return;
}
```

Should `i2c_smbus_write_i2c_block_data` be mapped to the i2ctransfer command or is there another fix to allow block commands to be used?

gardotd426, 2022-07-04:
> Seems like i2ctransfer commands work to talk to the EVGA 30 series GPUs:
>
> ```
> # Initialize LED control
> i2ctransfer -y 5 w6@0x2d 0xb2 0x04 0xc6 0xeb 0xea 0x15
> # Command control for ALL OFF [...]

@ZetaPhoenix I have an EVGA XC3 Ultra RTX 3090, am I correct in my reading of your comment  being that following your steps should allow lighting control of this card on Linux? I want to confirm this is what you're saying before I actually test it. Yes, I'm aware that this is all potentially risky and if I run anything it's at my own risk, but that's exactly why I want to confirm that's what you're saying beforehand. 

As others have said, even though it seems like it might be a different underlying issue, EVGA Ampere GPUs seem to also be controllable on Windows via OpenRGB but not on Linux, just like the ASUS cards. 

And again, I do have hardware myself I can test this on, and I believe I can try patching the modules without much trouble (though I do use DKMS, I don't imagine that should cause any issues?)

ZetaPhoenix, 2022-07-04:
Yes, according to other users on the OpenRGB discord, these commands work to turn off the RGB. The settings will not persist as the save command is not sent. See the rest of the OpenRGB controller code for the command if you want to save.

OpenRGB will attempt to load the firmware version from the card and if that fails (due to `i2c_smbus_read_i2c_block_data` not returning successfully) then the card will not be added.

gardotd426, 2022-07-04:
> Yes, according to other users on the OpenRGB discord, these commands work to turn off the RGB. The settings will not persist as the save command is not sent. See the rest of the OpenRGB controller [...]

Ah, so so far it's only possible to turn the RGB off?

gardotd426, 2022-07-04:
@ZetaPhoenix No dice, unfortunately:

```
sudo i2ctransfer -y 5 w6@0x2d 0xb2 0x04 0xc6 0xeb 0xea 0x15  
Error: Adapter does not have I2C transfers capability
```

Until a couple weeks ago I had a single-GPU passthrough Windows VM that I used for Apex Legends, and through that using EVGA Precision X1 I was obviously able to control lighting. I'm not sure what more info I can provide but I'm happy to provide anything else needed to help get this figured out

gardotd426, 2022-07-04:
Eh even with patching the nv-i2c.c file and rebuilding and reinstalling the module it still gives the same results. I'm able to test whatever needs testing but I'm swamped right now trying to finish all the work required for the Arch port of Regolith DE moving to v2.0 (I'm the sole - and apparently now official - developer/maintainer of the Regolith DE port to Arch Linux, and v1.6 -> v2.0 basically changed everything from the ground up on a fundamental level regarding how it works, so I basically have to redo the whole thing). So I don't know how much time I'll have for deep dives in the Discord, but as soon as I finish the 2.0 transition I'll have more time. In the meantime I'm open to any guidance

amrit1711, 2022-07-22:
@CalcProgrammer1 
I tried to repro issue locally with ASUS Lightning 3050 card on below configuration setup but could not repro issue.

Alienware-Area-51-R4 + Intel(R) Core(TM) i7-7900X CPU @ 3.30GHz + kernel 5.15.0-41-generic + RTX 3050 + Driver 515.43.04 + Display GBT AORUS FI27Q-P

Repro Steps Tried - 
1) Executed openrgb command and got the warning saying One or More I2C/SMBus interfaces failed to initialize.
2) Under Information Tab, I can see version as 0.7.
However as per your earlier comment, version should be a garbage value in case of repro scenario but I can see correct version. --- NO REPRO

Request you or someone else to please share reliable test case along with steps.

ZetaPhoenix, 2022-07-22:
Please make sure to load the udev rules and try with the pipeline version. Adding -vv will print more data to the console and launching with --loglevel 6 will create a verbose log file in ~/.config/OpenRGB/logs

FYI, that card does seem to have been added to the controller so it will skip: https://gitlab.com/OpenRGBDevelopers/OpenRGB-Wiki/-/blob/stable/Supported%20Devices/Supported%20Devices.md

Does it have RGB?

PAR2020, 2022-07-26:
Looks like we finally have an internal repro of this issue.
As stated, using Nouveau works fine (proper version reported, LEDs can be changed).
Using the 515.57 driver (monolithic or OpenRM) with OpenRGB 0.7 shows the card is detected but the LEDs cannot be changed, version field is garbage.
System being used for dev triage.

ZetaPhoenix, 2022-07-26:
Good to hear.

Are there any patches to test for the block writes (EVGA GPU's)? That seems possible in some instances with i2ctransfer.

PAR2020, 2022-07-26:
Not yet. The current patch reported here does not work in the repro case. Will advise if a new one is proposed.

aritger, 2022-07-30:
I think this patch should resolve the issue:
[0001-fix-I2C_SMBUS_WORD_DATA-reads-and-writes.patch.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9224065/0001-fix-I2C_SMBUS_WORD_DATA-reads-and-writes.patch.txt)
I'd appreciate if anyone here could confirm that this fixes it for them.

aritger, 2022-08-01:
Sorry, the patch I mentioned above should also contain the fix mentioned earlier in this thread.  Here is an updated patch that contains both: 
[0001-fix-I2C_SMBUS_WORD_DATA-reads-and-writes.patch.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9237804/0001-fix-I2C_SMBUS_WORD_DATA-reads-and-writes.patch.txt)

ZetaPhoenix, 2022-08-01:
I see we now have `case NV_I2C_CMD_SMBUS_BLOCK_WRITE:`

Do we need one for Read? Is this expected to work for the EVGA cards? I did not see any changes that would change the feature advertisement support to the upper layers.

ZetaPhoenix, 2022-08-09:
@aritger Results testing with ubuntu and 3080Ti:

```
user@jammy:~$ sudo i2cdetect -l
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-1	i2c       	nvkm-0000:01:00.0-bus-0000      	I2C adapter
i2c-2	i2c       	nvkm-0000:01:00.0-bus-0001      	I2C adapter
i2c-3	i2c       	nvkm-0000:01:00.0-bus-0002      	I2C adapter
i2c-4	i2c       	nvkm-0000:01:00.0-bus-0003      	I2C adapter
i2c-5	i2c       	nvkm-0000:01:00.0-aux-0003      	I2C adapter
i2c-6	i2c       	nvkm-0000:01:00.0-bus-0004      	I2C adapter
i2c-7	i2c       	nvkm-0000:01:00.0-aux-0004      	I2C adapter
i2c-8	i2c       	nvkm-0000:01:00.0-bus-0005      	I2C adapter
i2c-9	i2c       	nvkm-0000:01:00.0-aux-0005      	I2C adapter
i2c-10	i2c       	nvkm-0000:01:00.0-bus-0006      	I2C adapter
i2c-11	i2c       	nvkm-0000:01:00.0-aux-0006      	I2C adapter
i2c-12	i2c       	nvkm-0000:01:00.0-bus-0007      	I2C adapter
i2c-13	i2c       	nvkm-0000:01:00.0-aux-0007      	I2C adapter
i2c-14	i2c       	nvkm-0000:01:00.0-bus-0008      	I2C adapter
i2c-15	i2c       	nvkm-0000:01:00.0-aux-0008      	I2C adapter
i2c-16	i2c       	nvkm-0000:01:00.0-bus-0009      	I2C adapter
i2c-17	i2c       	nvkm-0000:01:00.0-aux-0009      	I2C adapter
i2c-18	i2c       	sor-0006-0f84                   	I2C adapter
i2c-19	i2c       	sor-0006-0f44                   	I2C adapter
i2c-20	i2c       	sor-0006-0f82                   	I2C adapter
user@jammy:~$ sudo i2cdetect 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
user@jammy:~$ sudo i2cdetect 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- 2d -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
user@jammy:~$ sudo i2cdump 2 0x2d i
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2, address 0x2d, mode i2c block
Continue? [Y/n] y
Error: Block read failed, return code -6
user@jammy:~$ 
```
OpenRGB did not seem to run at all on the new i2c interfaces (from --loglevel 6):

```
....
1010  |Info:    ------------------------------------------------------
1010  |Info:    |             Detecting I2C interfaces               |
1010  |Info:    ------------------------------------------------------
1010  |Info:    Registering I2C interface: /dev/i2c-3 Device 10DE:2208 Subsystem: 3842:3953
1010  |Info:    Registering I2C interface: /dev/i2c-20 Device 0000:0000 Subsystem: 0000:0000
1011  |Info:    Registering I2C interface: /dev/i2c-10 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-1 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-19 Device 0000:0000 Subsystem: 0000:0000
1011  |Info:    Registering I2C interface: /dev/i2c-17 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-8 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-15 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-6 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-13 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-4 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-11 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-2 Device 10DE:2208 Subsystem: 3842:3953
1011  |Info:    Registering I2C interface: /dev/i2c-0 Device 8086:A323 Subsystem: 1849:A323
1012  |Info:    Registering I2C interface: /dev/i2c-18 Device 0000:0000 Subsystem: 0000:0000
1012  |Info:    Registering I2C interface: /dev/i2c-9 Device 10DE:2208 Subsystem: 3842:3953
1012  |Info:    Registering I2C interface: /dev/i2c-16 Device 10DE:2208 Subsystem: 3842:3953
1012  |Info:    Registering I2C interface: /dev/i2c-7 Device 10DE:2208 Subsystem: 3842:3953
1012  |Info:    Registering I2C interface: /dev/i2c-14 Device 10DE:2208 Subsystem: 3842:3953
1012  |Info:    Registering I2C interface: /dev/i2c-5 Device 10DE:2208 Subsystem: 3842:3953
1012  |Info:    Registering I2C interface: /dev/i2c-12 Device 10DE:2208 Subsystem: 3842:3953
....
--------
....
1354  |Debug:   [EVGA GeForce RTX 3080Ti XC3 Gaming] is enabled
1354  |Trace:   [ResourceManager] Calling detection progress callbacks.
1354  |Debug:   [EVGA GeForce RTX 3080Ti XC3 Gaming] no devices found
1354  |Trace:   [EVGA GeForce RTX 3080Ti XC3 Gaming] detection end
1354  |Debug:   [EVGA GeForce RTX 3080Ti XC3 Ultra Gaming] is enabled
....
```

rskaliotis, 2022-08-13:
@aritger I can confirm your patch fixes smbus word reads for me. Where previously the MSB would always come back as 0x00, I'm now getting full valid 16-bit words.

Could you apply this to the open source driver and the proprietary one? I'm hitting this issue on both. Thank you!

ValZapod, 2022-08-13:
> Can I clarify? We can finaly get RGB colors instead of only green for my founders edition 2080 Ti? https://youtu.be/bcKQ3YFnYM4
> 
> That always made me so angry! yum

So does it work now, or it still requires flashing a new VBIOS?

ZetaPhoenix, 2022-08-13:
> > Can I clarify? We can finaly get RGB colors instead of only green for my founders edition 2080 Ti? https://youtu.be/bcKQ3YFnYM4
> > That always made me so angry! yum
> 
> So does it work now, or it still requires flashing a new VBIOS?

OpenRGB has recently added support for the `NV_API Illumination` Controller which supports the 2070 FE so control may be possible but only under Windows. If NVIDIA is willing to expose a similar API for Linux or document what the API does (IE i2c commands) then OpenRGB can add support. https://gitlab.com/CalcProgrammer1/OpenRGB/-/commit/81b385a67e5f308036c62c4c40d106983cfc381d

aritger, 2022-08-15:
Thank you for testing the patches, @rskaliotis.  The fix will be included in our next release branch series, 520.xx, though I don't have a firm ETA.

@ZetaPhoenix: I think this code handles both smbus word writes and reads.

If there are other problems/requests, let's please track those as separate Issues.  I think they're distinct from the 16-bit smbus problem originally reported here.

ZetaPhoenix, 2022-08-15:
@aritger 

I still thinks this falls under the general i2c issues that were brought up but if you want it tracked as another issue that is fine.

The patch "added" support but it did not work so I do not think this patch is ready.

aritger, 2022-08-15:
If I understand correctly, the issue that @CalcProgrammer1 described in the first post was specific to 16-bit smbus transactions with the ENE controller.  That much seems to be addressed by the patch.

Can you tell, @ZetaPhoenix, what controller your board has?  I suspect it is a different controller, if your test results don't match @rskalioti's?

ZetaPhoenix, 2022-08-15:
The information was contained in my update here: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/41#issuecomment-1208903060 along with the error after applying the patch and attempting to do a block read.

```
[EVGA GeForce RTX 3080Ti XC3 Gaming]
1011  |Info:    Registering I2C interface: /dev/i2c-2 Device 10DE:2208 Subsystem: 3842:3953
```

darrellenns, 2022-08-15:
@aritger The original post by @CalcProgrammer1 says the following:

> We have also observed issues with SMBus block operations, though doing the same block operations using I2C_RDWR ioctl (thus avoiding the SMBus layer) seem to work on the NVIDIA proprietary driver.

Has this part been addressed by the patch at all? Based on the patch contents and the tests by @ZetaPhoenix, it seems that the patch only fixes the mixed word/byte operations, and does not include anything for the block operations.

aritger, 2022-08-16:
Fair enough.  We'll look into the block operation issue.  I've filed NVIDIA internal bug 3757700 to investigate that.

amrit1711, 2022-08-26:
@ZetaPhoenix 
1. Can you please request provide the output of  - `#i2cdetect -F x` where x is the bus address which responds with 2d.
2. If there is a difference seen with Open RM and monolithic RM driver versions.
3. We are seeing below result with RTX 3080 FTW3 Ultra, can you please confirm if this the same issue you are reporting and can be considered as repro.
3757700# i2cdump 3 0x2d i
Error: Adapter does not have I2C block read capability

I disabled the check for block read capability in i2c-tools source to force a read anyway. It fails with -
i2c-tools-4.3# ./tools/i2cdump 3 0x2d i
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-3, address 0x2d, mode i2c block
Continue? [Y/n] Y
Error: Block read failed, return code -22

ZetaPhoenix, 2022-08-28:
> @ZetaPhoenix
>
>     1. Can you please request provide the output of  - `#i2cdetect -F x` where x is the bus address which responds with 2d.
>
>     2. If there is a difference seen with Open RM and [...]

1. If you are asking about without the new modules installed they look like this: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/41#issuecomment-1173613486
2. I was testing against the default Ubuntu drivers and the new Open GPU KM provided here with the patch. The I2C now shows up as `nvkm-0000:01:00.0-bus-0001` instead of `i2c-0	i2c       	NVIDIA i2c adapter 1 at f:00.0`
3. Yes, that matches. You can also run the OpenRGB pipeline version and enable verbose logging and you can see without the Open GPU KM, we attempt to talk to the device but return an error so we skip adding the device (Firmware dump fails as the card does not support the block read). If you run under Windows the card should be detected through the nvapi.dll as that supports the block reads/writes.

darrellenns, 2022-08-29:
Just as another data point - I also see i2c block read/write not supported on my EVGA 3070 FTW3 Ultra Gaming. This is with the 5.15.65.01 driver (via Arch Linux nvidia-dkms package). 

```
# i2cdetect -l
i2c-0   i2c             NVIDIA i2c adapter 1 at 1:00.0          I2C adapter
i2c-1   i2c             NVIDIA i2c adapter 2 at 1:00.0          I2C adapter
i2c-2   i2c             NVIDIA i2c adapter 5 at 1:00.0          I2C adapter
i2c-3   i2c             NVIDIA i2c adapter 6 at 1:00.0          I2C adapter
i2c-4   i2c             NVIDIA i2c adapter 7 at 1:00.0          I2C adapter
i2c-5   i2c             NVIDIA i2c adapter 8 at 1:00.0          I2C adapter
i2c-6   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-7   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-8   i2c             Synopsys DesignWare I2C adapter         I2C adapter

# i2cdetect 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0.
I will probe address range 0x08-0x77.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- 2d -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

# i2cdetect -F 0
Functionalities implemented by /dev/i2c-0:
I2C                              yes
SMBus Quick Command              yes
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               no
SMBus Block Write                yes
SMBus Block Read                 yes
SMBus Block Process Call         no
SMBus PEC                        no
I2C Block Write                  no
I2C Block Read                   no

```

ZetaPhoenix, 2022-10-10:
Any update on the block commands?

aritger, 2022-10-10:
I /think/ we have block commands working now, though it didn't make it into the 515 (or, now 520) release series.  It should be in the release series after 520, though I can't give a firm schedule.

ZetaPhoenix, 2022-10-10:
Is there a patch to test?

aritger, 2022-10-10:
Note that I can easily generate at the moment; sorry.

NoX1De, 2022-10-17:
Hello, while I've skimmed the fairly lengthy correspondence regarding this issue, I've not spent a significant amount of time to fully understand the implications of the looming patch here.  I see in the original comment:

"The cards we've been focused on lately are the ASUS 3xxx series cards, which all use a similar I2C RGB chip that is also found on some ASUS motherboards and various manufacturers' RGB DRAM modules. The chip comes from ENE..."

I'm on Fedora 35 and have an ASUS 3080 12GB ROG Strix GPU I'm currently running 515.76 drivers where I'm attempting to get OpenRGB working to change the color of the cards LED(s).  I've scoured the internet and found this thread to be most relevant/potentially useful to resolving the issue I'm having if I'm not mistaken.  Am I understanding this thread here correctly in that this looming patch should solve the issue where openrgb does not seem to detect this card nor let you modify it's lighting colors? 

```
 sudo openrgb -v -l
------------------------------------------------------
|               Start device detection               |
------------------------------------------------------
Initializing HID interfaces: Success
------------------------------------------------------
|             Detecting I2C interfaces               |
------------------------------------------------------
Registering I2C interface: /dev/i2c-3 Device 10DE:220A Subsystem: 1043:886B
Registering I2C interface: /dev/i2c-1 Device 10DE:220A Subsystem: 1043:886B
Registering I2C interface: /dev/i2c-4 Device 10DE:220A Subsystem: 1043:886B
Registering I2C interface: /dev/i2c-2 Device 10DE:220A Subsystem: 1043:886B
Registering I2C interface: /dev/i2c-0 Device 10DE:220A Subsystem: 1043:886B
Registering I2C interface: /dev/i2c-5 Device 10DE:220A Subsystem: 1043:886B
------------------------------------------------------
|               Detecting I2C devices                |
------------------------------------------------------
------------------------------------------------------
|               Detecting HID devices                |
------------------------------------------------------
[X570 AORUS ELITE] Registering RGB controller
[Gigabyte RGB Fusion 2 USB] successfully added
------------------------------------------------------
|              Detecting other devices               |
------------------------------------------------------
------------------------------------------------------
|                Detection completed                 |
0------------------------------------------------------
: X570 AORUS ELITE
  Type:           Motherboard
  Description:    IT8297BX-GBX570
  Version:        0x00060001
  Location:       HID: /dev/hidraw1
  Serial:         redacted
  Modes: [Direct] Static Breathing Blinking 'Color Cycle' Flashing
  Zones: 'D_LED1 Bottom' 'D_LED2 Top' Motherboard
  LEDs: 'Back I/O' 'CPU Header' PCIe 'LED C1/C2'
  ```
Looking for ASUS detections...
```
sudo openrgb -vv -l | grep ASUS
[ASUS Aura SMBus Motherboard] is enabled
[ASUS Aura SMBus Motherboard] no devices found
[ASUS Aura SMBus Motherboard] detection end
[ASUS Aura GPU (ENE)] is enabled
[ASUS Aura GPU (ENE)] no devices found
[ASUS Aura GPU (ENE)] detection end
[ASUS Aura GPU] is enabled
[ASUS Aura GPU] no devices found
[ASUS Aura GPU] detection end
```

 Separately in running i2cdetect I get the following but no apparent way to use openrgb to modify the color of my ASUS 3080 lighting from the default.
  ```
  sudo i2cdetect -l
i2c-0	i2c       	NVIDIA i2c adapter 1 at 29:00.0 	I2C adapter
i2c-1	i2c       	NVIDIA i2c adapter 3 at 29:00.0 	I2C adapter
i2c-2	i2c       	NVIDIA i2c adapter 5 at 29:00.0 	I2C adapter
i2c-3	i2c       	NVIDIA i2c adapter 6 at 29:00.0 	I2C adapter
i2c-4	i2c       	NVIDIA i2c adapter 7 at 29:00.0 	I2C adapter
i2c-5	i2c       	NVIDIA i2c adapter 8 at 29:00.0 	I2C adapter
```

Will this patch that's apparently (hopefully) coming in the version after 520 going to rectify this problem?

owenmylotte, 2022-10-21:
> I /think/ we have block commands working now, though it didn't make it into the 515 (or, now 520) release series. It should be in the release series after 520, though I can't give a firm schedule.

This is great news! I've been following this thread for months. Thank you for looking into this.

atta2022, 2022-10-28:
Any news?
Tried to apply openrgb on my rig with 6 cards, 4 of them works, 2 not recognised.

sudo openrgb -l
Attempting to connect to local OpenRGB server.
Connection attempt failed
Local OpenRGB server unavailable.
Running standalone.
[i2c_smbus_linux] Failed to read i2c device PCI device ID
[i2c_smbus_linux] Failed to read i2c device PCI device ID
0: Gigabyte RTX3060 EAGLE OC 12G V2
  Type:           GPU
  Description:    RGB Fusion GPU
  Location:       I2C: /dev/i2c-23, address 0x63
  Modes: [Direct] Breathing Flashing 'Dual Flashing' 'Color Cycle' 'Spectrum Cycle'
  Zones: 'GPU Zone'
  LEDs: 'GPU LED'

1: Gigabyte RTX3060 EAGLE OC 12G V2
  Type:           GPU
  Description:    RGB Fusion GPU
  Location:       I2C: /dev/i2c-18, address 0x63
  Modes: [Direct] Breathing Flashing 'Dual Flashing' 'Color Cycle' 'Spectrum Cycle'
  Zones: 'GPU Zone'
  LEDs: 'GPU LED'

2: ASUS TUF RTX 3080Ti O12G GAMING
  Type:           GPU
  Description:    ENE SMBus Device
  Version:        AUMA0-E6K5-0107
  Location:       I2C: /dev/i2c-28, address 0x67
  Modes: Direct [Off] Static Breathing Flashing 'Spectrum Cycle' Rainbow 'Chase Fade' Chase 'Random Flicker'
  Zones: Unknown
  LEDs: 'Unknown LED 1' 'Unknown LED 2' 'Unknown LED 3' 'Unknown LED 4'

3: ASUS ROG STRIX 3090 O24G GAMING
  Type:           GPU
  Description:    ENE SMBus Device
  Version:        AUMA0-E6K5-0107
  Location:       I2C: /dev/i2c-12, address 0x67
  Modes: Direct [Off] Static Breathing Flashing 'Spectrum Cycle' Rainbow 'Chase Fade' Chase 'Random Flicker'
  Zones: Unknown
  LEDs: 'Unknown LED 1' 'Unknown LED 2' 'Unknown LED 3' 'Unknown LED 4' 'Unknown LED 5' 'Unknown LED 6' 'Unknown LED 7' 'Unknown LED 8' 'Unknown LED 9' 'Unknown LED 10' 'Unknown LED 11' 'Unknown LED 12' 'Unknown LED 13' 'Unknown LED 14' 'Unknown LED 15' 'Unknown LED 16' 'Unknown LED 17' 'Unknown LED 18' 'Unknown LED 19' 'Unknown LED 20' 'Unknown LED 21' 'Unknown LED 22'

sudo i2cdetect -l
i2c-0   smbus           SMBus I801 adapter at f040              SMBus adapter
i2c-1   i2c             i915 gmbus dpc                          I2C adapter
i2c-2   i2c             i915 gmbus dpb                          I2C adapter
i2c-3   i2c             i915 gmbus dpd                          I2C adapter
i2c-4   i2c             AUX B/DDI B/PHY B                       I2C adapter
i2c-5   i2c             AUX D/DDI D/PHY D                       I2C adapter
i2c-6   i2c             NVIDIA i2c adapter 1 at 1:00.0          I2C adapter
i2c-7   i2c             NVIDIA i2c adapter 4 at 1:00.0          I2C adapter
i2c-8   i2c             NVIDIA i2c adapter 5 at 1:00.0          I2C adapter
i2c-9   i2c             NVIDIA i2c adapter 6 at 1:00.0          I2C adapter
i2c-10  i2c             NVIDIA i2c adapter 7 at 1:00.0          I2C adapter
i2c-11  i2c             NVIDIA i2c adapter 8 at 1:00.0          I2C adapter
i2c-12  i2c             NVIDIA i2c adapter 1 at 3:00.0          I2C adapter
i2c-13  i2c             NVIDIA i2c adapter 3 at 3:00.0          I2C adapter
i2c-14  i2c             NVIDIA i2c adapter 5 at 3:00.0          I2C adapter
i2c-15  i2c             NVIDIA i2c adapter 6 at 3:00.0          I2C adapter
i2c-16  i2c             NVIDIA i2c adapter 7 at 3:00.0          I2C adapter
i2c-17  i2c             NVIDIA i2c adapter 8 at 3:00.0          I2C adapter
i2c-18  i2c             NVIDIA i2c adapter 1 at 4:00.0          I2C adapter
i2c-19  i2c             NVIDIA i2c adapter 5 at 4:00.0          I2C adapter
i2c-20  i2c             NVIDIA i2c adapter 6 at 4:00.0          I2C adapter
i2c-21  i2c             NVIDIA i2c adapter 7 at 4:00.0          I2C adapter
i2c-22  i2c             NVIDIA i2c adapter 8 at 4:00.0          I2C adapter
i2c-23  i2c             NVIDIA i2c adapter 1 at 5:00.0          I2C adapter
i2c-24  i2c             NVIDIA i2c adapter 5 at 5:00.0          I2C adapter
[... 94 lines cut ...]
GPU Excluded:    No

cat /proc/driver/nvidia/gpus/0000\:06\:00.0/information 
Model:           NVIDIA GeForce RTX 3080 Ti
IRQ:             139
GPU UUID:        GPU-e9a3381e-0cef-29d1-d96c-a9abf81593b1
Video BIOS:      94.02.71.80.79
Bus Type:        PCIe
DMA Size:        47 bits
DMA Mask:        0x7fffffffffff
Bus Location:    0000:06:00.0
Device Minor:    4
GPU Excluded:    No

cat /proc/driver/nvidia/gpus/0000\:07\:00.0/information 
Model:           NVIDIA GeForce RTX 3080 Ti
IRQ:             140
GPU UUID:        GPU-43262d09-645f-44ea-e4b7-9514bebc8e75
Video BIOS:      94.02.71.80.cc
Bus Type:        PCIe
DMA Size:        47 bits
DMA Mask:        0x7fffffffffff
Bus Location:    0000:07:00.0
Device Minor:    5
GPU Excluded:    No

LarryDCJ, 2022-10-31:
I'm available at larrydcj@icloud.com if you need a tester. I have an EVGA 3090 FTW3 Ultra.

aritger, 2022-11-10:
I think all of the items mentioned in this Issue are addressed by the 525.35 release.  I'd appreciate if testers could confirm.

SapphirusBeryl, 2022-11-12:
> I think all of the items mentioned in this Issue are addressed by the 525.35 release. I'd appreciate if testers could confirm.

I've got an EVGA 3080 FTW3, and can confirm the i2c interface is now working with 525.35. All functionality in OpenRGB pertaining to the RGB bar is in fact working.

`$ i2cdetect -l
i2c-0	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-1	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-3	i2c       	NVIDIA i2c adapter 1 at 8:00.0  	I2C adapter
i2c-4	i2c       	NVIDIA i2c adapter 5 at 8:00.0  	I2C adapter
i2c-5	i2c       	NVIDIA i2c adapter 6 at 8:00.0  	I2C adapter
i2c-6	i2c       	NVIDIA i2c adapter 7 at 8:00.0  	I2C adapter
i2c-7	i2c       	NVIDIA i2c adapter 8 at 8:00.0  	I2C adapter
$ openrgb -l
Attempting to connect to local OpenRGB server.
Connection attempt failed
Local OpenRGB server unavailable.
Running standalone.
0: EVGA GeForce RTX 3080 FTW3 Ultra LHR
  Type:           GPU
  Description:    EVGA Ampere RGB GPU Device
  Version:        1.01.15
  Location:       I2C: /dev/i2c-3, address 0x2D
  Modes: Off Direct Breathing 'Spectrum Cycle' 'Color Cycle' ['Rainbow Wave'] Wave Star 'Color Stack'
  Zones: 'Front Logo' 'End plate Logo' 'Back Logo' 'Addressable Header'
  LEDs: 'Front Logo' 'End plate Logo' 'Back Logo' 'Addressable Header'`

atta2022, 2022-11-13:
Can confirm its still NOT working for me:

atta2k22@atta2k22-BTC-B250:~/LINUX/bzminer$ sudo openrgb -l
Attempting to connect to local OpenRGB server.
Connection attempt failed
Local OpenRGB server unavailable.
Running standalone.
[i2c_smbus_linux] Failed to read i2c device PCI device ID
[i2c_smbus_linux] Failed to read i2c device PCI device ID
0: Gigabyte RTX3060 EAGLE OC 12G V2
  Type:           GPU
  Description:    RGB Fusion GPU
  Location:       I2C: /dev/i2c-23, address 0x63
  Modes: [Direct] Breathing Flashing 'Dual Flashing' 'Color Cycle' 'Spectrum Cycle'
  Zones: 'GPU Zone'
  LEDs: 'GPU LED'

1: Gigabyte RTX3060 EAGLE OC 12G V2
  Type:           GPU
  Description:    RGB Fusion GPU
  Location:       I2C: /dev/i2c-18, address 0x63
  Modes: [Direct] Breathing Flashing 'Dual Flashing' 'Color Cycle' 'Spectrum Cycle'
  Zones: 'GPU Zone'
  LEDs: 'GPU LED'

2: ASUS TUF RTX 3080Ti O12G GAMING
  Type:           GPU
  Description:    ENE SMBus Device
  Version:        AUMA0-E6K5-0107
  Location:       I2C: /dev/i2c-28, address 0x67
  Modes: Direct [Off] Static Breathing Flashing 'Spectrum Cycle' Rainbow 'Chase Fade' Chase 'Random Flicker'
  Zones: Unknown
  LEDs: 'Unknown LED 1' 'Unknown LED 2' 'Unknown LED 3' 'Unknown LED 4'

3: ASUS ROG STRIX 3090 O24G GAMING
  Type:           GPU
  Description:    ENE SMBus Device
  Version:        AUMA0-E6K5-0107
  Location:       I2C: /dev/i2c-12, address 0x67
  Modes: Direct [Off] Static Breathing Flashing 'Spectrum Cycle' Rainbow 'Chase Fade' Chase 'Random Flicker'
  Zones: Unknown
  LEDs: 'Unknown LED 1' 'Unknown LED 2' 'Unknown LED 3' 'Unknown LED 4' 'Unknown LED 5' 'Unknown LED 6' 'Unknown LED 7' 'Unknown LED 8' 'Unknown LED 9' 'Unknown LED 10' 'Unknown LED 11' 'Unknown LED 12' 'Unknown LED 13' 'Unknown LED 14' 'Unknown LED 15' 'Unknown LED 16' 'Unknown LED 17' 'Unknown LED 18' 'Unknown LED 19' 'Unknown LED 20' 'Unknown LED 21' 'Unknown LED 22'

Sun Nov 13 12:23:44 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.53       Driver Version: 525.53       CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+

CalcProgrammer1, 2022-11-13:
Which card isn't working for you?  It looks like your two ASUS cards are properly detected, as it reads the version string.  RGB Fusion GPU should work even without the changes made here as it just uses byte accesses.  Do note that OpenRGB has not been well tested against systems with multiple GPUs, maybe you're running into an issue due to that.  Would you be able to test them one at a time?

NoX1De, 2022-11-14:
I can confirm sadly that it is still not working for for me after upgrading to the new 525.53 drivers (ASUS 3080 12GB ROG Strix GPU on Fedora 35)...what can I do to help?

```
Mon Nov 14 12:17:43 2022
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.53	  Driver Version: 525.53       CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
```

```
:~$ sudo i2cdetect -l
i2c-0	i2c       	NVIDIA i2c adapter 1 at 29:00.0 	I2C adapter
i2c-1	i2c       	NVIDIA i2c adapter 3 at 29:00.0 	I2C adapter
i2c-2	i2c       	NVIDIA i2c adapter 5 at 29:00.0 	I2C adapter
i2c-3	i2c       	NVIDIA i2c adapter 6 at 29:00.0 	I2C adapter
i2c-4	i2c       	NVIDIA i2c adapter 7 at 29:00.0 	I2C adapter
i2c-5	i2c       	NVIDIA i2c adapter 8 at 29:00.0 	I2C adapter
```
```
:~$ sudo openrgb -vv -l | grep ASUS
[ASUS Aura SMBus Motherboard] is enabled
[ASUS Aura SMBus Motherboard] no devices found
[ASUS Aura SMBus Motherboard] detection end
[ASUS Aura GPU (ENE)] is enabled
[ASUS Aura GPU (ENE)] no devices found
[ASUS Aura GPU (ENE)] detection end
[ASUS Aura GPU] is enabled
[ASUS Aura GPU] no devices found
[ASUS Aura GPU] detection end
```
```
:~$ sudo openrgb -l
Attempting to connect to local OpenRGB server.
Connection attempt failed
Local OpenRGB server unavailable.
Running standalone.
0: X570 AORUS ELITE
  Type:           Motherboard
  Description:    IT8297BX-GBX570
  Version:        0x00060001
  Location:       HID: /dev/hidraw1
  Serial:         redacted
  Modes: [Direct] Static Breathing Blinking 'Color Cycle' Flashing
  Zones: 'D_LED1 Bottom' 'D_LED2 Top' Motherboard
  LEDs: 'Back I/O' 'CPU Header' PCIe 'LED C1/C2'  
```

[41 more replies]

closed 2025-01-21 by CalcProgrammer1