NVIDIA/open-gpu-kernel-modules #44: Please if possible allow the "Discussion" feature on github for this project.

Quardah, 2022-05-12:
Hi.

As you might imagine many of us will be eyeing this project due to the potential impacts of its release.

It would be very productive for the community to have the "Discussion" feature activated for this project so people can openly share their questions and results with the work presented.

Thank you for considering.

MilesBHuff, 2022-05-12:
Should help to reduce non-issue spam in the Issues section.

shbaskaran, 2022-05-12:
Thanks for the suggestion, Quardah. I've tagged this as an enhancement and will consider with our project admins enabling discussions. As we started out, we wanted to take an incremental approach to enabling such features - to allow us to get settled with GitHub, and gauge interest and response from the community.

AaronWebster, 2022-05-12:
I think this is an excellent suggestion. As with any codebase that's been developed over many years, it's useful to have a place to have such discussions. For example, I just read an issue or someone was confused about naming conventions... Since most of us were invited when decisions were being made, sharing that knowledge gets you into the right mindset.

shbaskaran, 2022-05-13:
The feature was enabled. See you there! :)

closed 2022-05-13 by shbaskaran