NVIDIA/open-gpu-kernel-modules #47: Support more UNIX-like systems like FreeBSD and Solaris

X547, 2022-05-12:


X547, 2022-05-12:
I am also interested in [Haiku operating system](https://www.haiku-os.org/) support that I use and contribute.

MarbleXeno, 2022-05-12:
I think that we should focus on Linux. The Dev team is smaller than for the Windows version of the driver so supporting multiple operating systems seems like a bad idea. Nobody really uses Haiku or any other different OS than Linux/Windows/MacOS so we should focus on developing the best drivers we can for Linux.

extrowerk, 2022-05-12:
> Nobody really uses Haiku

Citation needed.

kraileth, 2022-05-12:
> I think that we should focus on Linux. The Dev team is smaller than for the Windows version of the driver so supporting multiple operating systems seems like a bad idea.

On the contrary: Linux users have been arguing for the importance of such support e.g. by pointing to the fact that there's not just Windows and macOS. If the community would now rush to just be part of the "big three" and be ok with starving the rest, this is a betrayal of the ideal of Open Source. And this would hurt the whole ecosystem - including Linux.

> Nobody really uses Haiku or any other different OS (for daily driving) than Linux/Windows/MacOS so we should focus on developing the best drivers we can for Linux.

You are very wrong, sir (This comment was written on my FreeBSD workstation).

MarbleXeno, 2022-05-12:
>
> > Nobody really uses Haiku or any other different OS (for daily driving) than Linux/Windows/MacOS...
> 
> You are very wrong, sir (This comment was written on my FreeBSD workstation).

I respect anyone who uses Haiku or other Unix like systems. What I meant is that Windows, MacOS and Linux are mainstream, where mostly only tech geeks use other systems. I'm not saying there shouldn't be support in other systems, I'm just saying that we should mostly focus on polishing current drivers on Linux.

mtijanic, 2022-05-12:
Thank you for your interest!

Traditional NVIDIA drivers use the same binary blob (`nv-kernel.o`) on FreeBSD and Solaris and only have a minor OS-specific adapter layer that is available in source form when you download the drivers. The new "GSP" driver architecture is also very amenable to this approach and we do hope to support them eventually. 

Thus far, our focus has been on Linux only, and that is not likely to change for the next few releases while we add support for more consumer facing features (such as power management).

We will gladly work with the community if they decide to pursue this ahead of our own schedules, though. If anyone is planning on tackling this, I would strongly recommend reaching out to us first for guidance.

Thank you.

kraileth, 2022-05-12:
> I respect anyone who uses Haiku or other Unix like systems. What I meant is that Windows, MacOS and Linux are mainstream, where mostly only tech geeks use other systems. I'm not saying there [...]

Fair enough. There certainly is more demand for great Linux drivers, I also have no doubts there. Especially with efforts from Valve and Google's Stadia. But I wanted to bust that myth that "nobody" was seriously using other operating systems. I've come a long way from my first Riva TNT card (didn't like the 3dfx hype back then) when I still ran Windows to kind of fighting with nvidia cards on Linux to finally finding the proprietary driver in good shape on FreeBSD when I made that switch several years ago.

I don't have the time to play a lot of games anymore and am more interested in other features like supporting multiple screens. And going with nvidia has worked very well for me in that regard. While it's not an ideology for me, I do prefer open source drivers and would love to see other systems supported, too. However on platforms where leadership didn't try as hard to piss off everybody who doesn't love the GPL (there are very valid reasons for that) and nvidia specifically, things worked well before. It's great that things are moving into the right direction now and I hope that this marks the end of certain hostilities. But it would be very unfortunate if that lead to support for other operating systems to fall short. But according to mtijanic's answer that does not seem to be the plan which is great!

owariee, 2022-05-12:
The driver is now open source, the FreeBSD guys can do the same thing they are doing for AMDGPU/Mesa drivers and just write a layer so the driver will run as expected. I think this is the best choice for the "not mainstream OSes" since this is what happens since the begin. Linux is growing(thanks VALVe/Community) and the win that linux acquired will benefit all the other open source OSes by no doubt.

X547, 2022-05-12:
I hope that NVidia will keep current architecture with most driver part OS independent and per-OS support layers. That approach may be broken if NVidia decides to integrate driver to Linux upstream code base because Linux kernel maintainers may force to drop OS abstraction layer.

dylanspag, 2022-05-12:
> Traditional NVIDIA drivers use the same binary blob (`nv-kernel.o`) on FreeBSD and Solaris and only have a minor OS-specific adapter layer that is available in source form when you download the [...]

@mtijanic This is not at all my area of expertise, but wondering if an approach similar to how the traditional drivers were handled would be feasible for bringing CUDA (and other libraries like cuDNN) to FreeBSD? It’d be awesome to run AI workloads on FreeBSD in the future! 😄

shkhln, 2022-05-12:
"may" is an understatement. As a FreeBSD user I would be very pissed off about the termination of an (official) support.

>  if an approach similar to how the traditional drivers were handled would be feasible for bringing CUDA

[Tested with mixed results.](https://github.com/shkhln/libc6-shim/issues/1#issuecomment-643077457) Works for less ambitious things like OpenCL, though.

dylanspag, 2022-05-13:
> "may" is an understatement. As a FreeBSD user I would be very pissed off about the termination of an (official) support.

I _think_ this was in reference to “were handled”. It was not my intention to imply a discontinuation of official support for the existing, proprietary drivers. I’ve edited the response to hopefully reflect this. Apologies about any confusion.

shkhln, 2022-05-13:
Eh, that part refers to the post containing the word "may". Hence the quotation marks.

dylanspag, 2022-05-13:
Ah missed that, my bad!

dchmelik, 2022-05-13:
UNIX (*BSD/OpenSolaris/IllumOS--'genetic Unix,' I don't use term for non-Unix-like but certified ones like EulerOS) is mainstream for scientists/programmers (and still networking, at a time was vast majority) because, as originally designed according to plan (instead of 'make up as you go along') has cleaner/stabler software such as kernels.  Some server software does much more on *BSD than GNU/Linux, at least without too much recompiling (or not at all, and even strictly Unix-like GNU/Linux is sometimes rather forced to adopt non-Unix-like software to stay relevant to servers, breaking things that still don't break on *BSD w/same software, so that's sometimes preferable.)

jthomas281, 2022-05-13:
> Nobody really uses Haiku or any other different OS (for daily driving) than Linux/Windows/MacOS so we should focus on developing the best drivers we can for Linux.

I never thought I'd see the day where Linux enthusiasts employed the very same arguments hardware manufacturers used for not supporting Linux _against_ non-Linux FOSS kernels and I'm at a loss for words right now.

emaste, 2022-05-13:
> We will gladly work with the community if they decide to pursue this ahead of our own schedules, though. If anyone is planning on tackling this, I would strongly recommend reaching out to us first for guidance.

Hello from the [FreeBSD Foundation](https://freebsdfoundation.org/). We are very interested in seeing FreeBSD support in these open source drivers.

Although we don't have any plans to directly support this (with staff time) in the immediate future, we do entertain [project grant proposals](https://freebsdfoundation.org/get-involved/project-proposal-overview/) from the community. If someone is capable and interested in working on FreeBSD support in the near future please get in touch.

Foundation staff can help with guidance and code review, either as part of a project grant or in collaboration with upstream.

amonpaike, 2022-05-13:
> Nobody really uses Haiku or any other different OS

The funny thing is that I recognize in the comments at least 3 HAIKU excellent develpers full of potential,
One of these has already done the entire porting of HAIKU on RISC-V, he ported the AMD GPU drivers himself to run them on his RISC-V platform and since it was not enough, he ported the  x86 AMD gpu drivers  bypassing all the "junk" of DRI linux to collect better performance and better integration with the HAIKU philosophy, and directly supporting Vulkan and run Zink on top of it to have good OpenGL support and at the same time save coding and work.
I am sure he would give a big boost to the improvement of these nvidia drivers for all platforms..
I'm not a developer, but I'm a deep observer of this "no one used HAIKU"
I tell you just one thing, this HAIKU used by no one, is full of pioneers who look far ...
do not be obtuse.