NVIDIA/open-gpu-kernel-modules #57: Null pointer dereference in

qarmin, 2022-05-12:
**NVIDIA Driver Version**
Git

https://github.com/NVIDIA/open-gpu-kernel-modules/blob/69791e9bf5161c786d620312fd6396756c37cff9/kernel-open/nvidia/nv-pci.c#L708-L716

`nvl->minor_num` is used if `nvl` is null

amrit1711, 2022-05-12:
Thanks qarmin for highlighting the bug, I have filed a bug 3643046 internally for tracking purpose.

sickcodes, 2022-05-12:
May as well track it externally 🎉 nice find

ValZapod, 2022-05-12:
Oh, classic.

https://www.open-std.org/jtc1/sc22/wg21/docs/cwg_active.html#232

**Indirection through a null pointer is not undefined behavior**, there is an open issue in the standard.  Well described here: https://stackoverflow.com/questions/28482809/c-access-static-members-using-null-pointer

What is much more funny is this in the file, since that is always true, change to &&:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/69791e9bf5161c786d620312fd6396756c37cff9/kernel-open/nvidia/nv-pci.c#L102

PAR2020, 2022-07-08:
Resolved in PR #61

commit 28d25047 (2022-08-02): nv-pci: Fix nullptr dereference if device was not found

Closes #57
commit 888f90fa (2022-08-23): nv-pci: Fix nullptr dereference if device was not found

Closes #57
closed 2022-08-02 by aritger