NVIDIA/open-gpu-kernel-modules #62: Question: gsp.bin firmware can be redistributed? What license do they have?

stalkerg, 2022-05-12:
The question in the title. I believe currently it's not clear but at the same time a very important aspect. 

Thanks.

aritger, 2022-05-12:
Hi stalkerg.  Yes, this is an important question.  gsp.bin contained in the driver .run file is governed by the license in the driver. run file (the file LICENSE in the top level directory when you extract the .run file).

I'm not a lawyer, but I think this section specifically allows redistribution of gsp.bin:
```
2.1.2  Linux/FreeBSD Exception.  Notwithstanding the foregoing terms
of Section 2.1.1, SOFTWARE designed exclusively for use on the Linux or
FreeBSD operating systems, or other operating systems derived from the
source code to these operating systems, may be copied and redistributed,
provided that the binary files thereof are not modified in any way
(except for unzipping of compressed files).
```

Certainly the intent is to allow gsp.bin to be redistributed.

That said, we hope to gsp.bin can ultimately be used by Nouveau, independently of the rest of the NVIDIA driver stack.  We'll need to work with NVIDIA's lawyers to update this license file to allow that.  There is a clause in section 2.1.3 that may technically prohibit that today:
```
No Separation of Components.  The SOFTWARE is licensed as a
single product.  Its component parts may not be separated for use
on more than one computer, nor otherwise used separately from the
other parts.
```
Does this answer your question?  Were you specifically interested solely in redistribution?  Or redistribution separately from the rest of the NVIDIA driver?

stalkerg, 2022-05-12:
Yes, I am primarily interested in 
> Or redistribution separately from the rest of the NVIDIA driver?

without it, Nouveau can't be working out the box. I thought about redistribution alongside AMD and Intel firmware as part of the Linux kernel.

stalkerg, 2022-05-12:
@ValZapod yes it's an interesting but probably separate question. 
Also, do we need current firmware as well or `gsp.bin` is enough for everything?

mtijanic, 2022-05-12:
> Also, do we need current firmware as well or gsp.bin is enough for everything?

@stalkerg in addition to `gsp.bin`, a driver needs the relevant booter microcode from [src/nvidia/generated/g_bindata_*.c](https://github.com/NVIDIA/open-gpu-kernel-modules/tree/main/src/nvidia/generated). 
I am even less of a lawyer than @aritger but my understanding is that distribution of these binaries is governed by the MIT license header in the file. 

All other firmware, like the ones listed in the kernel commit just above, are not needed with `gsp.bin`.

stalkerg, 2022-05-12:
@mtijanic cool, it means for final happy we need wait nVidia lawyers about `gsp.bin`.

TechnologyClassroom, 2022-05-13:
From an integration standpoint, the easiest way to solve this issue would be to release the code to compile gsp.bin under the same license as this repository.

ValZapod, 2022-05-13:
> the easiest way to solve this issue would be to release the code to compile gsp.bin under the same license as this repository.

I think it is signed, so not possible (?). Anyway, ucode code is mosgly just insane binary arrays in the c code.

TechnologyClassroom, 2022-05-13:
I think I read something about a [signing key being leaked](https://www.theregister.com/2022/03/05/nvidia_stolen_certificate/) a few months back.  Would that be related to this blob?

ValZapod, 2022-05-13:
That was a key for dll and exe.

SuparnaWrites, 2022-05-15:
Open GPU kernel modules are available under open-source license, dual GPLv2 + MIT. It's based upon GSP and supports GPU drivers having Turing architecture.

throwaway1037, 2022-05-22:
As discussed in #35, the ethical and practical course of action is to liberate the firmware too.

aritger, 2022-12-05:
@stalkerg: Making the license more explicit about allowing separately redistributable gsp firmware is progressing through the byzantine path it needs to take.  Thanks for your patience.

stalkerg, 2022-12-09:
@aritger, thank you for the update! Seems like we have progressed just need to wait.

Conan-Kudo, 2023-02-24:
@aritger Any progress on this? It'd be great if we could have this for Linux 6.2 / Fedora 38...

TechnologyClassroom, 2023-05-31:
What is the license of gsp.bin, gsp_tu10x.bin, and gsp_ga10x.bin? Will the source code for these files be published?

ttabi, 2023-05-31:
> Will the source code for these files be published?

No.