NVIDIA/open-gpu-kernel-modules #70: Feature request: add CUDA

RoelVdP, 2022-05-12:
Dear NVIDIA, Thank you for this. Many people react with "Finally!". Now the next greatest thing would be to open source CUDA, or at least make it much easier to obtain (i.e. avoid developer site signup and signin). 

This is a feature request to add CUDA to this repo. 

We can also remember that CUDA rests itself, to some extent, on open source technology, and that utility libraries built on top of CUDA, like CUBLAS, CUFFT, and CUDAPP, are open source. Let's take this one step further! 

Thank you & God bless,
Roel Van de Paar

antoscha, 2022-05-12:
I think it's rather obvious that open-sourcing CUDA would be harmful for business.

RoelVdP, 2022-05-12:
That is the logical first thought. Indeed. But think a bit further.

AarSeBail, 2022-05-12:
We could also do an open source alternative to CUDA. Much easier with the driver in hand.

ValZapod, 2022-05-12:
> We could also do an open source alternative to CUDA. Much easier with the driver in hand.

And how would you do it for example for nvJPEG2000? That is like a new "standard library".

AarSeBail, 2022-05-12:
My, admittedly flawed, philosophy is that if it works for me, I don't care about anyone else.

RoelVdP, 2022-05-12:
> We could also do an open source alternative to CUDA. Much easier with the driver in hand.

This would seem to be one good reason for NVIDIA to add open source CUDA first: i.e. before exactly this happens.

mtijanic, 2022-05-12:
Thank you for your comments. I'm afraid there are currently no plans to open source the CUDA usermode. See also the FAQ on [this blog post](https://developer.nvidia.com/blog/nvidia-releases-open-source-gpu-kernel-modules/):
> **Will the source for user-mode drivers such as CUDA be published?**
> 
> These changes are for the kernel modules; while the user-mode components are untouched. So the user-mode will remain closed source and published with pre-built binaries in the driver and the CUDA toolkit.

As this repository is for GPU _kernel_ modules, I'm going to go ahead and close this issue. A good place to discuss CUDA/UMD related requests is the [developer forums](https://forums.developer.nvidia.com/).

mtijanic, 2022-05-12:
As a sidenote, please use appropriate labels to tag your issues. It is also fine to leave them untagged and let the maintainers take care of it, but using the wrong tag just creates confusion and more work. Thank you.

closed 2022-05-12 by mtijanic

RoelVdP, 2022-05-12:
@mtijanic this is a feature request. And one that could have potentially received valuable community input/thoughts. It may have been a bit premature to close this imho. If NVIDIA wants to start open sourcing more things, like this great move, then it may also want to shift it's policy towards more open ended discussion. 

I did not add the bug label, it was automatically added (even though the system says I did). It's a shortcoming in your GitHub setup it seems.

Leseratte10, 2022-05-12:
@mtijanic This happens due to #15 , if you don't want that, you need to add a "feature request" template or similar.

RoelVdP, 2022-05-12:
Can we please reopen this feature request and see what input/thoughts come out of this?

mtijanic, 2022-05-12:
@RoelVdP @Leseratte10 my bad on the label, thanks for pointing it out. We're looking into adding more templates. In the meantime, I'll just silently be removing the label where it doesn't apply.

mtijanic, 2022-05-12:
> Can we please reopen this feature request and see what input/thoughts come out of this?

I sympathize with your request, but I believe this is simply not the place to make it. The repo is specific to GPU kernel modules and the issues here are monitored almost exclusively by engineers working on the kernel code. The developer forums are more closely monitored by groups that the request actually applies to.

ryao, 2022-05-13:
The community could always write its own open source CUDA user mode libraries. For example, there is this:

https://gpuocelot.gatech.edu/about/

Just having an open source kernel driver means that the community could handle the userspace bits on its own.