NVIDIA/open-gpu-kernel-modules #72: Any support on older graphics cards

amiria703, 2022-05-12:
Hi everyone, first thanks nVidia for making this open-sourced, I'm wondering if it's open source, there would be any support for older legacy graphics card with new kernel and new xorgs?

StayBlueee, 2022-05-12:
> Hi everyone, first thanks nVidia for making this open-sourced, I'm wondering if it's open source, there would be any support for older legacy graphics card with new kernel and new xorgs?

Based on Nvidia's official response [here](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/19#issuecomment-1124360887), it seems like Nvidia won't be providing any support for Pascal and Maxwell with their open driver stack. It also seems like there won't be any way for Nouveau to leverage the firmware necessary to enable reclocking support.

I think, as of right now, this is only going to be meant for production in the Datacenter, as supported by the [article ](https://developer.nvidia.com/blog/nvidia-releases-open-source-gpu-kernel-modules/) released today by Nvidia alongside this driver release. 

I am hopeful that they will work closely with Nouveau to make *something* work on older cards, but that might very well be just wishful thinking.

Gordon01, 2022-05-12:
Older cards do not have a big GPU System Processor so Nvidia would be forced to actually open-source the driver, not just a glue kernel module like it happened here.

So no, not gonna happen.

mtijanic, 2022-05-12:
Thank you for your interest!

As @StayBlueee mentions above, GPU architectures prior to Turing do not have the necessary HW this driver requires and we won't be supporting them from this codebase. I have pinned #19 to consolidate all information regarding GPU support, so I will go ahead and close this as a duplicate. Please post any additional questions in #19.

Thank you.

closed 2022-05-12 by mtijanic