NVIDIA/open-gpu-kernel-modules #75: HDMI no audio @4K above 60Hz

adolfotregosa, 2022-05-12:
It as been an issue since driver 465.

With latest driver @4K we lose audio when refresh rate is above 60Hz.

ValZapod, 2022-05-12:
TruHD, Eac3 or PCM? What about 192k 24 bit PCM? Does that work? What about 20 bit, yep it does exist?

adolfotregosa, 2022-05-12:
> TrueHD, Eac3 or PCM? What about 192k 24 bit PCM? Does that work? What about 20 bit, yep it does exist?

Nothing. No audio comes out @4K with refresh rate above 60Hz

mtijanic, 2022-05-12:
Thank you adolfotregose for your report. I believe this one is already tracked internally as bug 200761422, and we are actively investigating the issue. As I understand it, it is not specific to the GSP driver architecture.

I have linked this issue with our internal tracker so the engineers working on it can reach out if needed.