NVIDIA/open-gpu-kernel-modules #82: Not work at all

sandikata, 2022-05-12:
**NVIDIA Driver Version**
515.43.04.

**GPU**
RTX 3050

**Describe the bug**
Not work at all

**To Reproduce**
Build the driver regarding the documentation, and using it for X.Org

**Expected behavior**
To work by design

**Please reproduce the problem, run nvidia-bug-report.sh, and attach the resulting nvidia-bug-report.log.gz.**
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8676676/nvidia-bug-report.log.gz)

[Xorg.0.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8676690/Xorg.0.log)

alcaparra, 2022-05-12:
nvidia moment

TheRealOne78, 2022-05-12:
> nvidia moment

LOL ya

sandikata, 2022-05-12:
I've managed to make it "work" but it freezes regularly, and only hard reset helps.

sandikata, 2022-05-12:
@mtijanic Do you have any idea about this, or do someone suffer it?

mtijanic, 2022-05-12:
Hi, thanks for the report!

From the attached logs, I see:
```
NVRM: Open nvidia.ko is only ready for use on Data Center GPUs.
NVRM: To force use of Open nvidia.ko on other GPUs, see the
NVRM: 'OpenRmEnableUnsupportedGpus' kernel module parameter described
NVRM: in the README.
```
Are you sure you are running the open source flavor of the driver? You can verify with `modinfo nvidia | grep license`

The logs also previously show failure to initialize nvidia.ko because another driver (probably nouveau) is already active.

Let's first make sure we're running the open source driver, and if so we can try to up the logging level and see what went wrong.

sandikata, 2022-05-12:
Hello again, well there's the correct bugreport.

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8681186/nvidia-bug-report.log.gz)

Don't have freeze at the moment (and i cannot provide any report while it freeze)

aritger, 2022-05-12:
@sandikata, here are few possible experiments:

(1) Do you see similar freezes with the 515.43.04 binary kernel modules, as well as the open kernel modules?

(2) Did you see similar freezes with previous NVIDIA driver releases?

(3) When the system freezes, is the system still accessible over the network?  Or, is the only option at that point to reboot?

sandikata, 2022-05-12:
Hello,

1. Yes only on OpenSource 515.43.04, proprietary from same version works.
2.  Here can say no
3. Mostly on heavy load (gaming, encoding and etc), not accessible in any way the only is reboot.

dylif, 2022-05-14:
I am having the same issue here:
Kernel: 5.17.7-arch1-1
Driver: nvidia-open 515.43.04

Seems like something in the kernel module is locking up whenever I run an OpenGL game (i.e. Minecraft), as there is a lot of reports of timeouts. Strangely running a Vulkan game (GTA 5 through Proton and DXVK) works perfectly. 

This does not happen with the proprietary kernel modules; both OpenGL and Vulkan games work perfectly.

Please note I have tried adding the `nvidia-drm.modeset=1` kernel parameter which changed nothing.

Attached is a snippet of a log that I think is of interest.
[nvidia.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8693995/nvidia.log)

Hopefully this helps

dylif, 2022-05-14:
I should add that if I am quick enough after the freeze occurs, I can switch to a tty and have it be fully functional.

ValZapod, 2022-05-15:
https://www.phoronix.com/scan.php?page=article&item=nvidia-r515-open&num=1

>Switching to just blacklisting the Nouveau DRM driver rather than using the nomodeset option works around that issue.

dylif, 2022-05-15:
I'm not sure if that was a reply to my comment, but I attempted to blacklist the nouveau driver, which didn't solve my issues. Upon further inspection, on my system nouveau was already blacklisted by the `nvidia-utils` package in `/usr/lib/modprobe.d`.

shamefulCake1, 2024-09-25:
Should be closed as outdated?