NVIDIA/open-gpu-kernel-modules #87: [Question] Is the source code for GTX 1650 drivers available yet ?

prorounak2005, 2022-05-12:
I use GTX 1650 ( code name - NV167 (TU117)) so I want to know whether the drivers for my gpu are available yet. So that I can compile my own drivers. Thank you so much for making nvidia drivers open source!!!! :heart: :smile:

StayBlueee, 2022-05-12:
> I use GTX 1650 ( code name - NV167 (TU117)) so I want to know whether the drivers for my gpu are available yet. So that I can compile my own drivers. Thank you so much for making nvidia drivers open source!!!! heart smile

Since your GPU is a Turing GPU, it is indeed available, but it is in an alpha stage and requires an optional kernel parameter to get it working (view [here](http://us.download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/kernel_open.html) for more information). It is still recommended by Nvidia to use the proprietary driver until the open GPU drivers matures enough to be ready for production on Geforce and Workstation applications (it is only currently ready for the Datacenter, as seen [here](https://developer.nvidia.com/blog/nvidia-releases-open-source-gpu-kernel-modules/)).

You can compile the driver yourself by following the instructions in the README if you wish to do so.

I am no expert on the matter, but I hope this answered your question!

prorounak2005, 2022-05-12:
> Since your GPU is a Turing GPU, it is indeed available, but it is in an alpha stage and requires an optional kernel parameter to get it working (view [...]

Oh thank god my drivers are open source but are open source drivers available for maxwell and pascal gpus like gt 710 etc etc ?

Leseratte10, 2022-05-12:
No, so far these drivers are only for 16xx and higher.

prorounak2005, 2022-05-12:
Ic sadge :'(

closed 2022-05-12 by prorounak2005