NVIDIA/open-gpu-kernel-modules #94: Crashes after waking from sleep state on OpenRC

TURING-V2, 2022-05-12:
**NVIDIA Driver Version**
515.43.04

**GPU**
1650 Ti Mobile

**Describe the bug**
I tried sleep state using s2ram works ,
In waking up from sleep power button shows wake state keyboard lights on mouse light offf and display off . Had to hard shutdown .
I saw they changed sleep servive in release notes for systemd.
Distro: Gentoo

**To Reproduce**
Try sleep in  OpenRC using s2ram.

**Expected behavior**
Works fine in 510 drivers

mtijanic, 2022-05-12:
Hello and thank you for your bug report!

Is this with the traditional closed source driver or the newly published one? You can check which one you're running with `modinfo nvidia | grep license`

If it is the open source driver, be aware that both suspend/resume and notebook support in general are not supported in the initial release and were not part of the test matrix. Honestly, I'm kinda surprised you even got that far.

If this is a regression in the closed source driver, best report it on the [developer forums](https://forums.developer.nvidia.com/c/gpu-graphics/linux/148) or send an email to [linux-bugs@nvidia.com](mailto:linux-bugs@nvidia.com), as outlined in [README.md](https://github.com/NVIDIA/open-gpu-kernel-modules#how-to-report-issues)

TURING-V2, 2022-05-12:
Its the new GPL license driver

mtijanic, 2022-05-12:
Thanks for confirming. Please sit tight, those features are coming in one of the future releases.

Gert-dev, 2022-05-15:
I think it's a more general issue with the open kernel modules, as I'm experiencing this also on stock Arch Linux x86-64 with systemd and GNOME 42.1 with an RTX 2070 Super Max-Q. Happens on both Wayland and X11, and does not happen with the proprietary, non-open drivers.

Sometimes the screen turns off, but the system never gets to suspend (I can tell by the power LED), at which point everything is stuck until you reboot, and sometimes the system actually does get to sleep, but resuming doesn't bring back the monitors.

I've got a feeling it's not actually crashing, though, since journalctl seems to imply stuff is still happening as usual, so possibly it's just the screens not being restored properly.

Possibly relevant logs:

```
 </TASK>
 ? entry_SYSCALL_64_after_hwframe+0x44/0xae
 ? do_syscall_64+0x67/0x80
 ? do_syscall_64+0x67/0x80
 ? do_syscall_64+0x67/0x80
 ? syscall_exit_to_user_mode+0x23/0x50
 ? do_syscall_64+0x67/0x80
 ? syscall_exit_to_user_mode+0x23/0x50
 ? exit_to_user_mode_prepare+0x78/0x170
 ? do_syscall_64+0x58/0x80
 ? __x64_sys_ioctl+0x8a/0xc0
 ? drm_mode_addfb_ioctl+0x10/0x10
 ? drm_ioctl+0x22a/0x410
 ? drm_ioctl_kernel+0xc3/0x170
 ? drm_mode_addfb_ioctl+0x10/0x10
 ? drm_mode_addfb2+0x40/0xd0
 ? drm_internal_framebuffer_create+0x329/0x420
 ? nv_drm_framebuffer_create+0x95/0xb0 [nvidia_drm f89a8950b3584323270460b2d0330878b9e7d7f4]
 ? nv_drm_internal_framebuffer_create+0x23b/0x350 [nvidia_drm f89a8950b3584323270460b2d0330878b9e7d7f4]
 CreateSurface+0x1e8/0x290 [nvidia_modeset 21a65b2950cc29858e53d480494b2190a965491a]
 nvkms_ioctl_from_kapi+0x22/0x80 [nvidia_modeset 21a65b2950cc29858e53d480494b2190a965491a]
 ? __kmalloc+0x182/0x350
 rwsem_down_read_slowpath+0x35d/0x3c0
 schedule+0x4b/0xc0
 ? nvDbg_vPrintf+0x84/0x1a0 [nvidia d6f22e9194d21bfd28ff8b38bb42b3395ccfe07f]
 __schedule+0x378/0x11d0
 <TASK>
Call Trace:
task:gnome-shell     state:D stack:    0 pid: 1213 ppid:  1138 flags:0x00000006
Freezing of tasks failed after 20.002 seconds (1 tasks refusing to freeze, wq_busy=0):
```

afeuerstein, 2022-06-07:
Can reproduce on Artix Linux w/ OpenRC. Propriatary drivers behave correctly, open-source drivers don't. Nice to see this will be worked upon. Thanks for your work so far!