NVIDIA/open-gpu-kernel-modules #130: NVML GSP mode function needs reworked

BlueGoliath, 2022-05-13:
In 510.68.02, a function was introduced called [nvmlDeviceGetGspFirmwareMode](https://github.com/NVIDIA/nvidia-settings/blob/7471c5b584c4d8df8d81c336c01b29b8e4b15b1d/src/nvml.h#L6631). This function has three issues with it:

1. On Linux, it will fill both pointers with zero(false) even on Pascal when it should instead return NVML_ERROR_NOT_SUPPORTED. This is different to what the previous GSP NVML function, [nvmlDeviceGetGspFirmwareVersion](https://github.com/NVIDIA/nvidia-settings/blob/7471c5b584c4d8df8d81c336c01b29b8e4b15b1d/src/nvml.h#L6614) does and gives users the impression that their GPU has GSP firmware when it does not. 

2. On Windows with a Maxwell based GPU, nvmlDeviceGetGspFirmwareMode seems to always return NVML_ERROR_NOT_SUPPORTED, which is different than on Linux and Pascal. If you can't already, it should be possible to get the firmware version even on Windows.

3. NVML_ERROR_NOT_SUPPORTED is returned despite not being documented.

Given that this new kernel driver requires GSP, making sure that it's possible to get reliable information as to whether their GPU has GSP firmware is important. A nvmlDeviceGetGspFirmwareMode_v2 function should be introduced and the old function deprecated to fix these problems.

ryao, 2022-05-13:
Both Pascal and Maxwell are unsupported by the GSP firmware. The only thing that appears to be needed here is to document the possible NVML_ERROR_NOT_SUPPORTED return value on unsupported hardware.

BlueGoliath, 2022-05-13:
The standard way of dealing with different behaviors is defining a new function version. See here:

https://github.com/NVIDIA/nvidia-settings/blob/7471c5b584c4d8df8d81c336c01b29b8e4b15b1d/src/nvml.h#L8728

For example, the symbol for nvmlDeviceGetGraphicsRunningProcesses_v2 exists in the library still despite being replaced by nvmlDeviceGetGraphicsRunningProcesses_v3.

Unrelated side note: just came across functions I haven't seen until now. Nerd sniped myself.

ValeZAA, 2023-11-23:
How to enable GSP on Windows 11?

aritger, 2023-11-29:
Blech; sorry.  The buttons are right next to each other...