NVIDIA/open-gpu-kernel-modules #133: Raspberry Pi 4 Compute Module support

ryao, 2022-05-13:
https://www.jeffgeerling.com/blog/2022/external-graphics-cards-work-on-raspberry-pi

There are issues with the BCM2711's PCIe root complex that make it difficult to get graphics cards working on the Raspberry Pi.

It would be extremely cool if some workarounds could be added to this driver to support the Raspberry 4 Compute Module.

I do not expect this to be a priority of any kind. I am putting the idea out there for those interested in nerd credit. :)

stalkerg, 2022-05-13:
@ryao honestly, I believe will be much easier to wait for Raspberry Pi 5 with fixed PCIe behavior, currently, it's just broken in some aspects. Broadcom should take attention here.

ApplePie420, 2022-05-13:
It would be neat to try on other ARM boards with PCIe capabilities, such as ROCKpro64 (RK3399 SoC), that *may* have the PCIe configured properly. It kinda sucks that RPi's almost always had their stuff somehow limited/bodged.

Qinka, 2022-05-17:
Will the Raspberry Pi 4 be any better with the UEFI firmware?

stalkerg, 2022-05-17:
@Qinka no, the main problem is that Raspberry Pi 4 not fully complain to PCIe standard by design.