NVIDIA/open-gpu-kernel-modules #135: ERROR: modpost: "backlight_device_register"  [...nvidia-modeset.ko] undefined

sndirsch, 2022-05-13:
**NVIDIA Driver Version**
515.43.04

**GPU**
Doesn't matter. Build issue.

**Describe the bug**
Build fails.
[...]
[   63s] ERROR: modpost: "backlight_device_register" [/home/abuild/rpmbuild/BUILD/nvidia-gfxG06-515.43.04/obj/azure/515.43.04/nvidia-modeset.ko] undefined!
[   63s] ERROR: modpost: "backlight_device_unregister" [/home/abuild/rpmbuild/BUILD/nvidia-gfxG06-515.43.04/obj/azure/515.43.04/nvidia-modeset.ko] undefined!

**To Reproduce**
Not so easy. For some reason this only occurs when building against -azure kernel flavor of SUSE Linux Enterprise 15 SP4.

**Expected behavior**
Build should work without any issues.

I'm attaching a fix which I stole from your 510.68.02 version. Not sure why this is not included in 515.43.04. It's also missing in GLUE sources for proprietary driver of 515.43.04.

[backlight.patch.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8683558/backlight.patch.txt)

ryao, 2022-05-13:
Which kernel version are you running?

sndirsch, 2022-05-13:
> Which kernel version are you running? i.e. uname -r

It's a heavily patched 5.14.21. But I'm afraid this information doesn't help. Question is. Has this code been removed by intention since 510.68.02 or is it just missing from 515.43.04. Obviously it's not useless. But this is a question only nVidia can answer.

mtijanic, 2022-05-13:
Hello @sndirsch and thank you for the interest. Is one of the patches of your "heavily patched" kernel the removal of `backlight_device_register()`?

The intention behind the removal was simply that the minimal supported kernel version was bumped to 3.10. My understanding is that all kernels >=3.10 (including 5.14.21) have the same interface for `backlight_device_register()`, so it was deemed that the conftest is no longer necessary.

hramrach, 2022-05-13:
It's not removed - the same source with different configuration works

azure kernel is configured for cloud instances that do not have a backlight so the driver is not needed, and is likely disabled

obj-$(CONFIG_BACKLIGHT_CLASS_DEVICE)    += backlight.o

mtijanic, 2022-05-13:
Makes sense, thanks. I believe @aritger is the person that maintains this code and can provide better info and decide how to proceed.

sndirsch, 2022-05-13:
Indeed backlight support is disabled in our azure config.

----  default 
+++ azure 
[...]
+# CONFIG_BACKLIGHT_APPLE is not set
+# CONFIG_BACKLIGHT_CLASS_DEVICE is not set
+# CONFIG_BACKLIGHT_GENERIC is not set
+# CONFIG_BACKLIGHT_GPIO is not set
+# CONFIG_BACKLIGHT_QCOM_WLED is not set
+# CONFIG_BACKLIGHT_SAHARA is not set
-CONFIG_DRM_NOUVEAU_BACKLIGHT=y
-CONFIG_FB_BACKLIGHT=m
-# Backlight & LCD device support
-CONFIG_BACKLIGHT_CLASS_DEVICE=y
-CONFIG_BACKLIGHT_KTD253=m
-CONFIG_BACKLIGHT_LM3533=m
-CONFIG_BACKLIGHT_PWM=m
-CONFIG_BACKLIGHT_APPLE=m
-CONFIG_BACKLIGHT_QCOM_WLED=m
-CONFIG_BACKLIGHT_RT4831=m
-CONFIG_BACKLIGHT_SAHARA=m
-CONFIG_BACKLIGHT_ADP8860=m
-CONFIG_BACKLIGHT_ADP8870=m
-CONFIG_BACKLIGHT_LM3630A=m
-CONFIG_BACKLIGHT_LM3639=m
-CONFIG_BACKLIGHT_LP855X=m
-CONFIG_BACKLIGHT_GPIO=m
-CONFIG_BACKLIGHT_LV5207LP=m
-CONFIG_BACKLIGHT_BD6107=m
-CONFIG_BACKLIGHT_ARCXCNN=m
-# end of Backlight & LCD device support
-CONFIG_HID_PICOLCD_BACKLIGHT=y
-CONFIG_LEDS_TRIGGER_BACKLIGHT=m
-CONFIG_CROS_KBD_LED_BACKLIGHT=m

hramrach, 2022-05-13:
The patch is needlessly convoluted - the test is not needed when NV_BACKLIGHT_DEVICE_REGISTER_PRESENT is replaced with CONFIG_BACKLIGHT_CLASS_DEVICE, BACKLIGHT_RAW should be present and there is no need to test for it.

aritger, 2022-05-13:
Hi Stefan.  Sorry for the regression.

It looks the pre-existing conftest for backlight_device_register was mistakenly removed from our code base when we updated the code to no longer perform conftests for pre-3.10 kernels.

I've filed NVIDIA-internal bug 3645334 for this, and we'll try to get it fixed ASAP.

FWIW, the attached backlight.patch.txt restores the conftest we had in earlier releases.  I agree it can be simplified.  That conftest did two things:

(1) check if the function was available
(2) check the number of arguments (4 or 5)

backlight_device_register was updated to take 5 arguments in Linux kernel commit a19a6ee6cad2 (dating back to v2.6.34).  Since we now support only 3.10 and later kernels, check (2) is no longer needed, but check (1) is.  The mistake was that we removed the conftest, only thinking about (2), and didn't consider (1).

sndirsch, 2022-05-14:
Hi Andy. Thanks for addressing that issue! :-)

hramrach, 2022-05-14:
Is there any reason why CONFIG_BACKLIGHT_CLASS_DEVICE cannot be used to determine if the backlight function is available and a test that replicates this define as NV_BACKLIGHT_DEVICE_REGISTER_PRESENT is added?

aritger, 2022-05-17:
Thanks for the suggestion.  Yes, it is simplest to check CONFIG_BACKLIGHT_CLASS_DEVICE in nvidia-modeset-linux.c.  We'll include that in a future release.

sndirsch, 2022-05-19:
FWIW, that's how Takashi resolved the issue by using existing ONFIG_BACKLIGHT_CLASS_DEVICE.
[backlight-workaround.patch.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8726875/backlight-workaround.patch.txt)

closed 2022-05-31 by aritger

sndirsch, 2022-05-31:
Hi @aritger I can confirm that  515.48.07 fixes this issue. None of these workarounds is stil needed. :-)