NVIDIA/open-gpu-kernel-modules #139: File ibmnpu_linux.h is missing causing PPC64LE build problem

mikey, 2022-05-13:
**NVIDIA Driver Version**
SHA1 

**GPU**
compile issue

**Describe the bug**
File "ibmnpu_linux.h" is missing resulting in this when compiling on ppc64le:

> nvidia-open-gpu-kernel-modules/kernel-open/nvidia/nv-ibmnpu.h:29:10: fatal error: ibmnpu_linux.h: No such file or directory
>    29 | #include "ibmnpu_linux.h"

**To Reproduce**
on a ppc64le machine do:
```
make modules
```

**Expected behavior**
It compiles

mikey, 2022-05-13:
Looks like this missing include provides at least the following that the driver needs:
```
void ibmnpu_init(void);
void ibmnpu_exit(void);
typedef ibmnpu_genregs_info_t;
```

ryao, 2022-05-13:
Does Nvidia support PPC64LE processors? I remember seeing something about it in the past, but I do not see it listed here:

https://www.nvidia.com/en-us/drivers/unix/

I also do not see a driver on the mirror:

https://download.nvidia.com/XFree86/

That said, I do see an old PPC64LE driver here:

https://www.nvidia.com/download/driverResults.aspx/128697/en-us

I also see a PPC64LE CUDA toolkit here:

https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=ppc64le&Distribution=RHEL&target_version=8&target_type=runfile_local

mikey, 2022-05-13:
> Does Nvidia support PPC64LE processors? 

Yes, The 2 of the top 3 supercomputers in the world (Summit and Sierra) have nvidia GPUs in PPC64LE machines. 

It's also pretty evident in the driver that ppc64e is (kinda) supported (ibmnpu == ppc64le)
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/kernel-open/nvidia/nv-ibmnpu.c

ryao, 2022-05-13:
@mikey Where can people download the nvidia driver for ppc64le? I looked, but could not find it.

Without a ppc64le build of 515.43.04, getting this working on ppc64le is not going to give people much.

That said, I am always happy to see support for more architectures. :)

aik, 2022-05-13:
There is such build:
https://developer.download.nvidia.com/compute/cuda/repos/rhel8/ppc64le/kmod-nvidia-latest-dkms-515.43.04-1.el8.ppc64le.rpm

aaronp24, 2022-05-13:
Thanks for trying this out. As per the README, ppc64le is not supported yet in this repository.
> Currently, the kernel modules can be built for x86_64 or aarch64.

Changing the label from `bug` to `enhancement`.

mikey, 2022-05-13:
@aaronp24 sure, would you take it as a feature? We've got it compiling by ripping out most of the IBM POWER NPU code.
https://github.com/mikey/open-gpu-kernel-modules/commits/ppc64le

shbaskaran, 2022-05-13:
I sorry, @mikey, due to our shared code base we cannot easily merge this feature into our internal repo so I don't think we can accept the changes. We do plan to support ppc64le in a future release. Let me see if I can nudge the build support forward - the NPU code is incompatible with the open source driver anyway since we do not support P9 + Volta.

ValZapod, 2022-05-13:
> We've got it compiling by ripping out most of the IBM POWER NPU code.

But that is not enough, it should also work... Who has access to SUMMIT supercomputer? 😋😋

ryao, 2022-05-14:
> But that is not enough, it should also work... Who has access to SUMMIT supercomputer? yumyum

It is not necessary. One of these would work just fine for testing PPC64LE:

https://www.raptorcs.com/

mikey, 2022-05-14:
@shbaskaran Thanks for the response. 

> I'm sorry, @mikey, due to our shared code base we cannot easily merge this feature into our internal repo so I don't think we can accept the changes. 

It's disappointing that you have an internal repo the public can't see that gates external contributions. Hopefully when you reach your goal of upstream in the Linux kernel, this won't be a problem.

> We do plan to support ppc64le in a future release. Let me see if I can nudge the build support forward - the NPU code is incompatible with the open source driver anyway since we do not support P9 + Volta.

Great. It would be good if we could just get the base PCIe driver working so we could use those pure PCIe based GPUs on PPC64le. That should hopefully involve no PPC64le specific code (just like most PCIe based drivers in the Linux kernel)

P9 + Volta + NVLink2 would be nice to have but I understand that's out of scope here.

riptl, 2024-07-29:
To build on upstream ppc64le, follow these steps

- Stub out any references to "npu"
- Add compile flags `-mno-altivec -mno-vsx -mno-mma -fno-asynchronous-unwind-tables`
  - Kernel code doesn't have SIMD extensions enabled, so would crash
  - .eh_frame generates weird relocs
- Strip eh_frame again for good measure `objcopy --remove-section .rela.eh_frame kernel-open/nvidia-modeset.ko kernel-open/nvidia-modeset.ko2`

Works for me on Fedora 40

riptl, 2024-07-29:
I'm hopeful we can maintain a minimal community patch set for ppc64le and kick it along as long as it takes for NVIDIA to upstream it. Except for 64K pages and the IOMMU, I would naively hope there aren't any big roadblocks.

carlosgonz0, 2024-07-30:
Hi 
I looking for this nvidia gpl2.0 driver fully support for power9.

Thanks

aritger, 2024-08-07:
Thanks for your comments @riptl.  Unfortunately, at this time, we don't have plans to support ppc64le with open-gpu-kernel-modules.  Sorry.

closed 2024-08-07 by aritger