NVIDIA/open-gpu-kernel-modules #141: Where is the code for MPS in the driver, please?

YixinSong-e, 2022-05-13:
I'm interested in the implementation of MPS in the GPU, but I didn't find a lot of relevant information in the project. I would be grateful if someone could give me some guidance on this.

aaronp24, 2022-05-13:
Thank you for your interest in the driver. I don't believe the code in this repository is responsible for handling MPS support so I don't think the developers here will be able to help you. Please refer to the [NVIDIA Developer Zone documentation](https://docs.nvidia.com/deploy/mps/index.html) for information on MPS and direct questions to the [CUDA developer forum](https://forums.developer.nvidia.com/c/accelerated-computing/cuda/206).

closed 2022-05-13 by aaronp24

YixinSong-e, 2022-05-13:
Thank you，so can I use MPS if I use an open source driver installation?

shbaskaran, 2022-05-13:
Yes, you should be able to use MPS on the open source driver. If you encounter any bugs that are unique to this driver please do report a bug here.