NVIDIA/open-gpu-kernel-modules #143: What are the SMC features of the GPU?

s66104444, 2022-05-13:
I see in the code that the GPU seems to have the ability to be partitioned into multiple vGPUs if SMC is enabled, anyway, I'm curious as to what exactly SMC means.

pityulaxd, 2022-05-13:
Not expert.
It seems its a Smart Memory Controller on the gpu itself, so i think on higher end cards they have the chip and than it support virtualization of memory.
Other than that this is what a search got me on it: https://www.supermicro.com/solutions/NVIDIA_vGPU_Virtual_Desktop_Reference_Architecture.pdf

mtijanic, 2022-05-13:
Hello and thank you for your interest!

SMC stands for "Simultaneous Multiple Contexts" and is a feature also known as MIG (Multi Instance GPU). Please see the [Ampere architecture whitepaper](https://images.nvidia.com/aem-dam/en-zz/Solutions/data-center/nvidia-ampere-architecture-whitepaper.pdf) for more details.

> The new Multi-Instance GPU (MIG) feature allows the A100 Tensor Core GPU to be securely
> partitioned into as many as seven separate GPU Instances for CUDA applications, providing
> multiple users with separate GPU resources to accelerate their applications and development
> projects.

s66104444, 2022-05-13:
Thanks! Help me a lot!

mtijanic, 2022-05-13:
Closing this, let's use #148 to consolidate all similar questions. Thanks!

closed 2022-05-13 by mtijanic