NVIDIA/open-gpu-kernel-modules #148: RM

s66104444, 2022-05-13:
I see a lot of RM in the code related to the RMAPI module, what does this RM mean exactly?

mtijanic, 2022-05-13:
Hello and thank you for your interest!

RM stands for "Resource Manager" and it is the internal name of the driver that gets compiled into `nvidia.ko` module - both the open source one (sometimes referred to as "OpenRM" internally) and the proprietary version. It can also refer to the team working on this code.

Depending on the context, GSP-RM can refer either to the new driver architecture that uses the "GSP" microcontroller, or it can refer to the code running on said microcontroller. To avoid ambiguity, the latter is often called "Physical RM" instead, and code running in kernel is called "Kernel RM".

RMAPI is the interface nvidia.ko (RM) exposes to other kernel modules and to userspace.

mtijanic, 2022-05-13:
Meta: Do you think it makes sense to group these questions into a single issue, which we could then pin?

s66104444, 2022-05-13:
Thanks! Yes, I think it's a good idea to group these questions into a single issue.

mtijanic, 2022-05-13:
Okay, this is now a pinned generic code-related Q&A issue.

DemiMarie, 2022-05-13:
Will this driver support GPU virtualization with untrusted guests on consumer hardware?  This would be a huge win for Qubes OS, which is currently forced to rely on software rendering.  Support for CUDA or for enterprise GPUs would not be as helpful, as Qubes OS is an end-user operating system focused on the desktop use-case.

mtijanic, 2022-05-13:
@DemiMarie At this point all I can say is that the current codebase does not support virtualization - neither as a host nor a guest. I don't currently have any information about future changes in this regard.