NVIDIA/open-gpu-kernel-modules #149: error building Arch

phush0, 2022-05-13:
I have error building source code.

```
make -C src/nvidia
make -C src/nvidia-modeset
make[1]: Entering directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia'
make[1]: Entering directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia-modeset'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia-modeset'
cd kernel-open/nvidia-modeset/ && ln -sf ../../src/nvidia-modeset/_out/Linux_x86_64/nv-modeset-kernel.o nv-modeset-kernel.o_binary
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[2]: Entering directory '/usr/lib/modules/5.17.6-arch1-1/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.1.0
  You are using:           cc (GCC) 12.1.0
  LD [M]  /tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko
  BTF [M] /tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko
dwarf_expr: unhandled 0x12 DW_OP_ operation
die__process_function: tag not supported 0x2f (template_type_parameter)!
Unsupported DW_TAG_unspecified_type(0x3b)
Encountered error while encoding BTF.
make[4]: *** [scripts/Makefile.modfinal:61: /tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko] Error 1
make[4]: *** Deleting file '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open/nvidia-modeset.ko'
make[3]: *** [scripts/Makefile.modpost:140: __modpost] Error 2
make[2]: *** [Makefile:1746: modules] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.17.6-arch1-1/build'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/tmp/makepkg/nvidia-open-dkms-tkg/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make: *** [Makefile:50: modules] Error 2
```

Arch 5.17.6 - all updates

ghtesting2020, 2022-05-13:
We already have our own package in the Arch (testing) repo called nvidia-open. The maintainers were fast, was ready mere hours after this went open.

https://old.reddit.com/r/archlinux/comments/unpt7t/help_test_nvidiaopen/

They are looking for testers as well if you want to help out.

phush0, 2022-05-13:
Problem was to add `!buildflags`, otherwise it will use makepkg flags which can be changed as in my case

aritger, 2022-05-13:
Thanks.  It sounds like this isn't a problem with the open-gpu-kernel-modules repository.  Closing.  Please let me know if I've misunderstood.

closed 2022-05-13 by aritger

ZSL98, 2022-11-25:
> 

Could you please elaborate on the solution? Thanks a lot!

sTiKyt, 2023-01-27:
> > 
> 
> Could you please elaborate on the solution? Thanks a lot!

See https://github.com/archlinux/svntogit-packages/blob/packages/nvidia-open/trunk/PKGBUILD