NVIDIA/open-gpu-kernel-modules #151: aarch64 build fails

frezbo, 2022-05-13:
**Describe the bug**
aarch64 compilation fails

**To Reproduce**
Building using qemu aarch64 emulation

```bash
#0 812.3 In file included from ./include/linux/kernel.h:16,
#0 812.3                  from ./include/linux/list.h:9,
#0 812.3                  from ./include/linux/preempt.h:11,
#0 812.3                  from ./include/linux/spinlock.h:55,
#0 812.3                  from /tmp/build/0/kernel-open/common/inc/nv-lock.h:29,
#0 812.3                  from /tmp/build/0/kernel-open/common/inc/nv-linux.h:32,
#0 812.3                  from /tmp/build/0/kernel-open/nvidia/nv-vm.c:26:
#0 812.3 /tmp/build/0/kernel-open/nvidia/nv-vm.c: In function 'nv_get_max_sysmem_address':
#0 812.3 ./include/linux/minmax.h:20:35: warning: comparison of distinct pointer types lacks a cast
#0 812.3    20 |         (!!(sizeof((typeof(x) *)1 == (typeof(y) *)1)))
#0 812.3       |                                   ^~
#0 812.3 ./include/linux/minmax.h:26:18: note: in expansion of macro '__typecheck'
#0 812.3    26 |                 (__typecheck(x, y) && __no_side_effects(x, y))
#0 812.3       |                  ^~~~~~~~~~~
#0 812.3 ./include/linux/minmax.h:36:31: note: in expansion of macro '__safe_cmp'
#0 812.3    36 |         __builtin_choose_expr(__safe_cmp(x, y), \
#0 812.3       |                               ^~~~~~~~~~
#0 812.3 ./include/linux/minmax.h:52:25: note: in expansion of macro '__careful_cmp'
#0 812.3    52 | #define max(x, y)       __careful_cmp(x, y, >)
#0 812.3       |                         ^~~~~~~~~~~~~
#0 812.3 /tmp/build/0/kernel-open/nvidia/nv-vm.c:225:26: note: in expansion of macro 'max'
#0 812.3   225 |         global_max_pfn = max(global_max_pfn, node_end_pfn(node_id));
#0 812.3       |                          ^~~
#0 813.0 /tmp/build/0/kernel-open/nvidia/nv-mmap.c: In function 'nv_encode_caching':
#0 813.0 /tmp/build/0/kernel-open/nvidia/nv-mmap.c:343:16: warning: this statement may fall through [-Wimplicit-fallthrough=]
#0 813.0   343 |             if (NV_ALLOW_CACHING(memory_type))
#0 813.0       |                ^
#0 813.0 /tmp/build/0/kernel-open/nvidia/nv-mmap.c:346:9: note: here
#0 813.0   346 |         default:
#0 813.0       |         ^~~~~~~
#0 813.8   CC [M]  /tmp/build/0/kernel-open/nvidia/os-pci.o
#0 814.9   CC [M]  /tmp/build/0/kernel-open/nvidia/os-registry.o
#0 816.0   CC [M]  /tmp/build/0/kernel-open/nvidia/os-usermap.o
#0 817.2   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-modeset-interface.o
#0 817.6   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-pci-table.o
#0 820.8   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-kthread-q.o
#0 821.9   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-memdbg.o
#0 822.2   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-ibmnpu.o
#0 824.0   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-report-err.o
#0 825.3   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-rsync.o
#0 828.1   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-msi.o
#0 831.0   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-caps.o
#0 834.4   CC [M]  /tmp/build/0/kernel-open/nvidia/nv-frontend.o
#0 836.0   CC [M]  /tmp/build/0/kernel-open/nvidia/nv_uvm_interface.o
#0 836.6   CC [M]  /tmp/build/0/kernel-open/nvidia/nvlink_linux.o
#0 838.9   CC [M]  /tmp/build/0/kernel-open/nvidia/nvlink_caps.o
#0 843.2   CC [M]  /tmp/build/0/kernel-open/nvidia/linux_nvswitch.o
#0 844.8   CC [M]  /tmp/build/0/kernel-open/nvidia/procfs_nvswitch.o
#0 845.3   CC [M]  /tmp/build/0/kernel-open/nvidia/i2c_nvswitch.o
#0 846.4   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_common.o
#0 851.0   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_linux.o
#0 854.1   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/nvstatus.o
#0 855.2   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/nvCpuUuid.o
#0 855.5   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/nv-kthread-q.o
#0 857.4   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/nv-kthread-q-selftest.o
#0 858.4   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm.o
#0 859.1   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_tools.o
#0 859.2   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_global.o
#0 860.3   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_gpu.o
#0 865.8   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_gpu_isr.o
#0 872.8   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_procfs.o
#0 874.8   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_va_space.o
#0 875.1   CC [M]  /tmp/build/0/kernel-open/nvidia-uvm/uvm_va_space_mm.o
[... 151 lines cut ...]
#0 1168.3    96 |     struct drm_plane *primary_plane = crtc->primary;
#0 1168.3       |                       ^~~~~~~~~~~~~
#0 1168.7 ld -r -o /tmp/build/0/kernel-open/nvidia/nv-interface.o /tmp/build/0/kernel-open/nvidia/nv.o /tmp/build/0/kernel-open/nvidia/nv-pci.o /tmp/build/0/kernel-open/nvidia/nv-dmabuf.o /tmp/build/0/kernel-open/nvidia/nv-acpi.o /tmp/build/0/kernel-open/nvidia/nv-cray.o /tmp/build/0/kernel-open/nvidia/nv-dma.o /tmp/build/0/kernel-open/nvidia/nv-i2c.o /tmp/build/0/kernel-open/nvidia/nv-mmap.o /tmp/build/0/kernel-open/nvidia/nv-p2p.o /tmp/build/0/kernel-open/nvidia/nv-pat.o /tmp/build/0/kernel-open/nvidia/nv-procfs.o /tmp/build/0/kernel-open/nvidia/nv-procfs-utils.o /tmp/build/0/kernel-open/nvidia/nv-usermap.o /tmp/build/0/kernel-open/nvidia/nv-vm.o /tmp/build/0/kernel-open/nvidia/nv-vtophys.o /tmp/build/0/kernel-open/nvidia/os-interface.o /tmp/build/0/kernel-open/nvidia/os-mlock.o /tmp/build/0/kernel-open/nvidia/os-pci.o /tmp/build/0/kernel-open/nvidia/os-registry.o /tmp/build/0/kernel-open/nvidia/os-usermap.o /tmp/build/0/kernel-open/nvidia/nv-modeset-interface.o /tmp/build/0/kernel-open/nvidia/nv-pci-table.o /tmp/build/0/kernel-open/nvidia/nv-kthread-q.o /tmp/build/0/kernel-open/nvidia/nv-memdbg.o /tmp/build/0/kernel-open/nvidia/nv-ibmnpu.o /tmp/build/0/kernel-open/nvidia/nv-report-err.o /tmp/build/0/kernel-open/nvidia/nv-rsync.o /tmp/build/0/kernel-open/nvidia/nv-msi.o /tmp/build/0/kernel-open/nvidia/nv-caps.o /tmp/build/0/kernel-open/nvidia/nv-frontend.o /tmp/build/0/kernel-open/nvidia/nv_uvm_interface.o /tmp/build/0/kernel-open/nvidia/nvlink_linux.o /tmp/build/0/kernel-open/nvidia/nvlink_caps.o /tmp/build/0/kernel-open/nvidia/linux_nvswitch.o /tmp/build/0/kernel-open/nvidia/procfs_nvswitch.o /tmp/build/0/kernel-open/nvidia/i2c_nvswitch.o
#0 1170.8 ld -r -o /tmp/build/0/kernel-open/nvidia-modeset/nv-modeset-interface.o /tmp/build/0/kernel-open/nvidia-modeset/nvidia-modeset-linux.o /tmp/build/0/kernel-open/nvidia-modeset/nv-kthread-q.o
#0 1170.9   LD [M]  /tmp/build/0/kernel-open/nvidia-modeset.o
#0 1173.7   LD [M]  /tmp/build/0/kernel-open/nvidia-uvm.o
#0 1178.1   LD [M]  /tmp/build/0/kernel-open/nvidia-drm.o
#0 1178.8   LD [M]  /tmp/build/0/kernel-open/nvidia-peermem.o
#0 1180.5   MODPOST /tmp/build/0/kernel-open/Module.symvers
#0 1181.2 ERROR: modpost: "__aarch64_ldadd4_acq_rel" [/tmp/build/0/kernel-open/nvidia.ko] undefined!
#0 1181.2 ERROR: modpost: "__aarch64_ldclr4_acq_rel" [/tmp/build/0/kernel-open/nvidia.ko] undefined!
#0 1181.2 ERROR: modpost: "__aarch64_cas4_acq_rel" [/tmp/build/0/kernel-open/nvidia.ko] undefined!
#0 1181.2 ERROR: modpost: "__aarch64_ldset8_acq_rel" [/tmp/build/0/kernel-open/nvidia.ko] undefined!
#0 1181.2 ERROR: modpost: "__aarch64_ldadd8_acq_rel" [/tmp/build/0/kernel-open/nvidia.ko] undefined!
#0 1181.2 ERROR: modpost: "__aarch64_cas8_acq_rel" [/tmp/build/0/kernel-open/nvidia.ko] undefined!
#0 1181.2 ERROR: modpost: "__aarch64_ldset4_acq_rel" [/tmp/build/0/kernel-open/nvidia.ko] undefined!
#0 1181.2 make[3]: *** [scripts/Makefile.modpost:134: /tmp/build/0/kernel-open/Module.symvers] Error 1
#0 1181.2 make[3]: *** Deleting file '/tmp/build/0/kernel-open/Module.symvers'
#0 1181.2 make[2]: *** [Makefile:1783: modules] Error 2
#0 1181.2 make[2]: Leaving directory '/src'
#0 1181.2 make[1]: *** [Makefile:82: modules] Error 2
#0 1181.2 make[1]: Leaving directory '/tmp/build/0/kernel-open'
#0 1181.2 make: *** [Makefile:50: modules] Error 2

```

frezbo, 2022-05-13:
the proprietary version of the modules build fine in the same environment

hramrach, 2022-05-13:
It does not build nearly as much code as this one.

Anyway, you can try to build with V=1 and see if there is any difference in the flags passed to the compiler.

aritger, 2022-05-13:
Note that rather than V=1, to enable verbose output for all of this build, add `NV_VERBOSE=1` to the command line.

To debug unresolved symbol problems, I normally do something like this:

`find . -name "*.o" | xargs nm -A  | grep __aarch64_ldadd4_acq_rel`

That will tell you which object files call the symbol and which object files define the symbol.

From a few google searches, I think the aarch64 build may need "-mno-outline-atomics" (some gcc configs apparently have -moutline-atomics enabled by default).

Does the following change fix the problem?
```

$ git diff
diff --git a/src/nvidia/Makefile b/src/nvidia/Makefile
index 9bdb826ad249..687c46c11b6c 100644
--- a/src/nvidia/Makefile
+++ b/src/nvidia/Makefile
@@ -67,6 +67,7 @@ ifeq ($(TARGET_ARCH),aarch64)
   CFLAGS += -mgeneral-regs-only
   CFLAGS += -march=armv8-a
   CFLAGS += -mstrict-align
+  CFLAGS += -mno-outline-atomics
 endif

 CFLAGS += -fno-pic

```

frezbo, 2022-05-13:
@aritger the above patch fixes the build, thanks :+1:

hramrach, 2022-05-13:
According to documentation of recent GCC -mno-outline-atomics is required if you are building for -march=armv8-a (without lse) and your environment does not have libgcc (which the kernel does not).

Building for -march=armv8-a+lse, -march=armv8.1-a, or any later architecture avoids the helpers for lse detection because the instructions are assumed to be available, and conversely -mno-outline-atomics generates pure armv8-a code without lse instructions.

The question is what's the desired outcome, how the architecture gets picked, and for what architecture nVidia builds the binary drivers.

Seeing the build output with the verbose flag would answer these questions. I can only guess that the build uses -march=native and builds for some arbitrary architecture that nVidia uses in-house while this emulated environment is armv8-a leading to gcc generating the compatibility helpers.

woachk, 2022-05-14:
Please condition the ` -mno-outline-atomics ` check on GCC version though, to not break older OS releases.

frezbo, 2022-05-14:
Attaching the full build log with `V=1` and `NV_VERBOSE=1` set

[V=1.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8693686/V.1.txt)

[NV_VERBOSE=1.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8693687/NV_VERBOSE.1.txt)

hramrach, 2022-05-15:
Thanks, so either flag does equally good job at showing the compiler flags. 

`-march=armv8-a `

is explicitly passed.

This flag is present in a few files in the project and you can amend it with eg.

`sed -i -e s/-march=armv8-a/-march=armv8-a+lse/g */*/Makefile */Kbuild`

nVidia is propbably using an older compiler that does not support these lse helpers and `-moutline-atomics` internally so adding `-mno-outline-atomics` unconditionally would fail.

Using `-march=armv8-a+lse` or `-march=armv8.1-a` might be easiest if all target systems support the newer instruction set and all target compilers support the architecture designation.

pendragons-code, 2022-05-15:
> Thanks, so either flag does equally good job at showing the compiler flags.
>
> `-march=armv8-a `
>
> is explicitly passed.
>
> This flag is present in a few files in the project and you can amend it with [...]

can confirm this seems accurate

aritger, 2022-05-16:
Thanks for the thorough comments, @hramrach.

In general, we try to build one binary that runs across a large set of CPUs.  For now, I'd be reluctant to change -march.

TEST_CC_ARG can be used in the makefile to test if the compiler supports the flag, so I hope this is a workable solution:

```
diff --git a/src/nvidia/Makefile b/src/nvidia/Makefile
index 9bdb826ad249..2a3773131076 100644
--- a/src/nvidia/Makefile
+++ b/src/nvidia/Makefile
@@ -67,6 +67,7 @@ ifeq ($(TARGET_ARCH),aarch64)
   CFLAGS += -mgeneral-regs-only
   CFLAGS += -march=armv8-a
   CFLAGS += -mstrict-align
+  CONDITIONAL_CFLAGS += $(call TEST_CC_ARG, -mno-outline-atomics)
 endif

 CFLAGS += -fno-pic
```

hramrach, 2022-05-16:
Why is it added to CONDITIONAL_CFLAGS and not CFLAGS?

Also this is only 1 of the 3 places that have -march=armv8-a

woachk, 2022-05-16:
@hramrach to not break compatibility with configs that use older compilers.

Enabling LSE by default would break CPUs as recent as the Cortex-A72 and the ThunderX... so it's not a good option at all.

aritger, 2022-05-25:
FYI - our next release should contain a fix for this.

@hamrach: you are correct, there were three places that required updating.  The assignment to `CONDITIONAL_CFLAGS` in my previous comment was intentional: it is appended to CFLAGS later:
```
CFLAGS += $(CONDITIONAL_CFLAGS)
```
The difference, though, is that CONDITIONAL_CFLAGS is a simply expanded gnu make variable:
```
CONDITIONAL_CFLAGS :=
```
whereas CFLAGS is a recursively expanded gnu make variable (these different types of variables are described on https://www.gnu.org/software/make/manual/make.html#Flavors )

Normally, recursively expanded is more idiomatic in make.  But, the use of TEST_CC_ARG triggers a test invocation of the compiler, and we want to do that only once, rather than each time CFLAGS is used.

closed 2022-05-31 by aritger