NVIDIA/open-gpu-kernel-modules #161: GNOME fails to initialize NVIDIA dGPU if secondary screen is directly attached to it in hybrid setups

Gert-dev, 2022-05-13:
**NVIDIA Driver Version**
515.43.04

**GPU**
NVIDIA RTX 2070 Super Max-Q

**Describe the bug**
In a setup with an Intel iGPU and the NVIDIA acting as dGPU (hybrid graphics), attaching a secondary monitor over DisplayPort on a port wired to the NVIDIA dGPU directly will cause errors on GNOME 42.1, making it fall back to using the NVIDIA dGPU only as KMS device (display target), but not as render device, resulting in very low performance on that monitor due to rendering happening on the iGPU instead.

mutter 42.1 prints:

```
gnome-shell[1280]: Secondary GPU initialization failed (Failed to create gbm_surface: Operation not permitted). Falling back to GPU-less mode instead, so the secondary monitor may be slow to update.
```

This is due to the change in [this MR](https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/2341), which is a fallback fix as it simply crashed mutter entirely before.

The NVIDIA driver prints:

```
NVRM osCreateOsDescriptorFromDmaBufPtr: osCreateOsDescriptorFromDmaBufPtr(): Error (86) while trying to import dma_buf!
```

Additional relevant mutter bug reports: 

* [[Wayland][NVIDIA] Display freezes when I plug external display into optimus laptop with NVIDIA proprietary drivers
](https://gitlab.gnome.org/GNOME/mutter/-/issues/1891)
* [NVIDIA/Wayland/optimus external monitors always blank](https://gitlab.gnome.org/GNOME/mutter/-/issues/2166)
* [Wayland not working on optimus laptop on display with proprietary NVIDIA driver.
](https://gitlab.gnome.org/GNOME/mutter/-/issues/1572#note_1308007)

This issue also exists with the proprietary driver, hence the existing reports, but it looks like the exact same issue exists with this driver.

Also, **to clarify, a single display works fine with hybrid graphics**. This is not quite the same as #98 as that also applies to a situation with just a single display.

**To Reproduce**
1. Use a laptop or device with an Intel iGPU and integrated display, and a DisplayPort (over USB-C, in my case) port wired directly to the NVIDIA dGPU.
2. Log in to your account on GDM and GNOME 42.1.
3. The screens will flash, and the secondary monitor eventually turns on, but performance is very poor on it.

Alternatively, just plugging in the secondary monitor after being logged in also exposes the issue.

It might also be reproducible with any single-GPU device that has an external (e.g. Thunderbolt) dGPU with a monitor attached to it.

**Expected behavior**
The above errors are not printed, and mutter is able to render optimally for the secondary monitor without dropping the NVIDIA dGPU. Preferably the NVIDIA dGPU both _renders_ data that need to be displayed on the secondary monitor _and_ outputs it to the secondary monitor directly _without_ unnecessarily copying back to the iGPU (assuming mutter assigns rendering of content on the secondary monitor to the NVIDIA dGPU as well, so a copy can be avoided when the NVIDIA dGPU acts both as renderer and KMS device).

**Please reproduce the problem, run nvidia-bug-report.sh, and attach the resulting nvidia-bug-report.log.gz.**

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8690924/nvidia-bug-report.log.gz)

aritger, 2022-05-13:
Thanks for the detailed bug report.  Knowing that it also reproduces with the proprietary kernel modules helps.  Could you also check if this is a recent regression across NVIDIA driver releases?

geminis3, 2022-05-13:
> Could you also check if this is a recent regression across NVIDIA driver releases?

Screens attached to dGPU have never worked for me on hybrid systems under Wayland, my current device has a GTX 1060 which isn't supported by this open driver however it's also affected.

According to another NVIDIA developer this issue is being tracked on bug 3644077:
https://forums.developer.nvidia.com/t/external-monitor-doesnt-work-on-wayland/214090/4

ghtesting2020, 2022-05-14:
> Thanks for the detailed bug report. Knowing that it also reproduces with the proprietary kernel modules helps. Could you also check if this is a recent regression across NVIDIA driver releases?

This is also an issue for KDE Plasma on Wayland. It's one of the showstoppers that makes me stuck on X11 and one of the reasons I was looking at getting an AMD card next time. However now that Nvidia is welcoming the open source community and this may get fixed, my plans may change and I can stick with Nvidia.

Gert-dev, 2022-05-14:
> Could you also check if this is a recent regression across NVIDIA driver releases?

As mentioned by geminis3 above, it is indeed not a regression, and has never worked with the (proprietary) drivers for me either. I've been testing GNOME Wayland with and without Optimus since EGLStreams was added for Wayland support, again on every new stable driver release, and IIRC Optimus on GNOME Wayland didn't work properly at all until GBM support landed a couple of releases ago, together with which this issue was immediately introduced.

A footnote is that it initially crashed mutter on GNOME entirely when plugging in a second monitor, until they added the graceful degradation in 42.1 (see OP), which simply ignores the NVIDIA dGPU for rendering, and is the state we are in now.

qumaciel, 2022-05-16:
> Thanks for the detailed bug report. Knowing that it also reproduces with the proprietary kernel modules helps. Could you also check if this is a recent regression across NVIDIA driver releases?

This issue is also present in Sway, and also happens using HDMI, not just DP. Cf. discussion when we tried to devise a general sway setup back then: [](https://forums.developer.nvidia.com/t/nvidia-495-on-sway-tutorial-questions-arch-based-distros/192212) .

pablodz, 2022-05-19:
Same here with an old gtx1050

ErickHoward, 2022-05-24:
This also doesn't work for me on Gnome with the NVIDIA proprietary drivers and the following setup:
CPU: AMD Ryzen 5800H
GPU: AMD Ryzen 5800H Vega (IGPU, connected to internal display)
GPU: NVIDIA RTX 3070 Max-Q (DGPU, connected to all display outs)

I have an external monitor connected via HDMI. If I try to login to Gnome on Wayland the external display will not display anything however X11 works fine.

thesword53, 2022-05-26:
Maybe, it's related to #175

erik-kz, 2022-05-27:
This is a known issue with GNOME (or mutter, more specifically). If Sway implemented multi-GPU support in a similar way it would also be affected. The root of the problem is that, while GBM API allows specifying the format of a buffer using DRM format modifiers, this is not sufficient to ensure the buffer can be shared across devices. For instance, there is no way to control whether the buffer is allocated in video memory or system memory. The full and proper solution will probably look something like the "allocation constraints" system described in this XDC talk https://www.youtube.com/watch?v=HZEClOP5TIk

For what it's worth, though, multi-GPU *does* work with KDE / Kwin because it uses an EGLSurface created from a pitch-linear gbm_surface as the "shared" buffer between devices. In that case, we make the slightly dodgy assumption that it should be allocated in system memory since this is needed for PRIME render offload to work. On the other hand, mutter uses a plain gbm_bo attached to a GL framebuffer object, so such that hack doesn't help.

Gert-dev, 2022-05-27:
@erik-kz If I understand correctly, it's a bit [like the tranches for DMA-BUF](https://wayland.app/protocols/linux-dmabuf-unstable-v1#zwp_linux_dmabuf_feedback_v1), but for buffers in GBM?

Is this issue known upstream already (Mesa, if I understood correctly that GBM is part of it)? I'm willing to create such an issue upstream if it is convenient, if you can point me to the right project, just to get the gears in motion.

jadahl, 2022-05-27:
> For what it's worth, though, multi-GPU _does_ work with KDE / Kwin because it uses an EGLSurface created from a pitch-linear gbm_surface as the "shared" buffer between devices

@erik-kz If you mean allocate with `GBM_BO_USE_LINEAR` then mutter does this too when a buffer is supposed to be shareable from the iGPU to the dGPU.

erik-kz, 2022-05-27:
> If you mean allocate with GBM_BO_USE_LINEAR then mutter does this too when a buffer is supposed to be shareable from the iGPU to the dGPU.

I think part of the problem with mutter is that the default "COPY_MODE_SECONDARY_GPU" tries to bind the linear shared buffer to GL_TEXTURE_2D on the secondary GPU, which our driver does not support This is indicated by eglQueryDmaBufFormatsEXT setting external_only to true for DRM_FORMAT_MOD_LINEAR.

Looking at the latest Kwin code more closely, the EGLSurface thing I mentioned might not be correct. I think the difference might actually be that Kwin's default works more like mutter's "COPY_MODE_PRIMARY_GPU" where the copy to the shared buffer happens on the primary GPU and then the secondary GPU just imports it and scans it out.

DivineBicycle, 2022-09-14:
Is there any progress on this issue?

NoTuxNoBux, 2022-12-01:
The error from the OP seems to be slightly different now for me (using new 525 drivers and mutter 43.1):

```
Secondary GPU initialization failed (Failed to create gbm_surface: No such file or directory). Falling back to GPU-less mode instead, so the secondary monitor may be slow to update.
```

In case it helps, [this is a link to the mutter code that fails and logs the warning](https://gitlab.gnome.org/GNOME/mutter/-/blob/978cbe2d/src/backends/native/meta-onscreen-native.c#L2082) ([this line](https://gitlab.gnome.org/GNOME/mutter/-/blob/978cbe2d/src/backends/native/meta-onscreen-native.c#L1897) creates the GBM error). Flags passed to `gbm_surface_create` appear to be `GBM_BO_USE_SCANOUT | GBM_BO_USE_RENDERING`.

lisuke, 2023-01-15:
OS    Arch Linux x86_64
DE    Hyprland
GPU   Intel CoffeeLake-H GT2 [UHD Graphics 630]
GPU   NVIDIA Quadro RTX 5000 Mobile / Max-Q

[ 1607.885648] NVRM osCreateOsDescriptorFromDmaBufPtr: osCreateOsDescriptorFromDmaBufPtr(): Error (86) while trying to import dma_buf!

w9n, 2023-01-15:
[ 1607.885648] NVRM osCreateOsDescriptorFromDmaBufPtr: osCreateOsDescriptorFromDmaBufPtr(): Error (86) while trying to import dma_buf!

same issue on sway with nvidia egpu over thunderbolt.

CoelacanthusHex, 2023-02-10:
```
Graphics:
  Device-1: NVIDIA GA107BM [GeForce RTX 3050 Mobile] driver: nvidia
    v: 525.89.02
  Device-2: AMD Cezanne [Radeon Vega Series / Radeon Mobile Series]
    driver: amdgpu v: kernel
  Device-3: IMC Networks Integrated Camera type: USB driver: uvcvideo
  Display: wayland server: X.org v: 1.21.1.7 with: Xwayland v: 22.1.7
    compositor: kwin_wayland driver: X: loaded: modesetting unloaded: nvidia
    dri: radeonsi,nouveau gpu: nvidia,amdgpu resolution: 1920x1080
  API: OpenGL v: 4.6 Mesa 22.3.4 renderer: AMD Radeon Graphics (renoir LLVM
    15.0.7 DRM 3.49 6.1.11-zen1-1-zen)
```
OS Arch Linux x86_64
DE KDE Wayland
NVRM osCreateOsDescriptorFromDmaBufPtr: osCreateOsDescriptorFromDmaBufPtr(): Error (86) while trying to import dma_buf!

petersaints, 2023-02-25:
I know that it's not an ideal solution, but other than using X11 instead of Wayland, you can just disable the Intel GPU if your laptop allows it. On my laptop I'm able to do that it runs well if both the laptop screen and the external monitor run from the NVIDIA GPU. However, this is not an ideal solution since you will use more battery when you use your laptop disconnected from any screen and you will always have to reboot to change this setting.

ankurdhama, 2023-06-16:
In version 535 the error message is `Secondary GPU initialization failed (Failed to create gbm_surface: Function not implemented)`.

Mouwrice, 2023-11-14:
> Is there any progress on this issue in 545 or do we have to wait few more decades ?

@ankurdhama fixes are being made around this issue on the gnome / mutter side of things.

There are multiple issues and MR's open but this should give you a good entry point:
https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/3304#note_1910885

But if it depends on Nvidia then we are probably better of switching to a respectable company that does not abuse its customers.

vanvugt, 2024-01-25:
Yes a complete fix is coming in [mutter!3304](https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/3304). But part of that is a workaround for [an Nvidia driver quirk](https://forums.developer.nvidia.com/t/gbm-surface-create-fails-if-flags-0/279951) causing the `Failed to create gbm_surface`.

Also I suspect "open-gpu-kernel-modules" is the wrong NVIDIA project to track this in.

ankurdhama, 2024-04-26:
I had a look at https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/3304 and it seems that the fix in that PR is to make mutter use "copy mode gpu" instead of much slower "copy mode cpu". In "copy mode gpu" it seems that mutter is using a shader program to copy the buffer from system ram to GPU framebuffer. This does improve the performance a bit but I think the end goal is to use "copy mode zero" for sharing the buffer between igpu and nvidia dgpu. I am not sure what changes Nvidia driver needs to make so that we can get "copy mode zero" working or is it even possible in a igpu dgpu machine. Is it that nvidia driver needs to support dma buf import to make this happen?

@vanvugt does the above summary looks correct?

Gert-dev, 2024-07-07:
Closing this since I have hardware again that has this setup and it's fixed for me now using mutter 46.3.1. The performance is also a lot better, but still not great, but I think [that is being tracked in mutter already](https://gitlab.gnome.org/GNOME/mutter/-/issues/3544).

closed 2024-07-07 by Gert-dev

vanvugt, 2024-07-08:
Sorry I missed the above message, @ankurdhama. The fix for this bug is https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/3304 which was released in Mutter 46.1. And @Gert-dev is right that there are still open discussions about how to make it faster.