NVIDIA/open-gpu-kernel-modules #162: Night light does not work in GNOME Wayland (missing GAMMA_LUT support)

Gert-dev, 2022-05-13:
**NVIDIA Driver Version**
515.43.04

**GPU**
NVIDIA RTX 2070 Super Max-Q

**Describe the bug**
Enabling night light in GNOME has no effect.

This bug also exists with the proprietary driver, but also seems to exist with the new, open, driver (hence this report).

**To Reproduce**
1. Enable night light in GNOME 42.1 on Wayland.
2. Set the time so that it is currently active.
3. Set the intensity high enough to be noticeable (warm).
4. Observe nothing happening.

**Expected behavior**
The night light works as intended by making the colours of the screen warmer.

**Please reproduce the problem, run nvidia-bug-report.sh, and attach the resulting nvidia-bug-report.log.gz.**

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8690924/nvidia-bug-report.log.gz)

ValZapod, 2022-05-14:
Why would they implement it when little cms is so much better? Also since when do you need a LUT to do global dimming???

Newbytee, 2022-05-14:
Upstream bug report: https://gitlab.gnome.org/GNOME/mutter/-/issues/290

quaqadm, 2022-08-25:
i am with same problem.

santhosh-reddy03, 2022-10-13:
Hi Team,
I am facing this issue on x11 as well, when i click on the night light and move the slider, it turns on for a movement and exits from there, there after.
Nvidia Driver Version: 515.65.01
Please look into this issue at the earliest possible time.

pyt0xic, 2022-10-14:
Not sure if it helps but Gammastep works flawlessly for me with wlroots compositors.
Maybe worth a look...

MahouShoujoMivutilde, 2022-10-17:
@pyt0xic could you tell me more about your system?

I am on 520.56.06 _proprietary_ driver (pre-turing gpu), but I would expect that `gammastep` would either work on both or fail on both since "This code base is shared with NVIDIA's proprietary drivers" (from readme).

Hyprland compositor.

Launching `gammastep -m wayland -PO 2600` gives

```
Warning: Zero outputs support gamma adjustment.
Warning: 1/1 output(s) do not support gamma adjustment.
```

What compositor do you use?

pyt0xic, 2022-10-18:
I was using Sway and am now running Hyprland.
I have an Optimus laptop but usually force dGPU.

I'll add more once I'm back from work.
But maybe double check all your Wayland related env vars.

yannjor, 2022-12-17:
> I was using Sway and am now running Hyprland. I have an Optimus laptop but usually force dGPU.
> 
> I'll add more once I'm back from work. But maybe double check all your Wayland related env vars.

Could you please share your env vars?

LukasF1337, 2023-02-24:
It is a known missing feature that GAMMA_LUT is not implemented in nvidias wayland driver yet:
http://us.download.nvidia.com/XFree86/Linux-x86_64/525.89.02/README/wayland-issues.html
which will be implemented at some point in the future according to their site.
@santhosh-reddy03 on my system setting the gamma LUT on X11 in KDE Plasma works.

santhosh-reddy03, 2023-02-24:
Actually somehack from ubuntu launchpad bugs helped me fix the issue, i don't remember the variable name but on setting it to false, everything worked as usual, and glad to know nvidia is on it , hope to see in some time in future.

rwjack, 2023-02-28:
@santhosh-reddy03 well which variable was it?

santhosh-reddy03, 2023-02-28:
@rwjack there were two issues, the pc was also not waking up from sleep, i am think its Ubuntu 22.04 specific issue in my case, i m attaching the [bug link](https://bugs.launchpad.net/mutter/+bug/1968040) here, it has worked for me at that time, since then i switched back to x11.

rwjack, 2023-02-28:
Yeah I'm switching back to x11 until I get a new GPU.

MahouShoujoMivutilde, 2023-02-28:
I am not using gnome, but if there is some software that lets you draw your own custom shader as an overlay for the whole screen - you can make your own night light.

In hyprland there is an option for that

```
# hyprland.conf

decoration {
    screen_shader = ~/.config/hypr/your-shader.glsl
}

```

Here is the shader I use

https://github.com/hyprwm/Hyprland/issues/1140#issuecomment-1335128437

<details>
  <summary>Result: </summary>

No shader:

[image]

Shader:

[image]

</details>

It is trivial to automate it to apply on sunset.

Idk if gnome can do custom shaders, maybe ask the devs nicely to make a fallback implementation for nightlight if nvidia gpu is detected, because it doesn't seem like GAMMA_LUT will happen on nvidia any time soon.

eszlari, 2023-03-09:
One year ago a Nvidia dev wrote:

> Support for GAMMA_LUT is in the queue, but no ETA at the moment.

https://github.com/Plagman/gamescope/pull/454#issuecomment-1076680841

eszlari, 2023-04-10:
https://forums.developer.nvidia.com/t/nvidia-bug-kde-wayland-games-are-force-vsynced/237880/3

> Work on Gamma LUT support is also in progress. The implementation is currently under internal code-review.

rwjack, 2023-04-10:
Yeah, that quote is from the 28th of December 2022.

aviallon, 2023-05-30:
@rwjack Finally!

> The vsync issue will be fixed in the 545 driver release. Unfortunately I wasn’t able to get it in in time for the 535 release which will go public in a few days, sorry. But 545 will be the next release after that (we’re skipping 540 for whatever reason).

https://forums.developer.nvidia.com/t/nvidia-bug-kde-wayland-games-are-force-vsynced/237880/21

Newbytee, 2023-05-30:
> @rwjack Finally!
>
>
> https://forums.developer.nvidia.com/t/nvidia-bug-kde-wayland-games-are-force-vsynced/237880/21

How is this related to night light/GAMMA_LUT?

rwjack, 2023-05-31:
> How is this related to night light/GAMMA_LUT?

Just Nvidia smokescreen trying to keep us onboard :joy:

aviallon, 2023-05-31:
> How is this related to night light/GAMMA_LUT?

Sadly, the two subjects were discussed at the same time in the same thread.
And I misread 😢

eszlari, 2023-06-14:
Plasma 6 will support night color on Nvidia without GAMMA_LUT:

https://invent.kde.org/plasma/kwin/-/merge_requests/4130

zer0-x, 2023-06-14:
> Plasma 6 will support night color on Nvidia without GAMMA_LUT:
> 
> https://invent.kde.org/plasma/kwin/-/merge_requests/4130

It uses a shader based fallback. It's just a work around, not an actual support.

It is not efficient and I think that colours will be also different when capturing the screen, since it's not a GPU side based solution (didn't test it, I'm not sure).

Sohail-XD, 2023-07-05:
@MahouShoujoMivutilde this worked thanks, btw can I get dots for the bar in your picture?

MahouShoujoMivutilde, 2023-07-05:
@Sohail-XD The bar evolved since then, here is how the current one looks:

<details>
  <summary> Waybar</summary>

[image]

</details>

configs:

<details>
  <summary> waybar/config </summary>

```jsonc

// vim: set ft=jsonc:
{
    "layer": "top",
    "position": "top", // Waybar position (top|bottom|left|right)
    "height": 30, // Waybar height (to be removed for auto height)
    // "width": 1280, // Waybar width
    "spacing": 0, // Gaps between modules (set in style.css for mor precision)
    "modules-left": [
        "cpu",
        "wlr/workspaces"
    ],
    "modules-center": [
        "hyprland/window"
    ],
    "modules-right": [
        "custom/isrecording",
        "custom/nvtemp",
        "custom/cputemp",
        "custom/memory",
        "idle_inhibitor",
        "pulseaudio",
        "bluetooth",
        "hyprland/language",
        "custom/sunset",
        "clock",
        "tray",
        "custom/notification"
    ],
    "idle_inhibitor": {
        "format": "{icon}",
        "format-icons": {
            "activated": "",
            "deactivated": ""
        }
    },
    "cpu": {
        "interval": 1,
        // cores including HT
        "format": "{icon0} {icon1} {icon2} {icon3} {icon4} {icon5} {icon6} {icon7} {icon8} {icon9} {icon10} {icon11} {icon12} {icon13} {icon14} {icon15}",
        "format-icons": [ "▁", "▂", "▃", "▄", "▅", "▆", "▇", "█" ],
    },
    "custom/isrecording": {
        "exec": "isrecording",
        "interval": 1
    },
    "custom/memory": {
        "exec": "~/.config/waybar/scripts/waybar-mem",
        "interval": 1
    },
    "custom/nvtemp": {
        "format": "3D {} °C",
        "exec": "nvtemp 65",
        "interval": 2
    },
[... 217 lines cut ...]
{
  margin-left: 7px;
  margin-right: 7px;
  border-radius: 26px;
}

#window {
  font-family: Sans;
  background: rgba(65, 8, 145, 0.45)
}

#bluetooth {
  /* font-family: Monospace; */
  font-size: 12px;
  /* background: rgba(27, 22, 255, 0.65); */
  background: rgba(7, 30, 49, 0.85);
}

#idle_inhibitor.activated {
  color: #bc8b3c;
}

```

</details>

mirh, 2023-10-17:
GAMMA_LUT landed in the first R545 beta

rwjack, 2023-10-17:
Wohoo, finally!!

Anyone have an estimate on how long till the beta driver isn't beta anymore, and is uploaded to the ubuntu apt?

NTsch, 2023-11-01:
It's out of Beta now: https://www.nvidia.com/download/driverResults.aspx/214100/en-us/

gilvbp, 2023-11-01:
> Link to the driver in question - https://www.nvidia.com/download/driverResults.aspx/212964/en-us/
> 
> This fixes _many_ of the remaining (X)Wayland issues.
> 
> Great job Nvidia!
> 
> EDIT: Nvidia now keeps track of the remaining Wayland issues here https://forums.developer.nvidia.com/t/wayland-information-for-r545-beta-release/214275

Yes! There is a lot of improvement! But it still missing the multi-GPU mosaic support! But it's on track here:

https://github.com/NVIDIA/egl-wayland/issues/91#issuecomment-1785740677

Gert-dev, 2024-07-07:
Closing this since I have hardware again that has this setup and it's fixed for me now using GNOME and recent NVIDIA drivers. I can successfully use night light now on Wayland.

closed 2024-07-07 by Gert-dev

exil0867, 2024-07-07:
Also fixed on the latest version of KDE Plasma.