NVIDIA/open-gpu-kernel-modules #165: Provide driver for muslc, to install it in musl distros

TheMBL09898, 2022-05-13:
Please provide driver for musl. It would make installing the nvidia drvers possible on Alipne Linux, or other distributions using musl, such as my Gentoo Linux with musl profile. As far as I know you only need to compile it with musl, it isn't that big of a deal. People have been asking for this since 2019 in your github at least, and now that you've "open-sourced" your drivers I think you should be able to do this as well.
Reference: https://github.com/NVIDIA/nvidia-installer/issues/10

X547, 2022-05-13:
I think that it is not subject for this repo. It is better suited for proprietary drivers support.

aaronp24, 2022-05-13:
Thank you for your interest in the NVIDIA driver. This repository contains code for driver components that run in the kernel, which has its own C library implementation. Please address questions about other NVIDIA driver components to [the NVIDIA developer forums](https://forums.developer.nvidia.com/c/gpu-graphics/linux/148).

closed 2022-05-13 by aaronp24