NVIDIA/open-gpu-kernel-modules #170: Makefile problem in arch linux

squarerootx86, 2022-05-14:
**NVIDIA Driver Version**
installed linux just few minutes ago

**GPU**
Nvidia 3070 Ti

**Describe the bug**
[ nvidia            ]  CC           ../common/uproc/os/libos-v2.0.0/debug/lines.c
 [ nvidia            ]  CC           ../common/uproc/os/libos-v2.0.0/debug/logdecode.c
src/libraries/mmu/mmu_walk.c: In function 'mmuWalkProcessPdes':
src/libraries/mmu/mmu_walk.c:759:78: warning: 'entryIndexFillStart' may be used uninitialized [-Wmaybe-uninitialized]
  759 | progress == (entryIndexFillEnd - entryIndexFillStart + 1),
      |                                                      ^

src/libraries/mmu/mmu_walk.c:684:22: note: 'entryIndexFillStart' was declared here
  684 |         NvU32        entryIndexFillStart;
      |                      ^~~~~~~~~~~~~~~~~~~
src/libraries/mmu/mmu_walk.c:418:82: warning: 'entryIndexFillStart' may be used uninitialized [-Wmaybe-uninitialized]
  418 | progress != (entryIndexFillEnd - entryIndexFillStart + 1))
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~

src/libraries/mmu/mmu_walk.c:298:19: note: 'entryIndexFillStart' was declared here
  298 |             NvU32 entryIndexFillStart;
      |                   ^~~~~~~~~~~~~~~~~~~
In file included from inc/kernel/core/printf.h:38,
                 from inc/kernel/core/prelude.h:44,
                 from src/kernel/rmapi/nv_gpu_ops.c:24:
inc/libraries/utils/nvprintf.h: In function 'nvGpuOpsDeviceCreate':
inc/libraries/utils/nvprintf.h:305:30: warning: 'linkBandwidthMBps' may be used uninitialized [-Wmaybe-uninitialized]
  305 | #define NVRM_PRINTF_FUNCTION nvDbg_Printf
      |                              ^~~~~~~~~~~~
src/kernel/rmapi/nv_gpu_ops.c:1636:11: note: 'linkBandwidthMBps' declared here
 1636 |     NvU32 linkBandwidthMBps;
      |           ^~~~~~~~~~~~~~~~~
/bin/sh: line 1: hostname: command not found
 [ nvidia            ]  CC           _out/Linux_x86_64/g_nvid_string.c
 [ nvidia            ]  LD           _out/Linux_x86_64/nv-kernel.o
 [ nvidia            ]  OBJCOPY      _out/Linux_x86_64/nv-kernel.o
make[1]: Leaving directory '/home/user/nvidia/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/user/nvidia/kernel-open'
make[2]: Entering directory '/home/user/nvidia/kernel-open'
make[2]: *** /lib/modules/5.15.38-1-lts/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/user/nvidia/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/user/nvidia/kernel-open'
make: *** [Makefile:50: modules] Error 2

[user@khalifa nvidia]$ sudo make modules -j`nproc`
make -C src/nvidia
make -C src/nvidia-modeset
make[1]: Entering directory '/home/user/nvidia/src/nvidia'
make[1]: Entering directory '/home/user/nvidia/src/nvidia-modeset'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/home/user/nvidia/src/nvidia-modeset'
cd kernel-open/nvidia-modeset/ && ln -sf ../../src/nvidia-modeset/_out/Linux_x86_64/nv-modeset-kernel.o nv-modeset-kernel.o_binary
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/home/user/nvidia/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/user/nvidia/kernel-open'
make[2]: Entering directory '/home/user/nvidia/kernel-open'
make[2]: *** /lib/modules/5.15.38-1-lts/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/user/nvidia/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/user/nvidia/kernel-open'
make: *** [Makefile:50: modules] Error 2

**To Reproduce**
git clone "url" "dir"
cd "dir"
make modules -j`nproc`

ra1nst0rm3d, 2022-05-14:
Damn. Install linux-lts-headers package

aritger, 2022-05-15:
Yes, the kernel modules cannot be built without access to the target kernel's header files.  Closing.

closed 2022-05-15 by aritger

ThisNekoGuy, 2022-05-15:
@aritger Similar issue, though I do have my kernel headers installed but with a custom kernel :/
- https://github.com/Frogging-Family/nvidia-all/issues/104#issuecomment-1126636105

```
. . .
1 warning generated.
1 warning generated.
date: version.mk: No such file or directory
 [ nvidia            ]  CC           _out/Linux_x86_64/g_nvid_string.c
 [ nvidia            ]  LD           _out/Linux_x86_64/nv-kernel.o
 [ nvidia            ]  OBJCOPY      _out/Linux_x86_64/nv-kernel.o
make[1]: Leaving directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[2]: Entering directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[2]: *** /usr/src/linux: No such file or directory.  Stop.
make[2]: Leaving directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/neko-san/nvidia/nvidia-all/src/open-gpu-kernel-modules-515.43.04/kernel-open'
make: *** [Makefile:50: modules] Error 2
```

aritger, 2022-05-16:
From the error message ("/usr/src/linux: No such file or directory.  Stop."), I guess it must be looking for your kernel header files in /usr/src/linux.  Is that what your /lib/modules/`uname -r` points to?  I fear this is a system configuration issue rather than a bug with open-gpu-kernel-modules

ThisNekoGuy, 2022-05-16:
> From the error message ("/usr/src/linux: No such file or directory. Stop."), I guess it must be looking for your kernel header files in /usr/src/linux. Is that what your /lib/modules/`uname -r` points to? I fear this is a system configuration issue rather than a bug with open-gpu-kernel-modules

```bash
neko-san@ARCH ~> uname -r
5.17.2-256-tkg-pds-llvm
neko-san@ARCH ~> uname rmodinfo /usr/lib/modules/$(uname -r)/
modinfo: ERROR: Module /usr/lib/modules/5.17.2-256-tkg-pds-llvm/ not found
```
Probably not typing that right? But that's how it comes out :/
I'm not sure if I

sTiKyt, 2022-05-16:
> > From the error message ("/usr/src/linux: No such file or directory. Stop."), I guess it must be looking for your kernel header files in /usr/src/linux. Is that what your /lib/modules/`uname -r` points to? I fear this is a system configuration issue rather than a bug with open-gpu-kernel-modules
> 
> ```shell
> neko-san@ARCH ~> uname -r
> 5.17.2-256-tkg-pds-llvm
> neko-san@ARCH ~> ls /usr/lib/modules/
> 5.4.17-1-lts  5.17.2-256-tkg-pds  5.17.2-256-tkg-pds-llvm
> ```
> 
> This is how it comes out :/

I'm not sure if i'm right, but to me it seems like your issue if different, cuz OP didn't have headers package installed, and you do, maybe worth making new issue? Correct me if i'm wrong

ptr1337, 2022-05-16:
Actually his issue is related to a kernel compile with llvm where the module compiles fails.