NVIDIA/open-gpu-kernel-modules #171: [Question] How to check if you have installed them correctly

NotZer0Two, 2022-05-14:
i want check if i installed the kernel modules and if its working, im sure i installed them because my pc is 200% more faster but i want to be sure. Thanks if someone want help me

ptr1337, 2022-05-14:
https://us.download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/kernel_open.html

```

E.g., the proprietary flavor will report:

    # modinfo nvidia | grep license
    license:        NVIDIA

    # cat /proc/driver/nvidia/version
    NVRM version: NVIDIA UNIX x86_64 Kernel Module  [...]

The open flavor will report:

    # modinfo nvidia | grep license
    license:        Dual MIT/GPL

    # cat /proc/driver/nvidia/version
    NVRM version: NVIDIA UNIX Open Kernel Module for x86_64  [...]
```

Also the dmesg log should be differnt

NotZer0Two, 2022-05-14:
ok i will check it thanks for the help

NotZer0Two, 2022-05-14:
yes i have them installed correctly because i was encounting issues about the drivers but i wasn't sure but when i installed them back with the ubuntu driver command was ok thanks for the help

closed 2022-05-14 by NotZer0Two