NVIDIA/open-gpu-kernel-modules #175: True DMABUF support

gnif, 2022-05-14:
Now that the sources are no longer restricted from using GPL symbols, can we please have true Linux DMABUF import support?

See:
https://01.org/linuxgraphics/gfx-docs/drm/driver-api/dma-buf.html
https://www.phoronix.com/scan.php?page=news_item&px=MTIwNDI
https://www.phoronix.com/scan.php?page=news_item&px=MTA1OTU

This would greatly benefit projects like [Looking Glass](https://looking-glass.io) were we are importing full screen textures directly from memory shared with a VM which is writing into the shared memory.

Note: This is already working on AMD and Intel GPUs, we have to recommend against using NVidia GPUs on a daily basis due to the lack of this feature.

YusufKhan-improves-shit, 2022-05-14:
REALLY WANT THIS!!!!!

ad1rie1, 2022-05-16:
Hello, Are you sure is not already the case? , the GPL function mentioned in the Phoronix article is used.
 https://github.com/NVIDIA/open-gpu-kernel-modules/blob/d8f3bcff924776518f1e63286537c3cf365289ac/kernel-open/nvidia/nv-dmabuf.c#L721

https://github.com/NVIDIA/open-gpu-kernel-modules/blob/d8f3bcff924776518f1e63286537c3cf365289ac/kernel-open/nvidia/nv-dmabuf.c#L739

you can see the use of the GPL function.

gnif, 2022-05-16:
> Hello, Are you sure is not already the case? , the GPL function mentioned in the Phoronix article is used. [...]

Yes it will create a DMABUF for export, but will not use one from another source for import.