NVIDIA/open-gpu-kernel-modules #179: Display Stream Compression (DSC) support

edisionnano, 2022-05-14:
DSC allows people to drive 4k/240hz and 8k/60hz displays with DP 1.4 and HDMI 2.1. The NVIDIA driver for Windows supports this however the Linux one doesn't. Many people in the Linux Gaming community want this and moving into the future 8K or high refresh rate 4k will become more common for movies or gaming. Linux is a great choice for building a media center pc and nvidia cards are best in class for decoding video such as 8k60 so such a feature would be more than welcome

ValZapod, 2022-05-14:
It is also lossy...

aritger, 2022-05-15:
Yes, thanks.  The Linux driver in general does support DSC, but not yet at the higher rates that you're referring to.  I don't have an ETA for that, but we know it is an important feature that many people are asking for.  Thanks.

Evernow, 2022-05-18:
> Yes, thanks. The Linux driver in general does support DSC, but not yet at the higher rates that you're referring to. I don't have an ETA for that, but we know it is an important feature that many people are asking for. Thanks.

Thank you for confirming this.